abperiasamy/rtl8812AU_8821AU_linux #146: Cannot compile due to issue with ioctl_cfg80211.c

anastasiadecleyre, 2016-09-10:
Was unable to compile using either a simple `make` and `make install` or using `dkms`.

The `sudo dkms build -m rtl8812AU_8821AU_linux -v 1.0` command fails with:
```
Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
'make' KVER=4.4.0-36-generic.........(bad exit status: 2)
ERROR (dkms apport): binary package for rtl8812AU_8821AU_linux: 1.0 not found
Error! Bad return status for module build on kernel: 4.4.0-36-generic (x86_64)
Consult /var/lib/dkms/rtl8812AU_8821AU_linux/1.0/build/make.log for more information.
```
and the `make.log` contains:
```
/var/lib/dkms/rtl8812AU_8821AU_linux/1.0/build/os_dep/linux/ioctl_cfg80211.c:1704:40: error: variable ‘info’ has initializer but incomplete type
                                 struct cfg80211_scan_info info = {
                                        ^
/var/lib/dkms/rtl8812AU_8821AU_linux/1.0/build/os_dep/linux/ioctl_cfg80211.c:1705:41: error: unknown field ‘aborted’ specified in initializer
                                         .aborted = aborted,
                                         ^
/var/lib/dkms/rtl8812AU_8821AU_linux/1.0/build/os_dep/linux/ioctl_cfg80211.c:1705:52: warning: excess elements in struct initializer
                                         .aborted = aborted,
                                                    ^
/var/lib/dkms/rtl8812AU_8821AU_linux/1.0/build/os_dep/linux/ioctl_cfg80211.c:1705:52: note: (near initialization for ‘info’)
/var/lib/dkms/rtl8812AU_8821AU_linux/1.0/build/os_dep/linux/ioctl_cfg80211.c:1704:59: error: storage size of ‘info’ isn’t known
                                 struct cfg80211_scan_info info = {
                                                           ^
scripts/Makefile.build:258: recipe for target '/var/lib/dkms/rtl8812AU_8821AU_linux/1.0/build/os_dep/linux/ioctl_cfg80211.o' failed
```
You can find a copy of the `make.log` file here: [make.log](https://github.com/abperiasamy/rtl8812AU_8821AU_linux/files/465580/make.log.txt)

asokolsky, 2016-09-11:
Same problem here
`System:    Host: Folio13 Kernel: 4.4.0-36-generic x86_64 (64 bit) Desktop: Cinnamon 3.0.7
           Distro: Linux Mint 18 Sarah
`

harshavardhana, 2016-09-11:
https://github.com/abperiasamy/rtl8812AU_8821AU_linux/commit/09a1e212b643c2ad01824fda27ce613489ba2de4 - issue was introduced here reverting.

harshavardhana, 2016-09-11:
Fixed in - https://github.com/abperiasamy/rtl8812AU_8821AU_linux/commit/6bd11fc5bb484114f83d5d78bcb09f8086352ef3

anastasiadecleyre, 2016-09-11:
Thanks, will check now.

anastasiadecleyre, 2016-09-11:
It compiles. Closing.

closed 2016-09-11 by anastasiadecleyre