abperiasamy/rtl8812AU_8821AU_linux #152: Ubuntu 16.10 kernel 4.8.0-17 make fails

wolfland, 2016-09-30:
Make fails with Ubuntu 16.10 kernel 4.8.0-17.

 I used the code with the fix for issue #135 to pick up that fix.

The error is as follows:

./include/net/cfg80211.h:4113:6: note: expected ‘struct cfg80211_scan_info *’ but argument is of type ‘bool {aka _Bool}’

This issue also occurs with the AC600 build and has been reported as issue #30 at their
git page:

https://github.com/Grawp/rtl8812au_rtl8821au.git

You probably already know of this issue but I thought I better report it just in case.

Please see the AC600 git page issue #30 for details

If this is a known issue or a duplicate feel free to mark it as such or close it.

Thanks,

Fjodor42, 2016-10-03:
Hi Wolfland,

You should probably use current HEAD, or, at least, pull in the code from PR #148 as well.

wolfland, 2016-10-05:
Ok.. For my education I assume that HEAD means the "lastest" version ?

I need to get up to speed on git and it's terminology.

I will also pull in PR #148 code.

Thanks

Fjodor42, 2016-10-05:
Hi Wolfland,

That is correct, yes. Incidentally, though, my own use might have been a bit too broad, as every branch has a "HEAD", so "HEAD of the master branch" was probably what I should have written.

ptpt52, 2016-11-04:
notice it is fixed

closed 2016-11-04 by ptpt52