abperiasamy/rtl8812AU_8821AU_linux #154: Undefined variable in os_dep/linux/ioctl_linux.c

notorand-it, 2016-10-03:
I have got this error:

```
rtl8812AU_8821AU_linux/os_dep/linux/ioctl_linux.c: In function ‘rtw_wx_set_priv’:
rtl8812AU_8821AU_linux/os_dep/linux/ioctl_linux.c:8659:2: error: ‘i’ undeclared (first use in this function)
  i = rtw_android_cmdstr_to_num(ext);
  ^
rtl8812AU_8821AU_linux/os_dep/linux/ioctl_linux.c:8659:2: note: each undeclared identifier is reported only once for each function it appears in
```

despite the code should be enclose within a `#ifdef CONFIG_ANDROID`

Anyway, I just went to an earlier line (no.8575) and changed `//int i;` to `int i;` in order to fix it.
Compilation went through smoothly. I'll check later about the actual effectiveness of the driver.

mkreisl, 2016-10-07:
Same here.
I'm wondering why this is in #define CONFIG_ANDROID section

ptpt52, 2016-10-08:
hi, can you try this code out? this is a new version base
https://github.com/ptpt52/rtl8812au

mkreisl, 2016-10-12:
@ptpt52 
Does not compile without errors

ptpt52, 2016-10-12:
@mkreisl 
just some warning fix

`
diff --git a/Makefile b/Makefile
index 68ef622..b39cf1d 100644
--- a/Makefile
+++ b/Makefile
@@ -3,7 +3,7 @@ EXTRA_CFLAGS += -O1
 #EXTRA_CFLAGS += -O3
 EXTRA_CFLAGS += -Wall
 EXTRA_CFLAGS += -Wextra
-EXTRA_CFLAGS += -Werror
+#EXTRA_CFLAGS += -Werror
 #EXTRA_CFLAGS += -pedantic
 #EXTRA_CFLAGS += -Wshadow -Wpointer-arith -Wcast-qual -Wstrict-prototypes -Wmissing-prototypes
 EXTRA_CFLAGS += -Wframe-larger-than=1536
`

mkreisl, 2016-10-12:
Much better, module will be built successfully :smile: but getting many many of those
`warning: comparison between signed and unsigned integer expressions [-Wsign-compare]` 
warnings. Ok, they can be hidden by adding `EXTRA_CFLAGS += -Wno-sign-compare` to Makefile but would be better to solve them in code directly

ptpt52, 2016-10-12:
> Much better, module will be built successfully 😄 but getting many many

I don't have that environment for arm building, so can't help to fix that
warning for you