abperiasamy/rtl8812AU_8821AU_linux #156: Driver not working with Linux 4.8

libcg, 2016-10-06:
```
[32676.616518] ------------[ cut here ]------------
[32676.616540] WARNING: CPU: 0 PID: 2518 at net/wireless/core.c:1089 cfg80211_netdev_notifier_call+0x5cf/0x610 [cfg80211]
[32676.616543] Modules linked in: 8812au(O) nls_utf8 isofs sr_mod cdrom rndis_host cdc_ether usbnet mii fuse ipt_MASQUERADE nf_nat_masquerade_ipv4 xfrm_user xfrm_algo iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 xt_addrtype xt_conntrack nf_nat nf_conntrack br_netfilter bridge stp llc dm_thin_pool dm_persistent_data dm_bio_prison dm_bufio libcrc32c crc32c_generic loop dm_mod xt_multiport iptable_filter bnep snd_hda_codec_realtek snd_hda_codec_generic snd_hda_codec_hdmi nls_iso8859_1 nls_cp437 vfat fat joydev input_leds led_class mousedev intel_rapl x86_pkg_temp_thermal intel_powerclamp coretemp mei_wdt snd_usb_audio snd_usbmidi_lib snd_rawmidi snd_seq_device cfg80211 btusb btrtl hid_generic kvm_intel kvm snd_hda_intel irqbypass iTCO_wdt evdev iTCO_vendor_support ppdev mac_hid crct10dif_pclmul
[32676.616624]  crc32_pclmul crc32c_intel ghash_clmulni_intel aesni_intel aes_x86_64 snd_hda_codec lrw gf128mul glue_helper ablk_helper snd_hda_core e1000e cryptd i915 snd_hwdep snd_pcm pcspkr intel_cstate intel_rapl_perf mei_me snd_timer snd i2c_i801 ptp psmouse soundcore mei pps_core intel_gtt i2c_smbus thermal shpchp fan hci_uart battery btbcm btqca btintel parport_pc bluetooth ftdi_sio parport uas usbhid usb_storage usbserial wmi i2c_hid rfkill hid video intel_lpss_acpi intel_lpss tpm_infineon tpm_tis acpi_als tpm_tis_core tpm kfifo_buf fjes acpi_pad industrialio button sch_fq_codel ip_tables x_tables ext4 crc16 jbd2 mbcache sd_mod serio_raw atkbd libps2 ahci xhci_pci libahci xhci_hcd libata scsi_mod usbcore usb_common i8042 serio amdgpu i2c_algo_bit drm_kms_helper syscopyarea sysfillrect sysimgblt
[32676.616718]  fb_sys_fops ttm drm [last unloaded: 8812au]
[32676.616728] CPU: 0 PID: 2518 Comm: NetworkManager Tainted: G        W  O    4.8.0-1-ARCH #1
[32676.616730] Hardware name: Gigabyte Technology Co., Ltd. To be filled by O.E.M./B150M-D3H-CF, BIOS F5 11/02/2015
[32676.616734]  0000000000000286 00000000f23191da ffff88085425b648 ffffffff812fe240
[32676.616740]  0000000000000000 0000000000000000 ffff88085425b688 ffffffff8107c85b
[32676.616746]  000004415425b6a0 ffff8807c9f55c00 ffff880853aef2a0 ffff8806dbe10000
[32676.616752] Call Trace:
[32676.616763]  [<ffffffff812fe240>] dump_stack+0x63/0x83
[32676.616769]  [<ffffffff8107c85b>] __warn+0xcb/0xf0
[32676.616773]  [<ffffffff8107c98d>] warn_slowpath_null+0x1d/0x20
[32676.616785]  [<ffffffffa0abda5f>] cfg80211_netdev_notifier_call+0x5cf/0x610 [cfg80211]
[32676.616790]  [<ffffffff810bfdb4>] ? __wake_up+0x44/0x50
[32676.616797]  [<ffffffff814c7b8f>] ? sock_def_error_report+0x6f/0x80
[32676.616802]  [<ffffffff8109cbca>] notifier_call_chain+0x4a/0x70
[32676.616806]  [<ffffffff8109cd06>] raw_notifier_call_chain+0x16/0x20
[32676.616810]  [<ffffffff814dc065>] call_netdevice_notifiers_info+0x35/0x60
[32676.616815]  [<ffffffff814e5850>] __dev_notify_flags+0x80/0xd0
[32676.616820]  [<ffffffff814e5f72>] dev_change_flags+0x52/0x60
[32676.616824]  [<ffffffff814f6c9f>] do_setlink+0x31f/0xcc0
[32676.616829]  [<ffffffff8132c4a1>] ? __nla_reserve+0x41/0x50
[32676.616835]  [<ffffffff810b1f2a>] ? update_curr+0x6a/0x190
[32676.616841]  [<ffffffff810b2cb4>] ? dequeue_entity+0x254/0xb20
[32676.616845]  [<ffffffff81597d0a>] ? inet6_fill_link_af+0x1a/0x30
[32676.616849]  [<ffffffff814f5078>] ? rtnl_fill_ifinfo+0xba8/0x1080
[32676.616853]  [<ffffffff8132c271>] ? nla_parse+0x31/0x120
[32676.616858]  [<ffffffff814f9576>] rtnl_newlink+0x5c6/0x860
[32676.616865]  [<ffffffff812b4100>] ? security_capable+0x20/0x60
[32676.616869]  [<ffffffff81085f98>] ? ns_capable_common+0x68/0x80
[32676.616872]  [<ffffffff81085fc3>] ? ns_capable+0x13/0x20
[32676.616877]  [<ffffffff814f98f6>] rtnetlink_rcv_msg+0xe6/0x210
[32676.616883]  [<ffffffff811e56b5>] ? __kmalloc_node_track_caller+0x205/0x2c0
[32676.616887]  [<ffffffff814ce667>] ? __alloc_skb+0x87/0x1e0
[32676.616892]  [<ffffffff814f9810>] ? rtnl_newlink+0x860/0x860
[32676.616896]  [<ffffffff81519204>] netlink_rcv_skb+0xa4/0xc0
[32676.616901]  [<ffffffff814f6208>] rtnetlink_rcv+0x28/0x30
[32676.616904]  [<ffffffff81518aec>] netlink_unicast+0x17c/0x240
[32676.616908]  [<ffffffff81518fc8>] netlink_sendmsg+0x348/0x3b0
[32676.616912]  [<ffffffff814c4307>] sock_sendmsg+0x17/0x30
[32676.616916]  [<ffffffff814c4aad>] ___sys_sendmsg+0x2cd/0x2e0
[32676.616921]  [<ffffffff810fef30>] ? futex_wake+0x90/0x170
[32676.616926]  [<ffffffff8121d4a0>] ? poll_select_copy_remaining+0x150/0x150
[32676.616932]  [<ffffffff81256566>] ? eventfd_ctx_read+0x166/0x1e0
[32676.616936]  [<ffffffff810a6c00>] ? wake_up_q+0x80/0x80
[32676.616941]  [<ffffffff8125663c>] ? eventfd_read+0x5c/0x90
[32676.616945]  [<ffffffff81227507>] ? __fget+0x77/0xb0
[32676.616949]  [<ffffffff812275a5>] ? __fget_light+0x25/0x60
[32676.616953]  [<ffffffff814c5c04>] __sys_sendmsg+0x54/0x90
[32676.616958]  [<ffffffff814c5c52>] SyS_sendmsg+0x12/0x20
[32676.616964]  [<ffffffff815f7cf2>] entry_SYSCALL_64_fastpath+0x1a/0xa4
[32676.616967] ---[ end trace 9ee5ecd620e3bbb9 ]---
```

ptpt52, 2016-10-08:
hi, can you try this code out? this is a new version base
https://github.com/ptpt52/rtl8812au

libcg, 2016-10-08:
Thanks I will.

Looking at the commit history of this repo I'm a bit concerned, it looks like there is no review at all. It's doing more harm than good

Grief, 2016-10-10:
@ptpt52 why do you spam with this fork in every single issue while "abperiasamy:master and ptpt52:master are identical."?

libcg, 2016-10-11:
@ptpt52 your repo doesn't work, but https://github.com/Grawp/rtl8812au_rtl8821au does. I think it's a slightly more recent version of the driver.

weedy, 2016-11-01:
Newest stable https://github.com/diederikdehaas/rtl8812AU/commits/driver-4.3.20
Repo also has beta 4.3.22