abperiasamy/rtl8812AU_8821AU_linux #158: TP-LINK T4UH driver not work in Ubuntu 16.04.1

ol3gart, 2016-10-08:
Hello!
I have a usb TP-LINK T4UH v.1.
And driver is not work on Ubuntu 16.04.1.
But sometime ago, it works!
I just reinstall my OS, and try to git && make || dkms - driver is installed, system see wifi, see the wifi-networks, but after enter password - OS cant establish the connection and ask me the password one more and getting the loop (password correct).

jshthornton, 2016-10-08:
Getting the exact same issue. Fresh install of 16.04.1, with 4.4.0-38-generic.
After installing the drivers the device can find the networks, but can not connect to any.

ol3gart, 2016-10-08:
I buy this wifi stick at 2016-08-31, having a trouble to install it in that day and than found  this git in 1-2 days, so in earlier September when i install this, driver works perfectly.

ptpt52, 2016-10-08:
hello, Please try this new version out
https://github.com/ptpt52/rtl8812au
help me to improve it

ol3gart, 2016-10-09:
@ptpt52 
Great work!
Device is working now!
Thx!

ol3gart, 2016-10-15:
Found bug in network driver - it works, but one time in 3-5 minutes appear bug - lags in network connection for 5-7 seconds.
It was an old version driver too.

jshthornton, 2016-10-16:
@ptpt52 Does not work on my device. Get this error when building:
```
Building module:
cleaning build area....
make KERNELRELEASE=4.4.0-38-generic KVER=4.4.0-38-generic....(bad exit status: 2)
ERROR (dkms apport): binary package for rtl8812au: 4.3.14 not found
Error! Bad return status for module build on kernel: 4.4.0-38-generic (x86_64)
Consult /var/lib/dkms/rtl8812au/4.3.14/build/make.log for more information.
Makefile.dkms:18: recipe for target 'build' failed
make: *** [build] Error 10
```

ptpt52, 2016-10-16:
the latest HEAD fixed it.

closed 2016-10-16 by ptpt52

jshthornton, 2016-10-16:
@ptpt52 doesn't work at all now. The driver compiles fine, but my device doesn't even recognise it (green light doesn't light up anymore).

ptpt52, 2016-10-16:
post some dmesg info please

jshthornton, 2016-10-16:
```
[   19.353018] usb 2-5: new high-speed USB device number 2 using ehci-pci
[   19.485919] usb 2-5: New USB device found, idVendor=0bda, idProduct=8179
[   19.485922] usb 2-5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   19.485924] usb 2-5: Product: 802.11n NIC
[   19.485925] usb 2-5: Manufacturer: Realtek
[   19.485926] usb 2-5: SerialNumber: 243C201EBAC4
[   20.514880] r8188eu: module is from the staging directory, the quality is unknown, you have been warned.
[   20.515687] cfg80211: World regulatory domain updated:
[   20.515690] cfg80211:  DFS Master region: unset
[   20.515691] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[   20.515693] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[   20.515694] cfg80211:   (2457000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[   20.515696] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (N/A, 2000 mBm), (N/A)
[   20.515697] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[   20.515699] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[   20.515700] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
[   20.515701] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
[   20.515702] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)
[   20.515954] Chip Version Info: CHIP_8188E_Normal_Chip_TSMC_UNKNOWN_CUT(10)_1T1R_RomVer(0)
[   20.544979] usbcore: registered new interface driver r8188eu
[   20.546068] r8188eu 2-5:1.0 wlx243c201ebac4: renamed from wlan0
[   20.565791] IPv6: ADDRCONF(NETDEV_UP): wlx243c201ebac4: link is not ready
[   21.361622] MAC Address = 24:3c:20:1e:ba:c4
[   21.365278] IPv6: ADDRCONF(NETDEV_UP): wlx243c201ebac4: link is not ready
[   21.372861] R8188EU: ERROR indicate disassoc
[   21.376794] IPv6: ADDRCONF(NETDEV_UP): wlx243c201ebac4: link is not ready
[   21.418891] IPv6: ADDRCONF(NETDEV_UP): wlx243c201ebac4: link is not ready
[   24.265362] R8188EU: ERROR indicate disassoc
[   27.233522] R8188EU: ERROR indicate disassoc
[   30.209735] R8188EU: ERROR indicate disassoc
[   33.177961] R8188EU: ERROR indicate disassoc
[   48.783785] R8188EU: ERROR assoc success
[   48.783848] IPv6: ADDRCONF(NETDEV_CHANGE): wlx243c201ebac4: link becomes ready
```

ptpt52, 2016-10-16:
sorry ,but your device cannot be supported with this code
8188e chip not supported here.

jshthornton, 2016-10-16:
@ptpt52 I currently have two wireless adaptors in my device, one which is working (which I think is the 8188e) and the one which isn't working which is from a TP-LINK AC1200.

ptpt52, 2016-10-16:
remove your device and re-plugin again. then show me the  dmesg info

jshthornton, 2016-10-16:
```
[ 1458.201782] usb 8-1: USB disconnect, device number 2
[ 1461.537862] usb 8-1: new high-speed USB device number 3 using xhci_hcd
[ 1461.722224] usb 8-1: New USB device found, idVendor=2357, idProduct=0103
[ 1461.722230] usb 8-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1461.722233] usb 8-1: Product: 802.11n NIC
[ 1461.722236] usb 8-1: Manufacturer: Realtek
[ 1461.722238] usb 8-1: SerialNumber: 123456
```

ptpt52, 2016-10-16:
try this patch, then please tell me the result.

diff --git a/os_dep/linux/usb_intf.c b/os_dep/linux/usb_intf.c
index e14fb72..b49367c 100644
--- a/os_dep/linux/usb_intf.c
+++ b/os_dep/linux/usb_intf.c
@@ -312,7 +312,7 @@ static struct usb_device_id rtw_usb_id_tbl[] = {
        {USB_DEVICE(0x7392, 0xA813),.driver_info = RTL8821}, /* Edimax - EW-7811UAC */
        {USB_DEVICE(0x0BDA, 0xA811),.driver_info = RTL8821}, /* OUTLINK - Edimax */
        {USB_DEVICE(0x2357, 0x0101),.driver_info = RTL8821}, /* TP-Link - Archer T4U */
-       {USB_DEVICE(0x2357, 0x0103),.driver_info = RTL8812}, /* TP-Link - T4UH */
+       {USB_DEVICE(0x2357, 0x0103),.driver_info = RTL8821}, /* TP-Link - T4UH */
        {USB_DEVICE(0x04BB, 0x0953),.driver_info = RTL8821}, /* I-O DATA - Edimax */
        {USB_DEVICE(0x2001, 0x3314),.driver_info = RTL8821}, /* D-Link - Cameo */
        {USB_DEVICE(0x2001, 0x3318),.driver_info = RTL8821}, /* D-Link - Cameo */
@@ -322,7 +322,7 @@ static struct usb_device_id rtw_usb_id_tbl[] = {
        {USB_DEVICE(0x0411, 0x0242),.driver_info = RTL8821}, /* BUFFALO - Edimax */
        {USB_DEVICE(0x0411, 0x025D),.driver_info = RTL8821}, /* BUFFALO - WI-U3-866D */
        {USB_DEVICE(0x056E, 0x4007),.driver_info = RTL8821}, /* ELECOM - WDC-433DU2H */
-       {USB_DEVICE(0x148F, 0x9097),.driver_info = RTL8812}, /* Amped Wireless ACA1 */
+       {USB_DEVICE(0x148F, 0x9097),.driver_info = RTL8821}, /* Amped Wireless ACA1 */
 #endif
 
 #ifdef CONFIG_RTL8192E

ptpt52, 2016-10-16:
pull the latest HEAD and try again please

jshthornton, 2016-10-16:
@ptpt52 Was able to get the device to light up, connect, and run a speed test. Thanks very much, I owe you a beer!

sportbilly789, 2016-11-06:
Can not make work I am getting the message below
make KERNELRELEASE=4.4.0-43-generic KVER=4.4.0-43-generic.........(bad exit status: 2)
ERROR (dkms apport): binary package for rtl8812au: 4.3.14 not found
Error! Bad return status for module build on kernel: 4.4.0-43-generic (x86_64)
Consult /var/lib/dkms/rtl8812au/4.3.14/build/make.log for more information.
Makefile.dkms:18: recipe for target 'build' failed
make: *** [build] Error 10

ptpt52, 2016-11-06:
post your log:  /var/lib/dkms/rtl8812au/4.3.14/build/make.log

sportbilly789, 2016-11-06:
DKMS make.log for rtl8812au-4.3.14 for kernel 4.4.0-43-generic (x86_64)
Sun  6 Nov 15:41:36 GMT 2016
make[1]: Entering directory '/var/lib/dkms/rtl8812au/4.3.14/build'
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.4.0-43-generic/build M=/var/lib/dkms/rtl8812au/4.3.14/build  modules
make[2]: Entering directory '/usr/src/linux-headers-4.4.0-43-generic'
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_br_ext.o
In file included from include/net/ip.h:31:0,
                 from /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_br_ext.c:24:
include/net/route.h: In function ‘ip_route_output’:
include/net/route.h:141:3: error: missing initializer for field ‘flowic_iif’ of ‘struct flowi_common’ [-Werror=missing-field-initializers]
   .flowi4_tos = tos,
   ^
In file included from include/linux/skbuff.h:40:0,
                 from include/linux/if_arp.h:26,
                 from /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_br_ext.c:23:
include/net/flow.h:29:6: note: ‘flowic_iif’ declared here
  int flowic_iif;
      ^
In file included from include/net/ip.h:31:0,
                 from /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_br_ext.c:24:
include/net/route.h:142:3: error: missing initializer for field ‘flowic_scope’ of ‘struct flowi_common’ [-Werror=missing-field-initializers]
   .daddr = daddr,
   ^
In file included from include/linux/skbuff.h:40:0,
                 from include/linux/if_arp.h:26,
                 from /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_br_ext.c:23:
include/net/flow.h:32:7: note: ‘flowic_scope’ declared here
  __u8 flowic_scope;
       ^
cc1: all warnings being treated as errors
scripts/Makefile.build:258: recipe for target '/var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_br_ext.o' failed
make[3]: *** [/var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_br_ext.o] Error 1
Makefile:1403: recipe for target '_module_/var/lib/dkms/rtl8812au/4.3.14/build' failed
make[2]: *** [_module_/var/lib/dkms/rtl8812au/4.3.14/build] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-4.4.0-43-generic'
Makefile:1570: recipe for target 'modules' failed
make[1]: *** [modules] Error 2
make[1]: Leaving directory '/var/lib/dkms/rtl8812au/4.3.14/build'

ptpt52, 2016-11-06:
try remove the line core/rtw_br_ext.c:24
or comment out the line EXTRA_CFLAGS += -Werror in Makefile

sportbilly789, 2016-11-06:
I removed the line and It compiled and installed but cannot connect to the Internet. 
[  903.452796] usb 1-5: New USB device found, idVendor=2357, idProduct=0103
[  903.452800] usb 1-5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  903.452803] usb 1-5: Product: 802.11n NIC
[  903.452806] usb 1-5: Manufacturer: Realtek
[  903.452808] usb 1-5: SerialNumber: 123456
[  903.717357] RTL871X: rtw_ndev_init(wlan0)
[  904.745932] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  905.314187] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  905.366401] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  937.090445] RTL871X: rtw_set_802_11_connect(wlan0)  fw_state=0x00000008
[  937.262728] RTL871X: start auth
[  937.564588] RTL871X: auth success, start assoc
[  937.566871] RTL871X: rtw_cfg80211_indicate_connect(wlan0) BSS not found !!
[  937.566883] RTL871X: assoc success
[  937.566970] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  937.693701] RTL871X: send eapol packet
[  937.705038] RTL871X: send eapol packet
[  937.705127] RTL871X: set pairwise key camid:4, addr:a8:d3:f7:66:b5:64, kid:0, type:AES
[  937.708442] RTL871X: set group key camid:5, addr:a8:d3:f7:66:b5:64, kid:1, type:AES

jk89, 2018-01-22:
> binary package for rtl8812au: 4.3.14 not found

I get the same error