abperiasamy/rtl8812AU_8821AU_linux #159: tp-link archer t4u stopped working

xpue, 2016-10-09:
Driver works if bult from commit e4566b49ae3df63e37536b778d0118420676ca2e
but does not work if built from latest.

ptpt52, 2016-10-09:
hello, the latest commit maybe bug
Please try this new version out
https://github.com/ptpt52/rtl8812au

Grief, 2016-10-16:
@ptpt52 
```
Oct 16 03:19:12 rocket wpa_supplicant[2939]: wlan0: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
Oct 16 03:19:30 rocket NetworkManager[1486]: <info>  [1476577170.6056] manager: WiFi hardware radio set disabled
Oct 16 03:19:30 rocket wpa_supplicant[2939]: rfkill: WLAN soft blocked
Oct 16 03:19:30 rocket NetworkManager[1486]: <info>  [1476577170.6056] device (wlan0): state change: disconnected -> unavailable (reason 'none') [30 20 0]
Oct 16 03:19:30 rocket NetworkManager[1486]: <info>  [1476577170.6060] audit: op="radio-control" arg="wireless-enabled:0" pid=3433 uid=47433 result="success"
Oct 16 03:19:30 rocket NetworkManager[1486]: <info>  [1476577170.6064] manager: WiFi now disabled by radio killswitch
Oct 16 03:19:30 rocket kernel: ------------[ cut here ]------------
Oct 16 03:19:30 rocket kernel: WARNING: CPU: 2 PID: 1486 at net/wireless/core.c:1089 cfg80211_netdev_notifier_call+0x5f9/0x640 [cfg80211]
Oct 16 03:19:30 rocket kernel: Modules linked in: fuse rfcomm pci_stub vboxpci(OE) vboxnetadp(OE) vboxnetflt(OE) vboxdrv(OE) bnep 8812au(OE) cfg80211 binfmt_misc joydev hid_generic btusb btrtl btbcm btintel bluetooth rfkill snd_hda_codec_hdmi intel_rapl x86_pkg_temp_thermal intel_powerclamp coretemp mxm_wmi kvm_intel kvm irqbypass crct10dif_pclmul crc32_pclmul snd_soc_ssm4567 ghash_clmulni_intel efi_pstore aesni_intel aes_x86_64 lrw gf128mul glue_helper ablk_helper cryptd intel_cstate intel_rapl_perf snd_soc_rt5640 snd_soc_rl6231 serio_raw snd_soc_core snd_compress efivars ac97_bus snd_pcm_dmaengine snd_hda_codec_ca0132 nvidia_uvm(POE) snd_seq_midi snd_seq_midi_event snd_hda_intel snd_rawmidi snd_hda_codec snd_hda_core snd_hwdep snd_pcm snd_seq snd_seq_device snd_timer lpc_ich snd snd_soc_sst_acpi soundcore snd_soc_sst_match
Oct 16 03:19:30 rocket kernel:  battery elan_i2c wmi dw_dmac dw_dmac_core i2c_designware_platform i2c_designware_core acpi_pad intel_smartconnect spi_pxa2xx_platform mei_me evdev mei xpad shpchp ff_memless usbhid sg parport_pc ppdev lp parport efivarfs ip_tables x_tables autofs4 ext4 crc16 jbd2 fscrypto mbcache vmw_pvscsi vmxnet3 uvesafb uas usb_storage nvidia_drm(POE) nvidia_modeset(POE) crc32c_intel psmouse drm nvidia(POE) alx ahci mdio libahci e1000e ptp pps_core sdhci_acpi video sdhci fjes mmc_core i2c_hid hid gpio_lynxpoint button
Oct 16 03:19:30 rocket kernel: CPU: 2 PID: 1486 Comm: NetworkManager Tainted: P        W  OE   4.8.1-custom+ #4
Oct 16 03:19:30 rocket kernel: Hardware name: To Be Filled By O.E.M. To Be Filled By O.E.M./Z97 Professional, BIOS P2.50 05/06/2016
Oct 16 03:19:30 rocket kernel:  0000000000000286 00000000615d1215 ffffffff9af5a714 0000000000000000
Oct 16 03:19:30 rocket kernel:  0000000000000000 ffffffff9ac7f6ee ffff945323ef3400 ffff945320d05aa0
Oct 16 03:19:30 rocket kernel:  ffff94531fdf3000 0000000000000002 ffff945320d05800 0000563695fbf138
Oct 16 03:19:30 rocket kernel: Call Trace:
Oct 16 03:19:30 rocket kernel:  [<ffffffff9af5a714>] ? dump_stack+0x5c/0x78
Oct 16 03:19:30 rocket kernel:  [<ffffffff9ac7f6ee>] ? __warn+0xbe/0xe0
Oct 16 03:19:30 rocket kernel:  [<ffffffffc13aca29>] ? cfg80211_netdev_notifier_call+0x5f9/0x640 [cfg80211]
Oct 16 03:19:30 rocket kernel:  [<ffffffff9b223318>] ? __dev_remove_pack+0x58/0xc0
Oct 16 03:19:30 rocket kernel:  [<ffffffff9b318aba>] ? __unregister_prot_hook+0xca/0xd0
Oct 16 03:19:30 rocket kernel:  [<ffffffff9ac9fdb5>] ? notifier_call_chain+0x45/0x70
Oct 16 03:19:30 rocket kernel:  [<ffffffff9b22414d>] ? dev_close_many+0xdd/0x120
Oct 16 03:19:30 rocket kernel:  [<ffffffff9b22573e>] ? dev_close.part.83+0x3e/0x60
Oct 16 03:19:30 rocket kernel:  [<ffffffff9b225773>] ? dev_close+0x13/0x20
Oct 16 03:19:30 rocket kernel:  [<ffffffffc13abdef>] ? cfg80211_shutdown_all_interfaces+0x3f/0xa0 [cfg80211]
Oct 16 03:19:30 rocket kernel:  [<ffffffffc13abea2>] ? cfg80211_rfkill_set_block+0x22/0x30 [cfg80211]
Oct 16 03:19:30 rocket kernel:  [<ffffffffc104ac43>] ? rfkill_set_block+0x83/0x130 [rfkill]
Oct 16 03:19:30 rocket kernel:  [<ffffffffc104aed4>] ? rfkill_fop_write+0x114/0x1c0 [rfkill]
Oct 16 03:19:30 rocket kernel:  [<ffffffff9ae13eb3>] ? __vfs_write+0x33/0x160
Oct 16 03:19:30 rocket kernel:  [<ffffffff9ae2ca18>] ? dput+0x38/0x250
Oct 16 03:19:30 rocket kernel:  [<ffffffff9ae3391e>] ? __fd_install+0x2e/0xe0
Oct 16 03:19:30 rocket kernel:  [<ffffffff9aec1828>] ? security_file_permission+0x38/0xb0
Oct 16 03:19:30 rocket kernel:  [<ffffffff9ae145b0>] ? vfs_write+0xb0/0x190
Oct 16 03:19:30 rocket kernel:  [<ffffffff9ae15982>] ? SyS_write+0x52/0xc0
Oct 16 03:19:30 rocket kernel:  [<ffffffff9b33f136>] ? entry_SYSCALL_64_fastpath+0x1e/0xa8
Oct 16 03:19:30 rocket kernel: ---[ end trace 31063cdb5188c5c7 ]---
Oct 16 03:19:30 rocket systemd[1]: Starting Load/Save RF Kill Switch Status...
Oct 16 03:19:30 rocket systemd[1]: Started Load/Save RF Kill Switch Status.
Oct 16 03:19:30 rocket wpa_supplicant[2939]: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Oct 16 03:19:32 rocket NetworkManager[1486]: <info>  [1476577172.3803] manager: WiFi hardware radio set enabled
Oct 16 03:19:32 rocket NetworkManager[1486]: <info>  [1476577172.3805] audit: op="radio-control" arg="wireless-enabled:1" pid=3433 uid=47433 result="success"
Oct 16 03:19:32 rocket NetworkManager[1486]: <info>  [1476577172.3811] manager: WiFi now enabled by radio killswitch
Oct 16 03:19:32 rocket kernel: IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[... cut ...]
Oct 16 03:20:45 rocket sudo[4570]: pam_unix(sudo:session): session opened for user root by grief(uid=0)
Oct 16 03:20:45 rocket sudo[4570]: pam_unix(sudo:session): session closed for user root
Oct 16 03:20:45 rocket kernel: IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Oct 16 03:20:45 rocket NetworkManager[1486]: <info>  [1476577245.0540] device (wlan0): supplicant interface state: disabled -> inactive
Oct 16 03:20:45 rocket wpa_supplicant[2939]: wlan0: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
```

ptpt52, 2016-10-16:
@xpue 
try the latest HEAD

Grief, 2016-10-17:
@xpue Hello! I am on Ubuntu 16.10 with kernel 4.8.1 and I made Archer T4U to work. If you feel adventurous, feel free to try: https://github.com/Grief/archer-t4u

twhiteman, 2016-10-17:
@xpue I too had the same problem, and had to revert to commit e4566b4 (latest HEAD does not work - well, the driver loads but no wireless networks are found).

ptpt52, 2016-11-04:
try the latest HEAD of master, maybe it's fixed

archshift, 2016-11-12:
@ptpt52 I'm on latest master, I still can't find any connections using my T4U adapter.

x-direct-group, 2016-11-17:
For me, latest master 8db817 is working with Debian stretch, linux 4.8.0.1, I had the exact behaviour described by @Grief before - Bug seems closable to me.
Booting into an old kernel, fetching the latest master from here, going back into latest kernel and installing the driver via Makefile.dkms (one-liner in README.md) was all I had to do.

closed 2016-11-17 by ptpt52