abperiasamy/rtl8812AU_8821AU_linux #163: very new to linux, does this work on fedora

lizardkahn, 2016-10-11:
i just installed Fedora last week, and spent today connected via Ethernet cable to my router, i can't seem to get it to work. when i do make install i get

 cannot stat ‘rtl8812au.ko’: No such file or directory

from what i've seen the tp-link archer only works with Ubuntu and Raspberry Pi

andrerodr, 2016-10-12:
I dont have a tp-link unit but I have a comfast 1200ac wirh same rtl8812 and its currently working on opensuse and should work prety much on every distro!

andrerodr, 2016-10-12:
Did you actualy compiled the driver?

hyph3n, 2016-10-22:
Hi @lizardkahn,

Fedora 24 user here, and I finally got a TP-Link Archer T4U working today. Here's what worked for me.

> git clone https://github.com/abperiasamy/rtl8812AU_8821AU_linux.git

> cd rtl8812AU_8821AU_linux/

> sudo make -f Makefile.dkms install

Plug in your TP-Link Archer T4U, and you should get a list of wireless networks in NetworkManager. Choose one to connect to, and that should do it. Let me know if you need help.

closed 2016-11-04 by ptpt52