abperiasamy/rtl8812AU_8821AU_linux #166: compilation for dkms exits with error 2

wierdlmate, 2016-10-13:
Hello,

On my Fedora 24 system, I tried to compile the latest version, but the compilation fails.  Please advise.  I had no problem with installing the driver without dkms.  

# make -f Makefile.dkms 
make clean
make[1]: Entering directory '/home/apu/rtl8812AU_8821AU_linux-master'
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.7.6-200.fc24.x86_64/build M=/home/apu/rtl8812AU_8821AU_linux-master clean
make[2]: Entering directory '/usr/src/kernels/4.7.6-200.fc24.x86_64'
  CLEAN   /home/apu/rtl8812AU_8821AU_linux-master/.tmp_versions
  CLEAN   /home/apu/rtl8812AU_8821AU_linux-master/Module.symvers
make[2]: Leaving directory '/usr/src/kernels/4.7.6-200.fc24.x86_64'
make[1]: Leaving directory '/home/apu/rtl8812AU_8821AU_linux-master'
mkdir -p '/usr/src/rtl8812au-4.3.14'
cp -r dkms.conf Kconfig Makefile.dkms Makefile platform core hal include os_dep '/usr/src/rtl8812au-4.3.14'
cp Makefile '/usr/src/rtl8812au-4.3.14/Makefile'
sed 's/#MODULE_VERSION#/4.3.14/' dkms.conf > '/usr/src/rtl8812au-4.3.14/dkms.conf'
dkms add -m rtl8812au -v 4.3.14 2>/dev/null || true
dkms build -m rtl8812au -v 4.3.14

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j2 KERNELRELEASE=4.7.6-200.fc24.x86_64 KVER=4.7.6-200.fc24.x86_64...(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.7.6-200.fc24.x86_64 (x86_64)
Consult /var/lib/dkms/rtl8812au/4.3.14/build/make.log for more information.
Makefile.dkms:18: recipe for target 'build' failed
make: *** [build] Error 10

wierdlmate, 2016-10-14:
I remark that the driver (the one I compiled and installed without dkms) doesn't appear to be stable: my edimax ac1200 drops the connection within one hour. I'll be happy to debug this, just let me know what info you need.

I remark that this edimax device was working flawlessly on a mint 17 system for over a year---maybe even two years.  On the Mint box, I used the driver that was given at Edimax's site.  But that one doesn't compile on this Fedora 24 syetm.

ptpt52, 2016-10-14:
/var/lib/dkms/rtl8812au/4.3.14/build/make.log ?

anb0s, 2016-10-14:
Friend of mine told me the solution :)

In file dkms.conf line 3 add quotes to make,

PACKAGE_NAME="rtl8812au"
PACKAGE_VERSION="#MODULE_VERSION#"
MAKE[0]="’make’ KVER=$kernelver"
CLEAN="make clean"
BUILT_MODULE_NAME[0]="rtl8812au"
DEST_MODULE_LOCATION[0]="/kernel/drivers/net"
AUTOINSTALL="yes"

see https://linux.die.net/man/8/dkms

> ... KERNELRELEASE will be automatically appended to MAKE[#]. If you want to suppress this behavior, you can quote the make command: 'make'.

wierdlmate, 2016-10-14:
Yes, quoting make did the job. Thanks for the help!!

closed 2016-10-14 by wierdlmate