abperiasamy/rtl8812AU_8821AU_linux #169: Cannot connect

paneq, 2016-10-16:
```
[   77.384149] BUG: unable to handle kernel paging request at ffffc90001d74099
[   77.384178] IP: [<ffffffffc094d406>] join_cmd_hdl+0x157/0x322 [rtl8812au]
[   77.384221] PGD 40d8a1067 PUD 40d8a2067 PMD 4065b1067 PTE 0
[   77.384242] Oops: 0000 [#1] SMP 
[   77.384254] Modules linked in: rtl8812au(OE) xfrm_user xfrm_algo xt_addrtype xt_conntrack br_netfilter aufs ctr ccm xt_CHECKSUM iptable_mangle ipt_MASQUERADE nf_nat_masquerade_ipv4 iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack xt_tcpudp bridge stp llc iptable_filter ip_tables x_tables rfcomm bnep binfmt_misc snd_hda_codec_conexant snd_hda_codec_generic snd_hda_intel snd_hda_codec snd_hda_core snd_hwdep snd_pcm intel_rapl x86_pkg_temp_thermal intel_powerclamp snd_seq_midi coretemp snd_seq_midi_event kvm_intel uvcvideo arc4 videobuf2_vmalloc kvm videobuf2_memops rtl8192ce snd_rawmidi videobuf2_v4l2 btusb videobuf2_core v4l2_common btrtl btbcm videodev btintel irqbypass rtl_pci media rtl8192c_common bluetooth joydev rtlwifi mac80211 input_leds serio_raw cfg80211 lpc_ich snd_seq thinkpad_acpi nvram snd_seq_device snd_timer shpchp snd mei_me mac_hid mei soundcore parport_pc ppdev nls_iso8859_1 lp parport drbg ansi_cprng dm_crypt i915 crct10dif_pclmul crc32_pclmul i2c_algo_bit drm_kms_helper syscopyarea sysfillrect aesni_intel sysimgblt fb_sys_fops aes_x86_64 lrw drm gf128mul glue_helper ablk_helper cryptd psmouse e1000e ahci sdhci_pci libahci sdhci ptp pps_core wmi fjes video
[   77.384665] CPU: 3 PID: 3282 Comm: RTW_CMD_THREAD Tainted: G           OE   4.4.0-42-generic #62~14.04.1-Ubuntu
[   77.384697] Hardware name: LENOVO 4180HU1/4180HU1, BIOS 83ET76WW (1.46 ) 07/05/2013
[   77.384721] task: ffff8800d7246040 ti: ffff8803d1494000 task.ti: ffff8803d1494000
[   77.384744] RIP: 0010:[<ffffffffc094d406>]  [<ffffffffc094d406>] join_cmd_hdl+0x157/0x322 [rtl8812au]
[   77.384784] RSP: 0018:ffff8803d1497e30  EFLAGS: 00010297
[   77.384801] RAX: ffffc90001d7408d RBX: ffffc90001d6e000 RCX: 0000000000000000
[   77.384823] RDX: 0000000000000012 RSI: ffffc90001d6f2f8 RDI: ffffc90001d6f20a
[   77.384845] RBP: ffff8803d1497e60 R08: ffff1f106e1a2d00 R09: 0000000000000000
[   77.384867] R10: 0000ffff1f106e1a R11: 0000000000000000 R12: 0000000000004dcb
[   77.384889] R13: ffffc90001d74099 R14: ffffc90001d6f252 R15: 0000000000000001
[   77.384911] FS:  0000000000000000(0000) GS:ffff88041e2c0000(0000) knlGS:0000000000000000
[   77.384936] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[   77.384954] CR2: ffffc90001d74099 CR3: 0000000002e0c000 CR4: 00000000000406e0
[   77.384976] Stack:
[   77.384983]  0000000000000297 ffff8804052192a0 ffffc90001d6e000 00000000ffff2697
[   77.385008]  ffffc90001d6fc18 ffff8800d7246040 ffff8803d1497ec8 ffffffffc092b9fc
[   77.385032]  ffff8800d7246040 ffffc90001d6fc00 ffffc90001d6fc64 ffff8803f6bc8a00
[   77.385057] Call Trace:
[   77.385074]  [<ffffffffc092b9fc>] rtw_cmd_thread+0x2a4/0x3c0 [rtl8812au]
[   77.385102]  [<ffffffffc092b758>] ? rtw_stop_cmd_thread+0x53/0x53 [rtl8812au]
[   77.385126]  [<ffffffff8109b959>] kthread+0xc9/0xe0
[   77.385142]  [<ffffffff8109b890>] ? kthread_park+0x60/0x60
[   77.385162]  [<ffffffff817fa88f>] ret_from_fork+0x3f/0x70
[   77.385179]  [<ffffffff8109b890>] ? kthread_park+0x60/0x60
[   77.385196] Code: e8 a1 0e 00 00 66 89 83 ea 11 00 00 41 bf 00 00 00 00 41 bc 0c 00 00 00 4c 8d b3 52 12 00 00 44 89 e0 49 8d 44 06 70 4c 8d 68 0c <0f> b6 40 0c 3c bf 0f 84 e5 00 00 00 3c bf 77 11 3c 2d 74 51 3c 
[   77.385288] RIP  [<ffffffffc094d406>] join_cmd_hdl+0x157/0x322 [rtl8812au]
[   77.385320]  RSP <ffff8803d1497e30>
[   77.385331] CR2: ffffc90001d74099
[   77.391130] ---[ end trace 3066b9560727b4e5 ]---
```

I am not sure if this is related to #38 and #144 as this is "TP Link Archer T4U AC1200" which I try to make it working with 5Ghz network.

Grief, 2016-10-17:
@paneq Hello! I am on Ubuntu 16.10 with kernel 4.8.1 and I made Archer T4U to work. If you feel adventurous, feel free to try: https://github.com/Grief/archer-t4u

ptpt52, 2016-10-17:
@paneq hi,
help me checkout that weather the latest HEAD work for your t4u device ?

paneq, 2016-10-17:
@Grief looks nice, but i am on 14.04 and 4.4 so I think it won't work, right?

ptpt52, 2016-10-17:
@Grief have you test the latest HEAD code? does it work?

paneq, 2016-10-17:
@ptpt52 I compiled and tested on `de9c2b93c6302cdb0fe83d8d998160ecf4a65f41` which is current HEAD I believe. And the error that occurred is described. Can I do something more to help?

ptpt52, 2016-10-17:
@paneq what error?

ptpt52, 2016-10-17:
@paneq seems hard to fix...

Grief, 2016-10-17:
@ptpt52 have to say that your driver is working for me now with your latest changes. But I'd continue to use my own :)

Grief, 2016-10-19:
@ptpt52 I just had exactly the same stacktrace on lenovo g550 with Archer t4u with your latest code. However, with my code (https://github.com/Grief/archer-t4u) it works smoothly. Feel free to grab changes from my fork into yorth.

closed 2016-11-04 by ptpt52