abperiasamy/rtl8812AU_8821AU_linux #174: compilation failure on kernel 4.4.0-46

random-thoughts, 2016-11-03:
I am trying to compile the code (make) but the compilation fails because of the following errors. Anybody has had the same problems?

`In file included from include/net/ip.h:31:0,
                 from /home/orestis/wifi_drivers/rtl8812AU_8821AU_linux/core/rtw_br_ext.c:24:
include/net/route.h: In function ‘ip_route_output’:
include/net/route.h:141:3: error: missing initializer for field ‘flowic_iif’ of ‘struct flowi_common’ [-Werror=missing-field-initializers]
   .flowi4_tos = tos,
   ^
In file included from include/linux/skbuff.h:40:0,
                 from include/linux/if_arp.h:26,
                 from /home/orestis/wifi_drivers/rtl8812AU_8821AU_linux/core/rtw_br_ext.c:23:
include/net/flow.h:29:6: note: ‘flowic_iif’ declared here
  int flowic_iif;
      ^
In file included from include/net/ip.h:31:0,
                 from /home/orestis/wifi_drivers/rtl8812AU_8821AU_linux/core/rtw_br_ext.c:24:
include/net/route.h:142:3: error: missing initializer for field ‘flowic_scope’ of ‘struct flowi_common’ [-Werror=missing-field-initializers]
   .daddr = daddr,
   ^
In file included from include/linux/skbuff.h:40:0,
                 from include/linux/if_arp.h:26,
                 from /home/orestis/wifi_drivers/rtl8812AU_8821AU_linux/core/rtw_br_ext.c:23:
include/net/flow.h:32:7: note: ‘flowic_scope’ declared here
  __u8 flowic_scope; 
`

NicholasTurner, 2016-11-03:
Same issue here. Running Ubuntu 14.04 with kernel 3.13.

ptpt52, 2016-11-04:
maybe you have the wrong kernel headers

try sudo apt-get install linux-headers-generic or something like that.

hackeprot, 2016-11-11:
Got the same problem on Debian Jessie

```
# uname -r
3.16.0-4-amd64

dpkg -l | grep linux-headers
ii  linux-headers-3.16.0-4-amd64          3.16.36-1+deb8u2                     amd64        Header files for Linux 3.16.0-4-amd64
ii  linux-headers-3.16.0-4-common         3.16.36-1+deb8u2                     amd64        Common header files for Linux 3.16.0-4
ii  linux-headers-amd64                   3.16+63                              amd64        Header files for Linux amd64 configuration (meta-package)

```

mildmojo, 2016-11-13:
I hit the same error on kernel 3.19.0-32 in Linux Mint 17 (Ubuntu 14.04 derivative). I only have this one kernel and headers version installed, and it matches `uname -a`.

I worked around the compilation failure by commenting out `CFLAGS_EXTRA=-Werror` in `Makefile`. (When enabled, this option turns compiler warnings into errors.)

I haven't verified if the module works with my hardware.

closed 2016-12-07 by random-thoughts