abperiasamy/rtl8812AU_8821AU_linux #177: rtl8812AU_8821AU_linux and Edimax AC600 USB (7392:a812)

notorand-it, 2016-11-11:
This is not a real issue (not yet at least).
The "Edimax AC600 USB" adapter [is said](https://wiki.ubuntuusers.de/WLAN/Karten/Edimax/) to be managed as "rtl8812au".
Is this driver the right one?
Or should I go with an older [rtl8812au only driver](https://github.com/gnab/rtl8812au)?

ptpt52, 2016-11-17:
remove your old driver first

strainu, 2016-11-29:
@notorand-it: this is the right driver, I can confirm the adapter worked perfectly for me in the last year with this driver.

rajendrant, 2016-12-19:
I could confirm this it the right driver.
It worked for my new Edimax AC600 USB.

Goga777, 2017-01-22:
don't you have any issue with authentification ? for example error as bad password during handshake with router ?