abperiasamy/rtl8812AU_8821AU_linux #178: Error compiling

bhaap, 2016-11-11:
http://pastebin.com/5mwuw9mR

Im trying to compile this on my NanoPi 2 Fire with a S5P4418 SoC arm-v7-a Cortex-A9 chip.

Atm i have defined CONFIG_PLATFORM_ARM_RPI = y
and also commented out EXTRA_CFLAGS += -Werror so that it would not stop on all the 
"warning: comparison between signed and unsigned integer expressions [-Wsign-compare]"

regards,
Johan L

bhaap, 2016-11-11:
Got past this issue by doing changes to the ieee80211.h file.
Replace "extern __inline" with "static __inline" for declatations of is_broadcast_mac_addr, is_multicast_mac_addr, is_zero_mac_addr.

But now i have a new issue. I cannot modprobe the driver. 

$ sudo modprobe -avf rtl8812au
insmod /lib/modules/3.4.39-s5p4418/updates/compat/compat.ko 
modprobe: ERROR: could not insert 'rtl8812au': Exec format error

uname -a:
Linux FriendlyARM 3.4.39-s5p4418 #2 SMP PREEMPT Thu Aug 11 13:53:54 HKT 2016 armv7l armv7l armv7l GNU/Linux

notorand-it, 2016-11-22:
In C language "extern" has a completely different semantic than "static". They are actually opposite each other.
Your "solution" is just a message stopper, not a solution.