abperiasamy/rtl8812AU_8821AU_linux #180: modprobe -a rtl8812au does not work for raspberry pi 3

random-thoughts, 2016-11-27:
modprobe -a rtl8812au does not work for raspberry pi 3. 

`make` and `make install` work fine.

 `make` works after I download the headers for 3.10-3-rpi and link the `/lib/modules/3.10-3-rpi/build` folder to `/lib/modules/4.4.11-v7+/build`
Kernel version: 4.4.11-v7+

Error:  ERROR: could not insert '8812au': Exec format error
After executing `file /lib/modules/4.4.11-v7+/kernel/drivers/net/wireless/rtl8812au.ko ` I get the message : 
`rtl8812au.ko: ELF 32-bit LSB relocatable, ARM, EABI5 version 1 (SYSV), BuildID[sha1]=c36196dc3674c56d0ce42cda9f21dba0a8f786e6, not stripped`

Any ideas?

sthorne, 2017-01-28:
Looks like you posted this a while ago, but I just picked up an Archer T4UH today and needed this driver so I got it working doing the following.  Based on your error it looks you compiled against the wrong kernel.

If you don't have rpi-update you can add it with `sudo apt install rpi-update` and rpi-source can be found here [https://github.com/notro/rpi-source/wiki](https://github.com/notro/rpi-source/wiki).  It will save you a ton of time when attempting to get the headers for an install.  Then just follow the README instructions for the driver on this site.

`$ sudo rpi-update`
`$ sudo rpi-source`

orestisgorgas, 2017-05-23:
I don't have the usb dongle anymore,so I cannot test it.

closed 2017-05-23 by orestisgorgas