abperiasamy/rtl8812AU_8821AU_linux #182: Why is "CONFIG_CONCURRENT_MODE" commented out for "CONFIG_PLATFORM_I386_PC"?

chboehm, 2016-12-29:
Why is "CONFIG_CONCURRENT_MODE" commented out for "CONFIG_PLATFORM_I386_PC"?

It seems to work and it is required to use the device on 2.4Ghz and 5Ghz simultaneously.

ptpt52, 2017-01-10:
really?

chboehm, 2017-01-11:
The actual result is that I have two devices now but I still can not use 2.4Ghz and 5Ghz simultaneously.

fumantsu, 2017-01-11:
I have test this and I think I found somewhere a documentation about it. It creates 2 AP but the second is more like virtual and the bandwidth is been shared. So in case that you have a 2.4G n 150mbps, the maximum for each AP will be 75mbps. To use 2.4Ghz and 5Ghz simultaneously as AP you need your adapter to have two radios.

chboehm, 2017-01-12:
My EW-7811UAC was advertised as "dual band". So I assumed it would have two radios. Apparently this is not the case. Thank you!

closed 2017-01-12 by chboehm