abperiasamy/rtl8812AU_8821AU_linux #186: Interface wlan0 vanished from ifconfig output after trying to enter monitor mode

Schawen, 2017-01-10:
Interface wlan0 vanished from ifconfig output after trying to enter monitor mode, but 
iwconfig show:

wlan0     unassociated  Nickname:"<WIFI@REALTEK>"
          Mode:**Managed**  Frequency=2.412 GHz  Access Point: Not-Associated   

Installed the driver ( sudo make -f Makefile.dkms install ) to
Kali Linux for VMware (Kernel: 4.6.0-kali1-amd64 ) 
running on VMware® Workstation 12 Player (Version: 12.5.2 build-4638234)
WLAN USB-Device ALFA AWUS036ACH

root@kali:~# uname -r
4.6.0-kali1-amd64

**root@kali:~# ifconfig**
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        ...

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        ...

**wlan0**: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 00:c0:ca:8f:dc:b2  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

**root@kali:~# airmon-ng start wlan0**

Found 3 processes that could cause trouble.
If airodump-ng, aireplay-ng or airtun-ng stops working after
a short period of time, you may want to run 'airmon-ng check kill'

  PID Name
  576 NetworkManager
  767 wpa_supplicant
  768 dhclient

PHY	Interface	Driver		Chipset

phy0	wlan0		rtl8812au	Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac WLAN Adapter
**cat: /sys/class/ieee80211/phy0/device/net/wlan0mon/type: Datei oder Verzeichnis nicht gefunden**

Newly created monitor mode interface wlan0mon is *NOT* in monitor mode.
Removing non-monitor wlan0mon interface...

WARNING: unable to start monitor mode, please run "airmon-ng check kill"

**root@kali:~# ifconfig**
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        ...

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        ...

_**Second try, killed all previously mentioned services:**_

**root@kali:~# ps aux | grep Network**
root      1698  0.0  0.0  12724   964 pts/0    S+   21:19   0:00 grep Network
root@kali:~# ps aux | grep dhcli
root      1700  0.0  0.0  12724   980 pts/0    S+   21:20   0:00 grep dhcli
root@kali:~# ps aux | grep wpa
root      1702  0.0  0.0  12724   944 pts/0    S+   21:20   0:00 grep wpa

**root@kali:~# ifconfig**
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        ...

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        ...

wlan0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 00:c0:ca:8f:dc:b2  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

**root@kali:~# airmon-ng start wlan0**

PHY	Interface	Driver		Chipset

phy0	wlan0		rtl8812au	Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac WLAN Adapter
cat: /sys/class/ieee80211/phy0/device/net/wlan0mon/type: Datei oder Verzeichnis nicht gefunden

Newly created monitor mode interface wlan0mon is *NOT* in monitor mode.
Removing non-monitor wlan0mon interface...

WARNING: unable to start monitor mode, please run "airmon-ng check kill"
root@kali:~#

**root@kali:~# ifconfig**
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        ...

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        ...

root@kali:~# 

_**No interface wlan0 ..., but**_

**root@kali:~# iwconfig**
lo        no wireless extensions.

wlan0     unassociated  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

eth0      no wireless extensions.

michael5486, 2017-02-08:
Not sure if it helps, but I have been using this driver with the TP-Link Archer T4U and just got it to work with monitor mode. I am running Kali in VMWare on a MacBook Pro. With the NIC plugged in and connected to Kali, I ran the following commands:
`ifconfig`
`iw wlan0 set monitor none`
`iw wlan0 info`
`airmon-ng start wlan0`
`airmon-ng check kill`
`airodump-ng wlan0`

Airmon-ng outputted an error, but a good one. It said wlan0 was already in monitor mode, and I have been able to successfully dump packets.

michael5486, 2017-02-08:
You may also need this too:

`sudo apt-get install kali-linux-wireless`

https://www.kali.org/news/kali-linux-metapackages/

PLEASEFORGETME2, 2019-05-26:
I installed meta-packages, but I can't run
`iw wlan0 set monitor none`
command failed: no such device (-19)
I have **ALMOST** the same problem... 
A little thing is different;
`iwconfig`
wlan0 unassociated ESSID:"" nickname:<WIFI@REALTEK>

The guy who opened the issue didn't got ESSID:""
I'm beginner, I don't know what it means

`uname -r`
4.19.0-kali5-amd64

Please help me guys, I'm working on this shitty wifi card for a week and I really need it to work.

The card is a REALTEK 8188ETV, but also 802.11n I don't know exactly what is the good name.
I'm working on a virtual box, host is a windows 10
thanks, cheers