abperiasamy/rtl8812AU_8821AU_linux #187: Can't install driver.

VautoMF, 2017-01-14:
I updated my Ubuntu 16.10 today and TP-LINK T4UH stopped working. When I try to install this driver terminal shows:

make clean
make[1]: Entering directory '/home/vauto/driver2'
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.8.0-34-generic/build M=/home/vauto/driver2 clean
make[2]: Entering directory '/usr/src/linux-headers-4.8.0-34-generic'
make[2]: Leaving directory '/usr/src/linux-headers-4.8.0-34-generic'
make[1]: Leaving directory '/home/vauto/driver2'
mkdir -p '/usr/src/rtl8812au-4.3.14'
cp -r dkms.conf Kconfig Makefile.dkms Makefile platform core hal include os_dep '/usr/src/rtl8812au-4.3.14'
cp Makefile '/usr/src/rtl8812au-4.3.14/Makefile'
sed 's/#MODULE_VERSION#/4.3.14/' dkms.conf > '/usr/src/rtl8812au-4.3.14/dkms.conf'
dkms add -m rtl8812au -v 4.3.14 2>/dev/null || true
dkms build -m rtl8812au -v 4.3.14
Module rtl8812au/4.3.14 already built for kernel 4.8.0-34-generic/4
dkms install -m rtl8812au -v 4.3.14
Module rtl8812au/4.3.14 already installed on kernel 4.8.0-34-generic/x86_64

And I still can't use wifi! Please help!

kwokyinc, 2017-01-15:
Same issue here with Ubuntu 16.10 with kernel 4.8.0-34. Not only did it stop working. The system actually halted claiming a kernel panick.
 
Now I have to revert back to an older kernel 4.8.0-32.

VautoMF, 2017-01-15:
Same here! I need tô choose the older kernel on Grub to make it work...

ptpt52, 2017-07-07:
just remote it use dkms first

closed 2017-07-07 by ptpt52

vinyanalista, 2017-07-13:
Today I saw those same messages (`Module rtl8812au/4.3.14 already built` and `Module rtl8812au/4.3.14 already installed`) using openSUSE Leap 42.2 with kernel 4.4.74.

Just in case anyone needs, here is how to remove the module using dkms:

```
# dkms remove rtl8812au/4.3.14 --all
```

Source: https://wiki.archlinux.org/index.php/Dynamic_Kernel_Module_Support#Remove_modules