abperiasamy/rtl8812AU_8821AU_linux #188: SOLVED: Support for new TP-Link T4U USB 3 WiFi router

sarlacii, 2017-01-19:
Greetings, I'd like to contribute changes to enable support for the new TP-Link T4U USB 3 WiFi router stick. I bought one from a local supplier in SA ([TP LINK T4U AC1200 WLSS D-BND USB ADAPT](https://www.incredible.co.za/tp-link-t4u-ac1200-wlss-d-bnd-usb-adapt)).

To get it to work I had to modify the usb_intf.c file, around line 300 to add {USB_DEVICE(0x2357,0x010d),driver_info=RTL8812} under the similar line for product ID 0101.

If you'd like me to make the changes in a branch, please let me know.

Thank you.

Sarlacii (AKA Julian)

harshavardhana, 2017-01-19:
You can send a PR @sarlacii

arzaz, 2017-01-20:
Dear Salacii,

How to get it work on my set top box Forever HD using shipset, RALINK RT73, RALINK RT3070, RALINK RT2870 or RALINK RT5370.

Your assistance is highly appreciated

closed 2017-01-22 by harshavardhana