abperiasamy/rtl8812AU_8821AU_linux #190: Archer T4UH v.1.0 looses abitilty to connect to previously connected network

RuralYak, 2017-01-24:
Summary:

After successful connection to 5ghz network Archer T4UH works for approx 4-6 hours then reports disconnection and does not connect back to the same network until reboot. Please see attached kern.log
[kern.zip](https://github.com/abperiasamy/rtl8812AU_8821AU_linux/files/727295/kern.zip)
.

Steps to reproduce.
1. Connect to wifi network
2. Leave it connected for 4-6 hours.
3. You will receive disconnection notification
4. Manual retires to connect it back will be unsuccessful until reboot. 

Kernel: 4.9.0-11-linuxium
System: Intel atom z-8300 Wintel cx-w8 pro
Ubuntu 17.04
driver: 4.3.8

PS. Give me a shout if you need additional info.

brmc, 2017-03-18:
any update on this? I have the same problemm, though it is not dependent on time.  sometimes I can go for days without a problem, other times it will happen several times in an hour, and I have yet to find a way to reproduce the issue

kmaterka, 2017-04-13:
I had the same issue with Asus USB-AC56. Problem is resolved in version 5.1.5 of the driver which is available in this repository:
https://github.com/uminokoe/rtl8812AU/tree/driver-5.1.5
For TP-Link Archer support you may try one of these forks:
https://github.com/mk-fg/rtl8812au
https://github.com/ojnickel/rtl8812au

More details for developers:

I compiled drivers with debug logs and tried to pinpoint the root cause of the issue. It doesn't matter if there is high traffic or connection is idle. 
Here is (shortened) output when hang happened:

kwi 10 22:38:28.985346 Kejk-PC kernel: RTL871X: chk_ap_is_alive(wlx704d7b153e6c) rx:(m:0, c:0, d:0), beacon:0, probersp_to_self:0, retry:0
kwi 10 22:38:28.985356 Kejk-PC kernel: RTL871X: chk_ap_is_alive(wlx704d7b153e6c) tx_pkts:18690, link_count:2
kwi 10 22:38:29.497333 Kejk-PC kernel: RTL871X: rtw_sctx_wait timeout: rtw_ack_tx_wait
kwi 10 22:38:30.009338 Kejk-PC kernel: RTL871X: rtw_sctx_wait timeout: rtw_ack_tx_wait
kwi 10 22:38:30.521319 Kejk-PC kernel: RTL871X: rtw_sctx_wait timeout: rtw_ack_tx_wait
kwi 10 22:38:30.521346 Kejk-PC kernel: RTL871X: issue_probereq_ex(wlx704d7b153e6c) to ac:22:0b:1f:0c:6c, ch:36, 3/3 in 1536 ms
kwi 10 22:38:31.033346 Kejk-PC kernel: RTL871X: rtw_sctx_wait timeout: rtw_ack_tx_wait
kwi 10 22:38:31.033377 Kejk-PC kernel: RTL871X: rtl8812au_xmitframe_complete #1, connot alloc xmitbuf!!!! 
kwi 10 22:38:31.033388 Kejk-PC kernel: RTL871X: rtl8812au_xmitframe_complete #1, connot alloc xmitbuf!!!! 
kwi 10 22:38:31.033395 Kejk-PC kernel: RTL871X: rtl8812au_xmitframe_complete #1, connot alloc xmitbuf!!!! 
kwi 10 22:38:31.545346 Kejk-PC kernel: RTL871X: rtw_sctx_wait timeout: rtw_ack_tx_wait
kwi 10 22:38:32.057334 Kejk-PC kernel: RTL871X: rtw_sctx_wait timeout: rtw_ack_tx_wait
kwi 10 22:38:32.057361 Kejk-PC kernel: RTL871X: issue_nulldata(wlx704d7b153e6c) to ac:22:0b:1f:0c:6c, ch:36, 3/3 in 1536 ms
...
kwi 10 22:38:41.273401 Kejk-PC kernel: RTL871X: chk_ap_is_alive(wlx704d7b153e6c) rx:(m:0, c:0, d:0), beacon:0, probersp_to_self:0, retry:4
kwi 10 22:38:41.273407 Kejk-PC kernel: RTL871X: chk_ap_is_alive(wlx704d7b153e6c) tx_pkts:18748, link_count:3
kwi 10 22:38:41.785343 Kejk-PC kernel: RTL871X: rtw_sctx_wait timeout: rtw_ack_tx_wait
kwi 10 22:38:42.297337 Kejk-PC kernel: RTL871X: rtw_sctx_wait timeout: rtw_ack_tx_wait
kwi 10 22:38:42.809365 Kejk-PC kernel: RTL871X: rtw_sctx_wait timeout: rtw_ack_tx_wait
kwi 10 22:38:42.809386 Kejk-PC kernel: RTL871X: issue_probereq_ex(wlx704d7b153e6c) to ac:22:0b:1f:0c:6c, ch:36, 3/3 in 1536 ms
kwi 10 22:38:43.321339 Kejk-PC kernel: RTL871X: rtw_sctx_wait timeout: rtw_ack_tx_wait
kwi 10 22:38:43.833355 Kejk-PC kernel: RTL871X: rtw_sctx_wait timeout: rtw_ack_tx_wait
kwi 10 22:38:44.345335 Kejk-PC kernel: RTL871X: rtw_sctx_wait timeout: rtw_ack_tx_wait
kwi 10 22:38:44.345362 Kejk-PC kernel: RTL871X: issue_nulldata(wlx704d7b153e6c) to ac:22:0b:1f:0c:6c, ch:36, 3/3 in 1536 ms
kwi 10 22:38:44.345370 Kejk-PC kernel: RTL871X: linked_status_chk(wlx704d7b153e6c) disconnect or roaming
kwi 10 22:38:51.581274 Kejk-PC kernel: RTL871X: There are some pkts to transmit 2
kwi 10 22:38:51.581295 Kejk-PC kernel: RTL871X: free_xmitbuf_cnt: 4, free_xmit_extbuf_cnt: 28

Driver keeps checking if connection with Access Point is still alive (chk_ap_is_alive). When there is no traffic it sends empty (?) packets (issue_nulldata, rtw_sctx_wait) and waits for the response. For some reason device hangs and is not sending packages (or handling retrieved responses). The issue_nulldata and rtw_sctx_wait functions are timing out, connection is closed and disconnect signal is send to operating system.
Situation is even tricker: sometimes device is able to reset itself (by happy coincident). Driver has build in power saving. If there is no active connection (as in our case) devices in turned off. Usually network manager will try to scan for new WIFI networks after few seconds so device is again powered on and everything is back to normal. After this reset connection is working fine. But if there is something left in xmitbuf or xmit_extbuf buffers then it writes "There are some pkts to transmit" error message to logs and refuses to power off. Unloading kernel module or reconnecting device to USB helps, restart is not required.
It looks that the problem is solely in the driver because 5.1.5 version is working fine for me - I tested it for a few days - no hangs, timeouts or disconnects.

@abperiasamy are you planning to use 5.1.5 version?

BTW, official Asus drivers for Windows also has this issue! I even got warranty replacement but issue still persisted.

uminokoe, 2017-04-15:
@kmaterka 
Thanks for your effort. I would like to remind you that your output seems to contain the MAC address of your wireless adapter. You may wish to edit / delete and re-post the comment for privacy reasons. (Especially if it is not the one replaced under warranty)

If you have a RTL8811AU / RTL8821AU chipset, please note that my repository is very experimental. I don't have the version 5.1.5 HAL files of the aforementioned chipsets (see [this commit](https://github.com/uminokoe/rtl8812AU/commit/5fc45ac1418cd36ec420ee78fa710d2a0d09bfcd)) and part of the code regarding the "ant band switch issue" is commented out as a temporary workaround (see [this commit](https://github.com/uminokoe/rtl8812AU/commit/c9b2f4186c7f21917e28cfa1e3395dc7efb5369c)).

kmaterka, 2017-04-17:
@uminokoe
It is RTL8812AU. I guess that issue is somewhere within HAL but I don't have enough knowledge about drivers to verify it.

mk-fg, 2019-01-18:
@rks80 Tree up to this commit - https://github.com/mk-fg/rtl8812au/commit/495c2d992cbe318572ed4fdcf5f5cd8f37e93a5a - should be exactly what was in that repo.
Though iirc 5.2.20 was a much better and complete version, unlike 5.1.5 that was pieced together from different ones like uminokoe described.
That's as of year ago, when I last looked at that driver, there're probably newer releases by now.

cheukswiss, 2022-11-10:
i got same problem of this. the driver print massage something like this 
```
<4>[55574.378111] RTW: There are some pkts to transmit
<4>[55574.382896] RTW: free_xmitbuf_cnt: 0, free_xmit_extbuf_cnt: 32
<4>[55576.330851] RTW: xmit_status_check REG_TXDMA_STATUS:0x00004020
```
the bug occur time is random, it's hard to dbg. and i also got the msg
```
RTW: rtw_set_ps_mode(wlan0) Enter 802.11 power save - WIFI-TRAFFIC_IDLE
```
i found when the problem occur, the driver is no longer exit power save mode

kmaterka, 2022-11-14:
I'm using https://github.com/aircrack-ng/rtl8812au now and it works fine.