abperiasamy/rtl8812AU_8821AU_linux #192: Can't see WLAN networks

shrugal, 2017-01-26:
Hey,

I have the [CSL WLS41186](http://www.linux-hardware-guide.de/2016-02-12-csl-ac1200-stick-dual-band-wlan-usb-3-0-adapter-ac1200-dual-band) and kernel version 4.9.5-200.fc25.x86_64.

The driver installed successfully and I see the stick in iwconfig, but it can't find any wireless networks.

spesut, 2017-01-27:
I had the same problem after a recent update too.
The issue seems to be caused by an update to wpa_supplicant (wpa_supplicant-2.6-1.fc25).

Downgrading to the previous package (wpa_supplicant-2.5-5.fc24) resolved the issue for me:
>dnf downgrade wpa_supplicant

shrugal, 2017-01-27:
That worked, thank you!

shartrec, 2017-02-01:
I have the same problem with kernel 4.9.6-200.fc25.x86_64 and wpa_supplicant-1:2.6-1.fc25.x86_64.

piraz, 2017-02-18:
It seems like when I've added the network manually the wireless interface started to find the network. Here is the log from the operation:

```
Feb 18 12:32:36 machine.local NetworkManager[3829]: <info>  [1487439156.4442] device (wlp0s19f2u5): supplicant interface state: scanning -> disconnected
Feb 18 12:32:36 machine.local NetworkManager[3829]: <info>  [1487439156.4681] device (wlp0s19f2u5): state change: failed -> disconnected (reason 'none') [120 30 0]
Feb 18 12:32:36 machine.local NetworkManager[3829]: <warn>  [1487439156.7145] device (wlp0s19f2u5): set-hw-addr: new MAC address XX:XX:XX:XX:XX:XX not successfully set (scanning)
Feb 18 12:32:36 machine.local NetworkManager[3829]: <warn>  [1487439156.8219] device (wlp0s19f2u5): set-hw-addr: new MAC address XX:XX:XX:XX:XX:XX not successfully set (scanning)
Feb 18 12:32:36 machine.local NetworkManager[3829]: <warn>  [1487439156.8232] firewall: [0x561d427124d0,remove:"wlp0s19f2u5"]: complete: request failed (UNKNOWN_INTERFACE: 'wlp0s19f2u5' is not in any zone)
Feb 18 12:32:36 machine.local NetworkManager[3829]: <info>  [1487439156.8267] device (wlp0s19f2u5): supplicant interface state: disconnected -> disabled
Feb 18 12:32:36 machine.local NetworkManager[3829]: <info>  [1487439156.8418] device (wlp0s19f2u5): supplicant interface state: disabled -> disconnected
Feb 18 12:32:40 machine.local NetworkManager[3829]: <warn>  [1487439160.5445] device (wlp0s19f2u5): set-hw-addr: new MAC address XX:XX:XX:XX:XX:XX not successfully set (scanning)
```

It seems like the firewall was introduced to the interface and everything is running smooth again.

Should you add some instruction to the install script to make the firewall be introduced to the interface?

piraz, 2017-02-18:
Ok, I got the issue here. The problem is that the script for the interface doesn't exists on /etc/sysconfig/network-scripts. When I've added the network manually one file was created and now wpa_supplicant is behaving like an adult.

I don't know if this is a problem just with my device or something related to the wpa_supplicant-1:2.6-1.fc25.x86_64 version.

I think this latest version of wpa_supplicant is behaving differently than the previous version when trying to set the MAC address to the device. It is failing if no script is found on etc/sysconfig/network-scripts directory.

ojnickel, 2017-03-15:
i've got the same problem. Well, i had :-).
I've tried a couple (lot) of open source drivers. Some of them worked for a while and some (including this one) didn't work at all.
On my rpi works just fine. I didn't know, what the problem was with my gentoo. 'Til i read this thread.
I'm using kernel 4.4.39 and wpa_supplicant 2.6. 
With iwconfig the interface was listed, but in NM i didn't see any WLAN Networks.
So, i tried connecting as it were a hidden network and it worked like a charm!
Thank u so much shartrec

spesut, 2017-03-16:
There's an alternate fix for this issue.  I updated to 4.3.22-beta and everything is working correctly.  You can snag it from the following repo:
https://github.com/diederikdehaas/rtl8812AU

Kernel - 4.9.13-201
wpa_supplicant - 2.6

gsora, 2017-05-21:
I'm having this problem too, with a Netis WF2190 on Linux 4.11.1: NetworkManager refuses to see any network until I manually connect to a wireless network through `nmcli`.

shartrec, 2017-07-14:
I just upgraded to Fedora 26,  NetworkManager-1.8.0 and  the driver now sees networks at startup  and connects automatically.