abperiasamy/rtl8812AU_8821AU_linux #194: edimax ew-7811dac doesn't work - iwlist no scan results

Goga777, 2017-01-28:
I have edimax ew-7811dac based on 8811au chipset and it doesn't work

 dmesg | grep 8812
[   15.786488] 8812au: loading out-of-tree module taints kernel.
[   15.789019] 8812au: module verification failed: signature and/or required key missing - tainting kernel
[   15.797248] RTL871X: rtl8812au v4.3.20_16317.20160108
[   15.999787] usbcore: registered new interface driver rtl8812au
[   16.853687] rtl8812au 1-2:1.0 wlx74da3867a2c1: renamed from wlan1
[ 2079.856247] RTL871X: rtl8812au v4.3.14_13455.20150212_BTCOEX20150128-51
[ 2079.856250] RTL871X: rtl8812au BT-Coex version = BTCOEX20150128-51
[ 2079.856290] Error: Driver 'rtl8812au' is already registered, aborting...
[ 2162.449753] usbcore: deregistering interface driver rtl8812au
[ 2178.136814] RTL871X: rtl8812au v4.3.14_13455.20150212_BTCOEX20150128-51
[ 2178.136817] RTL871X: rtl8812au BT-Coex version = BTCOEX20150128-51
[ 2178.350634] usbcore: registered new interface driver rtl8812au
[ 2178.363823] rtl8812au 1-2:1.0 wlx74da3867a2c1: renamed from wlan0

usb 1-2: USB disconnect, device number 3
[ 2177.962227] usb 1-2: new high-speed USB device number 4 using ehci-pci
[ 2178.111443] usb 1-2: New USB device found, idVendor=7392, idProduct=a812
[ 2178.111461] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2178.111470] usb 1-2: Product: Edimax AC600 USB
[ 2178.111478] usb 1-2: Manufacturer: Realtek
[ 2178.111486] usb 1-2: SerialNumber: 00e04c000001
[ 2178.136803] RTL871X: module init start
[ 2178.136814] RTL871X: rtl8812au v4.3.14_13455.20150212_BTCOEX20150128-51
[ 2178.136817] RTL871X: rtl8812au BT-Coex version = BTCOEX20150128-51
[ 2178.345163] RTL871X: rtw_ndev_init(wlan0)
[ 2178.350634] usbcore: registered new interface driver rtl8812au
[ 2178.350644] RTL871X: module init ret=0
[ 2178.363823] rtl8812au 1-2:1.0 wlx74da3867a2c1: renamed from wlan0

ifconfig
enp1s0f0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 30:65:ec:05:b0:4e  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 26  

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        loop  txqueuelen 1  (Local Loopback)
        RX packets 2  bytes 100 (100.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 2  bytes 100 (100.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlp5s0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.64  netmask 255.255.255.0  broadcast 192.168.1.255
        ether 0c:84:dc:0d:6a:17  txqueuelen 1000  (Ethernet)
        RX packets 32893  bytes 15165843 (14.4 MiB)
        RX errors 0  dropped 1  overruns 0  frame 0
        TX packets 16352  bytes 2982547 (2.8 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlx74da3867a2c1: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 74:da:38:67:a2:c1  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

iw list
Wiphy phy3
	max # scan SSIDs: 9
	max scan IEs length: 2304 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
[... 329 lines cut ...]
          Encryption key:off
          Power Management:off
          Link Quality=49/70  Signal level=-61 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:11   Missed beacon:0

lo        no wireless extensions.

wlx74da3867a2c1  unassociated  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

enp1s0f0  no wireless extensions.

 iwlist wlx74da3867a2c1 scanning 
wlx74da3867a2c1  No scan results

iwlist wlx74da3867a2c1 scanning essid Keenetic-71
wlx74da3867a2c1  No scan results

andyfleas, 2017-03-04:
Am having same issues with a Brostrend A2-600 8811AU chipset. 
I figured out all terminal commands for the 8812AU for A1 & A3 WiFi adapter, are the same chipsets so same commands, works good.

Goga777, 2017-03-04:
did you find the solution to solve the issue ? may other drivers ?

andyfleas, 2017-03-04:
Unfortunately, no. Am still working on it tho. Looking in all places. Trying to find other adapter brands using same chipset 8811AU that may have commands in UBUNTU/Linux forums.

andyfleas, 2017-03-04:
Apparently, these following brands are same chipset 8811AU, I have not investigated these drivers for Linux...yet. I'm assuming for #7, 8, 10, the N/A at end meant they did not work for Win OS:
 
Laptop (assuming, he didn't say Win OS or vs used)- Dell Inspirion 15 - D 24.43 U 5.24 (I don't use Dell)
1 - Glam Hobby - ASIN X000WFVCL5 - D 12.86 U 5.11
2 - Aukey - ASIN X000ZPDDA4T - D 27.85 U 5.15 | (2nd speed test) D 25.05 U 5.05
3 - Edup - ASIN X0012T4ZKF - D 14.5 U 0.70
4 - BrosTrend - ASIN X000UIGE9J - D 15.75 U 5.21
5 - Urant - ASIN X00138BC01 - D 14.65 U 5.18
6 - Panda Wireless - ASIN B00JDVRCI0 - D 22.49 U 5.16 | (2nd speed test) D 22.50 U 5.20
7 - Motoraux - ASIN X000XID79V - NA
8 - LB1 High Performance - ASIN X00103I4RD - NA
9 - TurnRaise - ASIN X000W9A7TT - D 9.43 U 5.10
10 - LB1 High Performance - ASIN X000TD4Y5L - D 17.44 U 5.24
11 - Linkpal - ASIN X000ZDQFMZ - D 9.5 U 2.47
12 - Linkpal - ASIN X000Z159TR - NA