abperiasamy/rtl8812AU_8821AU_linux #195: make fails on csum_ipv6_magic

djjerdog, 2017-02-04:
Trying to build driver on Odroid-C2 running Debian 8.

  CC [M]  /root/rtl8812AU_8821AU_linux/core/rtw_br_ext.o
/root/rtl8812AU_8821AU_linux/core/rtw_br_ext.c: In function ‘nat25_db_handle’:
/root/rtl8812AU_8821AU_linux/core/rtw_br_ext.c:1346:7: error: implicit declaration of function ‘csum_ipv6_magic’ [-Werror=implicit-function-declaration]
       hdr->icmp6_cksum = csum_ipv6_magic(&iph->saddr, &iph->daddr,
       ^
cc1: all warnings being treated as errors
scripts/Makefile.build:308: recipe for target '/root/rtl8812AU_8821AU_linux/core/rtw_br_ext.o' failed
make[2]: *** [/root/rtl8812AU_8821AU_linux/core/rtw_br_ext.o] Error 1
Makefile:1278: recipe for target '_module_/root/rtl8812AU_8821AU_linux' failed
make[1]: *** [_module_/root/rtl8812AU_8821AU_linux] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-3.14.79+'
Makefile:1570: recipe for target 'modules' failed
make: *** [modules] Error 2

kevin0932, 2017-06-23:
Hi @djjerdog , did you solve the problem? I met with the same error when I tried to compile for Jetson TX1.  I am really appreciated if you can share some experience with me!

ptpt52, 2017-06-26:
kernel version ?

z5n0w, 2019-11-13:
instal rtl8812-dkms on Armbian with 4.19.63-sunxi64 kernel meet the same issue