abperiasamy/rtl8812AU_8821AU_linux #198: Not compiling / no DKMS install with Kernel 4.9.7

Gymnae, 2017-02-13:
uname -r
`4.9.7-cubox`

```
sudo make -f Makefile.dkms install
make clean
make[1]: Entering directory '/root/rtl8812AU_8821AU_linux'
make ARCH= CROSS_COMPILE= -C  M=/root/rtl8812AU_8821AU_linux clean
make[2]: Entering directory '/root/rtl8812AU_8821AU_linux'
make[2]: *** M=/root/rtl8812AU_8821AU_linux: No such file or directory.  Stop.
make[2]: Leaving directory '/root/rtl8812AU_8821AU_linux'
Makefile:1591: recipe for target 'clean' failed
make[1]: *** [clean] Error 2
make[1]: Leaving directory '/root/rtl8812AU_8821AU_linux'
Makefile.dkms:11: recipe for target 'src_install' failed
make: *** [src_install] Error 2

```

phouverneyuff, 2017-02-25:
Same here! with kernel 4.9.10 fedora 25

Gymnae, 2017-03-04:
Might have found a fix:
https://github.com/diederikdehaas/rtl8812AU/pull/62
At least this works for i.mx6 based systems. You could try to hardcode the architecture for the system you are using, for example for raspberry pi:

ifeq ($(CONFIG_PLATFORM_ARM_RPI), y)
EXTRA_CFLAGS += -DCONFIG_LITTLE_ENDIAN
EXTRA_CFLAGS += -DCONFIG_IOCTL_CFG80211 -DRTW_USE_CFG80211_STA_EVENT
ARCH := arm
CROSS_COMPILE ?=
KVER ?= $(shell uname -r)
KSRC := /lib/modules/$(KVER)/build
MODDESTDIR := /lib/modules/$(KVER)/kernel/drivers/net/wireless/
INSTALL_PREFIX :=
endif

ptpt52, 2017-03-07:
your code is the latest HEAD?

ptpt52, 2017-03-07:
$ sudo make -f Makefile.dkms install
make clean
make[1]: Entering directory '/mnt/HOME/terminator/Sources/rtl8812AU_8821AU_linux'
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.10.1-ptpt52/build M=/mnt/HOME/terminator/Sources/rtl8812AU_8821AU_linux clean
make[2]: Entering directory '/mnt/HOME/terminator/Sources/kernel/linux.git'
make[2]: Leaving directory '/mnt/HOME/terminator/Sources/kernel/linux.git'
make[1]: Leaving directory '/mnt/HOME/terminator/Sources/rtl8812AU_8821AU_linux'
mkdir -p '/usr/src/rtl8812au-4.3.14'
cp -r dkms.conf Kconfig Makefile.dkms Makefile platform core hal include os_dep '/usr/src/rtl8812au-4.3.14'
cp Makefile '/usr/src/rtl8812au-4.3.14/Makefile'
sed 's/#MODULE_VERSION#/4.3.14/' dkms.conf > '/usr/src/rtl8812au-4.3.14/dkms.conf'
dkms add -m rtl8812au -v 4.3.14 2>/dev/null || true

Creating symlink /var/lib/dkms/rtl8812au/4.3.14/source ->
                 /usr/src/rtl8812au-4.3.14

DKMS: add completed.
dkms build -m rtl8812au -v 4.3.14

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make KERNELRELEASE=4.10.1-ptpt52 KVER=4.10.1-ptpt52............
cleaning build area....

DKMS: build completed.
dkms install -m rtl8812au -v 4.3.14

rtl8812au:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.10.1-ptpt52/updates/dkms/

depmod....

DKMS: install completed.

closed 2017-03-07 by ptpt52