abperiasamy/rtl8812AU_8821AU_linux #199: error: initialization from incompatible pointer type [-Werror]

philm001, 2017-02-22:
I am receiving this compile error in the file ioctl_cfg80211.c on line 5971.

Specifically, .del_station = cfg80211_rtw_del_station

kevin0932, 2017-06-23:
Hi @philm001 , did you solve the problem?

ptpt52, 2017-06-26:
... kernel version ?

mariavd, 2017-07-01:
Hello,

I am also receiving this error but for another pointer:

```
/home/MYUSERNAME/Downloads/rtl8812AU_8821AU_linux/os_dep/linux/os_intfs.c:809:2: error: initialization from incompatible pointer type [-Werror]
  .ndo_select_queue = rtw_select_queue,
  ^
/home/MYUSERNAME/Downloads/rtl8812AU_8821AU_linux/os_dep/linux/os_intfs.c:809:2: error: (near initialization for ‘rtw_netdev_ops.<anonymous>.ndo_select_queue’) [-Werror]
```

Compiling on CentOS 7 with kernel version 3.10.0-514.21.1.el7.x86_64.

bhmevik, 2017-07-02:
I get the same (originally reported) error:
```
/var/lib/dkms/rtl8812au/4.3.14/build/os_dep/linux/ioctl_cfg80211.c:5971:2: error: initialization from incompatible pointer type [-Werror]
  .del_station = cfg80211_rtw_del_station,
  ^
/var/lib/dkms/rtl8812au/4.3.14/build/os_dep/linux/ioctl_cfg80211.c:5971:2: error: (near initialization for ‘rtw_cfg80211_ops.del_station’) [-Werror]
cc1: all warnings being treated as errors
```

This is on a Linux Min 17.3 system, which is based on Ubuntu 14.04, with
```
$ uname -r
3.19.0-32-generic
$ gcc --version
gcc (Ubuntu 4.8.4-2ubuntu1~14.04.3) 4.8.4
```

bhmevik, 2017-07-02:
Following the tip in [Issue 176](https://github.com/abperiasamy/rtl8812AU_8821AU_linux/issues/176), I commented out the line `EXTRA_CFLAGS += -Werror` in `Makefile` and ran `sudo make -f Makefile.dkms install` again, and then it worked. Plugging in the nic, everything seems to work, and I was able to connect to the wireless network.

kimocoder, 2017-07-02:
Confirmed. we encountered this in another fork (astsam's fork) few weeks back too, sorry for not  paying attention hele earlier