abperiasamy/rtl8812AU_8821AU_linux #200: Can't compile

Silveris, 2017-03-01:
Hello there,
First of all I'm a fresh linux user. I've installed Solus distro (kernel 4.9.12) on my pc which connects to the Internet using WiFi via TP-LINK Archer T4UH AC1200 High Gain Wireless Dual Band USB Adapter. Plug and Play doesn't work here so I've read that I need to install rtl8812au driver.
I've installed: git, linux-headers, kernel-headers, make, gcc, g++
I'm using these commands: git clone (repo link here), cd (directory here), make, (later make install, modprobe (name))
And here is the output with errors:
http://pastebin.com/gbjTKwSy
How can I make it work? Thanks for any advices.

ptpt52, 2017-03-07:
32bit x86 ?

donbright, 2017-08-27:
i have compiled and installed without problem on 32bit x86 ubuntu 16.04