abperiasamy/rtl8812AU_8821AU_linux #203: T4U on kali linux

Gagootron, 2017-05-03:
I tried to use this driver on Kali linux (kernel version 4.9) with a 8853A-T4U.
So i cloned it to /usr/src used make to build and install. when i then plug it in I see it in the Network manager as a Usb Wifi adapter but it can't find any wifi networks, airmon-ng can't put it in monitor mode and the led doesn't blink either.

ptpt52, 2017-05-13:
post some logs? dmesg or something

Gagootron, 2017-05-13:
here is my dmesg log.
if you need any other logs tell me witch.
the wifi adapter is in this case wlan1 and before i plugged the adapter in i uninstalled the current driver, cloned main, compiled and installed it again, just to be up-to-date.

[dmesg.log](http://gagootron.org/logs/dmesg.txt)

AlexKvazos, 2018-02-25:
Where you able to fix? This same issue, no compilation errors but the usb card simply won't blink or anything. I validated the card works on macOS.

kimocoder, 2018-02-25:
apt-get install realtek-rtl88xxau-dkms

AlexKvazos, 2018-02-25:
It doesn't work