abperiasamy/rtl8812AU_8821AU_linux #207: Compilation failure with -Werror=switch-unreachable (fix inside)

hschletz, 2017-05-31:
Compilation stopped working for me:
```
/var/lib/dkms/rtl8812au/4.3.14/build/hal/rtl8812a/rtl8812a_mp.c: In function ‘Hal_SetAntenna’:
/var/lib/dkms/rtl8812au/4.3.14/build/hal/rtl8812a/rtl8812a_mp.c:616:7: error: statement will never be executed [-Werror=switch-unreachable]
   u32 reg0xC50 = 0;
```
This was most likely caused by a GCC upgrade from 6.3 to 7.1 which appears to be more strict by default.

This is how I fixed it:
```
diff --git a/hal/rtl8812a/rtl8812a_mp.c b/hal/rtl8812a/rtl8812a_mp.c
index e20aa87..a6dd35d 100644
--- a/hal/rtl8812a/rtl8812a_mp.c
+++ b/hal/rtl8812a/rtl8812a_mp.c
@@ -579,6 +579,7 @@ void Hal_SetAntenna(PADAPTER pAdapter)
        //u8    chgTx = 0, chgRx = 0;
        u32     r_ant_select_ofdm_val = 0;
 
+       u32 reg0xC50 = 0;
 
        p_ofdm_tx = (R_ANTENNA_SELECT_OFDM *)&r_ant_select_ofdm_val;
        p_cck_txrx = (R_ANTENNA_SELECT_CCK *)&r_ant_select_cck_val;
@@ -613,7 +614,6 @@ void Hal_SetAntenna(PADAPTER pAdapter)
        }
 
        switch (pAdapter->mppriv.antenna_rx) {
-               u32 reg0xC50 = 0;
        case ANTENNA_A:
                PHY_SetBBReg(pAdapter, rRxPath_Jaguar, bMaskByte0, 0x11);                                                                                                          
                PHY_SetRFReg(pAdapter, ODM_RF_PATH_B, RF_AC_Jaguar, 0xF0000, 0x1); // RF_B_0x0[19:16] = 1, Standby mode
```

harshavardhana, 2017-05-31:
Please send a proper PR @hschletz - we will happily accept.

hschletz, 2017-05-31:
Added PR #208.

harshavardhana, 2017-06-01:
Fixed by #208

closed 2017-06-01 by harshavardhana