abperiasamy/rtl8812AU_8821AU_linux #210: Stability issues.

nibbles-, 2017-06-08:
I am running:
`Linux hostname 4.11.3-200.fc25.x86_64 #1 SMP Thu May 25 19:03:07 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux`
`Fedora release 25 (Twenty Five) ` with a `ASUS USB AC56`

I can compile and install the driver (master branch) but it is extreamly unstable. It can run fine for about an hour streaming Netflix then it will disconnect and seem to dissapear from the system and I have to reload the module to get it working again.
Also (perhaps unrelated but doesnt seem like it) when it works I can do 'iwlist devicename scanning' and see all available SSIDs but networkmanager will not find anything and I have to manually create a connection to make the interface connect.
To fix it when it crashes I run the following:
`modprobe -r rtl8812au
modprobe rtl8812au
nmcli device connect wlp0s26u1u6`
I am at a loss as what todo so any help is appriciated.
I have attached lsusb -D /devpath and dmes for the module

[lsusb-asus-ac56.txt](https://github.com/abperiasamy/rtl8812AU_8821AU_linux/files/1062200/lsusb-asus-ac56.txt)
[dmesg-rtl871x.txt](https://github.com/abperiasamy/rtl8812AU_8821AU_linux/files/1062201/dmesg-rtl871x.txt)

nibbles-, 2017-06-08:
https://github.com/xxNull-lsk/rtl8812AU This is a new version of the driver which seems to fix atleast the networkmanager part

kimocoder, 2017-06-08:
Jepp, @ Kali we working on https://github.com/astsam/rtl8812au/tree/v5.1.5 too :)

dschwartz783, 2017-08-11:
@kimocoder it would be nice if people would stop forking this code and just make contributions to one repo.

kimocoder, 2017-08-11:
astsam's fork got monitor & injection patches, this one does not. besides, if people stopped forking & working.. well.. what do we have Github for? anyways, everyone if free to collect/cherry pick patches of each repo to one freely.

dschwartz783, 2017-08-11:
@kimocoder the point is that many improvements have been made to the various forks, but they have not all have found their way into a single repo. This is the problem with forking being so easy on GitHub. I've seen this issue recur with a number of projects.

kimocoder, 2017-08-11:
I agree, but maintainers may collect those patches, they may be pulled & commited both ways and 3rd party users may also collect all together as one  so.. this is simply how it works & it's a good in several ways too.