abperiasamy/rtl8812AU_8821AU_linux #211: Doesn't Work on Fedora 25: "link is not ready"

vatrat, 2017-06-11:
I installed the driver like this:

```
git clone https://github.com/abperiasamy/rtl8812AU_8821AU_linux.git
cd rtl8812AU_8821AU_linux
make
sudo make install
sudo modprobe rtl8812au
```

NetworkManager can see the device, but says it's disconnected. Iwconfig says that it no devices are found when scanning with the device.
I'm using the TP-LINK Archer T4U. The variant is US/1.0.
dmesg shows a bunch of messages saying "link is not ready".

```
[ 3032.120963] usb 2-1: new high-speed USB device number 7 using ehci-pci
[ 3032.239906] usb 2-1: New USB device found, idVendor=2357, idProduct=0101
[ 3032.239912] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 3032.239916] usb 2-1: Product: 802.11n NIC
[ 3032.239920] usb 2-1: Manufacturer: Realtek
[ 3032.239924] usb 2-1: SerialNumber: 123456
[ 3032.576038] RTL871X: rtw_ndev_init(wlan1)
[ 3033.265910] rtl8812au 2-1:1.0 wlp0s29f7u1: renamed from wlan1
[ 3033.290263] IPv6: ADDRCONF(NETDEV_UP): wlp0s29f7u1: link is not ready
[ 3033.893944] IPv6: ADDRCONF(NETDEV_UP): wlp0s29f7u1: link is not ready
[ 3034.004502] IPv6: ADDRCONF(NETDEV_UP): wlp0s29f7u1: link is not ready
[ 3034.050348] IPv6: ADDRCONF(NETDEV_UP): wlp0s29f7u1: link is not ready
[ 3034.156689] IPv6: ADDRCONF(NETDEV_UP): wlp0s29f7u1: link is not ready
[ 3034.377098] IPv6: ADDRCONF(NETDEV_UP): wlp0s29f7u1: link is not ready
[ 3034.597201] IPv6: ADDRCONF(NETDEV_UP): wlp0s29f7u1: link is not ready
[ 3034.818389] IPv6: ADDRCONF(NETDEV_UP): wlp0s29f7u1: link is not ready
[ 3035.055081] IPv6: ADDRCONF(NETDEV_UP): wlp0s29f7u1: link is not ready
[ 3035.278006] IPv6: ADDRCONF(NETDEV_UP): wlp0s29f7u1: link is not ready
[ 3040.277929] IPv6: ADDRCONF(NETDEV_UP): wlp0s29f7u1: link is not ready
[ 3040.506543] IPv6: ADDRCONF(NETDEV_UP): wlp0s29f7u1: link is not ready
[ 3040.728434] IPv6: ADDRCONF(NETDEV_UP): wlp0s29f7u1: link is not ready
[ 3040.950589] IPv6: ADDRCONF(NETDEV_UP): wlp0s29f7u1: link is not ready
[ 3041.182093] IPv6: ADDRCONF(NETDEV_UP): wlp0s29f7u1: link is not ready
[ 3041.404795] IPv6: ADDRCONF(NETDEV_UP): wlp0s29f7u1: link is not ready
[ 3041.645269] IPv6: ADDRCONF(NETDEV_UP): wlp0s29f7u1: link is not ready
[ 3041.880518] IPv6: ADDRCONF(NETDEV_UP): wlp0s29f7u1: link is not ready
[ 3042.099022] IPv6: ADDRCONF(NETDEV_UP): wlp0s29f7u1: link is not ready
[ 3042.335112] IPv6: ADDRCONF(NETDEV_UP): wlp0s29f7u1: link is not ready
[ 3042.555709] IPv6: ADDRCONF(NETDEV_UP): wlp0s29f7u1: link is not ready
[ 3042.778465] IPv6: ADDRCONF(NETDEV_UP): wlp0s29f7u1: link is not ready
[ 3042.999660] IPv6: ADDRCONF(NETDEV_UP): wlp0s29f7u1: link is not ready
[ 3043.219330] IPv6: ADDRCONF(NETDEV_UP): wlp0s29f7u1: link is not ready
[ 3043.454833] IPv6: ADDRCONF(NETDEV_UP): wlp0s29f7u1: link is not ready
```

If I unplug the device and plug it back in, dmesg shows this:

```
[ 3248.252867] usb 2-1: USB disconnect, device number 7
[ 3248.275889] RTL871X: rtw_ndev_uninit(wlp0s29f7u1)
[ 3248.291911] RTL871X: rtw_cmd_thread: DriverStopped(1) SurpriseRemoved(1) break at line 550
[ 3248.291972] RTL871X: rtw_dev_unload: driver not in IPS
[ 3251.705887] usb 2-1: new high-speed USB device number 8 using ehci-pci
[ 3251.824991] usb 2-1: New USB device found, idVendor=2357, idProduct=0101
[ 3251.824998] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 3251.825003] usb 2-1: Product: 802.11n NIC
[ 3251.825007] usb 2-1: Manufacturer: Realtek
[ 3251.825011] usb 2-1: SerialNumber: 123456
[ 3252.164646] RTL871X: rtw_ndev_init(wlan1)
[ 3253.092105] rtl8812au 2-1:1.0 wlp0s29f7u1: renamed from wlan1
[ 3253.130915] IPv6: ADDRCONF(NETDEV_UP): wlp0s29f7u1: link is not ready
[ 3253.741610] IPv6: ADDRCONF(NETDEV_UP): wlp0s29f7u1: link is not ready
[ 3253.855371] IPv6: ADDRCONF(NETDEV_UP): wlp0s29f7u1: link is not ready
[ 3253.896355] IPv6: ADDRCONF(NETDEV_UP): wlp0s29f7u1: link is not ready
[ 3254.003090] IPv6: ADDRCONF(NETDEV_UP): wlp0s29f7u1: link is not ready
[ 3254.228735] IPv6: ADDRCONF(NETDEV_UP): wlp0s29f7u1: link is not ready
[ 3254.448438] IPv6: ADDRCONF(NETDEV_UP): wlp0s29f7u1: link is not ready
[ 3254.668909] IPv6: ADDRCONF(NETDEV_UP): wlp0s29f7u1: link is not ready
[ 3254.913654] IPv6: ADDRCONF(NETDEV_UP): wlp0s29f7u1: link is not ready
```

Of course, it continues to give the "link is not ready" message. I've tried this on several computers.
```
$ uname -a
Linux laptop1 4.11.3-202.fc25.x86_64 #1 SMP Mon Jun 5 16:38:21 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

```

ptpt52, 2017-06-16:
I think it is working.

vatrat, 2017-06-16:
What? No. NetworkManager shows the device as finding no networks. When I use nmtui, it doesn't let me use the device.

vatrat, 2017-06-16:
This worked on Arch Linux, but does not work on Fedora.

ptpt52, 2017-06-17:
try this https://github.com/astsam/rtl8812au/tree/v5.1.5
and give me some usage reports

vatrat, 2017-06-20:
Okay, I'll try that now.

vatrat, 2017-06-20:
After cloning:
```
$ make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.11.4-200.fc25.x86_64/build M=/home/vatrat/rtl8812au  modules
make[1]: Entering directory '/usr/src/kernels/4.11.4-200.fc25.x86_64'
  CC [M]  /home/vatrat/rtl8812au/core/rtw_cmd.o
In file included from /home/vatrat/rtl8812au/include/drv_types.h:32:0,
                 from /home/vatrat/rtl8812au/core/rtw_cmd.c:22:
/home/vatrat/rtl8812au/include/osdep_service.h: In function ‘thread_enter’:
/home/vatrat/rtl8812au/include/osdep_service.h:343:2: error: implicit declaration of function ‘allow_signal’ [-Werror=implicit-function-declaration]
  allow_signal(SIGTERM);
  ^~~~~~~~~~~~
/home/vatrat/rtl8812au/include/osdep_service.h: In function ‘flush_signals_thread’:
/home/vatrat/rtl8812au/include/osdep_service.h:353:6: error: implicit declaration of function ‘signal_pending’ [-Werror=implicit-function-declaration]
  if (signal_pending (current))
      ^~~~~~~~~~~~~~
/home/vatrat/rtl8812au/include/osdep_service.h:355:3: error: implicit declaration of function ‘flush_signals’ [-Werror=implicit-function-declaration]
   flush_signals(current);
   ^~~~~~~~~~~~~
cc1: some warnings being treated as errors
scripts/Makefile.build:294: recipe for target '/home/vatrat/rtl8812au/core/rtw_cmd.o' failed
make[2]: *** [/home/vatrat/rtl8812au/core/rtw_cmd.o] Error 1
Makefile:1496: recipe for target '_module_/home/vatrat/rtl8812au' failed
make[1]: *** [_module_/home/vatrat/rtl8812au] Error 2
make[1]: Leaving directory '/usr/src/kernels/4.11.4-200.fc25.x86_64'
Makefile:1264: recipe for target 'modules' failed
make: *** [modules] Error 2
```

vatrat, 2017-06-20:
I can't find the correct instructions to install this. I modified the makefile by adding this line:
`EXTRA_CFLAGS += -Wno-error=implicit-function-declaration`
This allowed me to compile the driver, but the driver doesn't work. I tried
```
sudo make install
sudo modprobe 8812au
```
but it tells me there was an error and I should check dmesg. Dmesg gives me this:
```
[214835.218601] 8812au: loading out-of-tree module taints kernel.
[214835.227048] 8812au: module verification failed: signature and/or required key missing
 - tainting kernel                                                                      
[214835.227278] 8812au: Unknown symbol signal_pending (err 0)
[214835.227507] 8812au: Unknown symbol allow_signal (err 0)
[214852.663288] No module found in object
[214858.095489] 8812au: module license 'unspecified' taints kernel.
[214858.095492] Disabling lock debugging due to kernel taint
[214858.095621] 8812au: Unknown symbol init_module (err 0)
[214858.095658] 8812au: Unknown symbol cleanup_module (err 0)
[214999.438630] 8812au: Unknown symbol signal_pending (err 0)
[214999.438851] 8812au: Unknown symbol allow_signal (err 0)
```
It seems that the functions allow_signal, signal_pending, and flush_signals aren't usable. As well, init_module and cleanup_module.
Is there another file that I need? A file with the functions allow_signal, signal_pending, and flush_signals?

ptpt52, 2017-06-25:
rtl drivers code is really dirty...can't help sorry

jollyboyjohn, 2017-08-20:
I had the same issue the `link is not ready`. 

I found out that it is expected behavior when something is trying to use an unassociated connection.

Doing a `iwlist scan`, this showed me that the driver was working fine. 

In my case, it was wpa_supplicant-related issue.

vatrat, 2017-08-20:
How did you fix it?

jollyboyjohn, 2017-08-20:
As I stated earlier, I used `iwlist scan`. This confirmed to me the driver wasn't at fault first. I saw some networks, just not the ones I was wanting.

Then I switched my domain in my wpa_supplicant.conf, and in the wireless cfg80211 regdom module options.

Then finally I hard-coded my SSID and passphrase into wpa_supplicant.conf - no NetworkManager. I also removed the update_config line to stop external factors from changing the region, which has happened before.

If the same driver works on ArchLinux but not something else like Fedora, it's highly likely to be an external factor like wireless libraries or user permissions like not being in the netdev group or similar - this might explain why nmtui isn't working. I've heard some people just get it working by installing 'NetworkManager-wifi'.