abperiasamy/rtl8812AU_8821AU_linux #214: performance issues with TRENDnet TEW-805UB on RPI3

dschwartz783, 2017-06-25:
Basically what the header says. It seems that the adapter is being limited to a single stream of 802.11ac. Both the adapter and the access point support three-stream 802.11ac (1.3Gbps, AirPort Extreme), and one version of this driver on GitHub was also able to support the faster speeds (although I'm not sure which), but unfortunately limited me to non-DFS channels, and so I had to switch to this one. It would be nice if we could work out the issue here... Thanks.

ptpt52, 2017-06-26:
really don't know how to fix this. Any example code?

dschwartz783, 2017-06-26:
I'm really not certain, but I believe it was the code from this repo that allowed multiple streams: https://github.com/gnab/rtl8812au

Where is the stream count controlled in the driver? I'm fairly new to this codebase.

dschwartz783, 2017-06-26:
I got it working my manually replacing RF_1T1R with RF_2T2R at https://github.com/abperiasamy/rtl8812AU_8821AU_linux/blob/00aedbde5cb6388ff5abf433d6798370c6a74102/hal/rtl8812a/rtl8812a_hal_init.c#L2192
Which tells me that it must be detecting the number of available antennas incorrectly.

dschwartz783, 2017-06-26:
Think I was mistaken, never mind.

closed 2017-06-26 by dschwartz783