abperiasamy/rtl8812AU_8821AU_linux #215: Page fault in rtw_mlme_ext.c on Ubuntu 16.04 LTS using latest master branch

maksyms, 2017-06-30:
Hi guys,

  I thought I'd report the page fault I'm regularly experiencing using RTL8811AU chipset and this driver:
`[ 5387.538393] RTL871X: rtw_set_802_11_connect(wlan0)  fw_state=0x00000009
[ 5387.555267] RTL871X: ERROR rtw_chk_start_clnt_join(wlan0) cur_ch:0
[ 5387.555271] ------------[ cut here ]------------
[ 5387.555306] WARNING: CPU: 0 PID: 1818 at /var/lib/dkms/rtl8812au/4.3.14_13455.20150212/build/core/rtw_mlme_ext.c:1                                                 3647 rtw_chk_start_clnt_join+0x77/0xa1 [rtl8812au]()
[ 5387.555307] Modules linked in: pci_stub vboxpci(OE) vboxnetadp(OE) vboxnetflt(OE) vboxdrv(OE) binfmt_misc rtl8812a                                                 u(OE) cfg80211 xfs snd_hda_codec_hdmi reiserfs nvidia(POE) snd_hda_codec_realtek snd_hda_codec_generic snd_hda_intel                                                  input_leds snd_hda_codec snd_hda_core snd_hwdep serio_raw snd_pcm snd_seq_midi snd_seq_midi_event snd_rawmidi lpc_ich                                                  snd_seq snd_seq_device snd_timer snd drm 8250_fintek x38_edac soundcore shpchp edac_core mac_hid kvm_intel kvm irqby                                                 pass ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 xt_hl ip6t_rt nf_conntrack_ipv6 nf_defrag_ipv6 ipt_REJECT nf_reject_ipv4                                                  xt_comment nf_log_ipv4 nf_log_common xt_LOG xt_multiport xt_limit xt_tcpudp xt_addrtype nf_conntrack_ipv4 nf_defrag_i                                                 pv4 xt_conntrack ip6table_filter ip6_tables nf_conntrack_netbios_ns nf_conntrack_broadcast nf_nat_ftp nf_nat
[ 5387.555358]  nf_conntrack_ftp nf_conntrack iptable_filter ip_tables x_tables cuse hwmon_vid coretemp parport_pc pp                                                 dev lp parport autofs4 btrfs raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq l                                                 ibcrc32c raid0 multipath linear raid1 pata_acpi hid_generic usbhid hid firewire_ohci firewire_core ahci crc_itu_t lib                                                 ahci sky2 pata_jmicron fjes
[ 5387.555389] CPU: 0 PID: 1818 Comm: RTW_CMD_THREAD Tainted: P        W  OE   4.4.0-83-generic #106-Ubuntu
[ 5387.555391] Hardware name: Shuttle SX38/FX38, BIOS 6.00 PG 12/16/2008
[ 5387.555393]  0000000000000286 000000001be9a9a0 ffff8800c1a0bda8 ffffffff813f9513
[ 5387.555397]  0000000000000000 ffffffffc1318bc0 ffff8800c1a0bde0 ffffffff81081322
[ 5387.555400]  ffffc90000ff5000 ffffc90000ff62da 0000000000000022 ffffc90000ff62ce
[ 5387.555403] Call Trace:
[ 5387.555409]  [<ffffffff813f9513>] dump_stack+0x63/0x90
[ 5387.555423]  [<ffffffff81081322>] warn_slowpath_common+0x82/0xc0
[ 5387.555425]  [<ffffffff8108146a>] warn_slowpath_null+0x1a/0x20
[ 5387.555442]  [<ffffffffc129e35a>] rtw_chk_start_clnt_join+0x77/0xa1 [rtl8812au]
[ 5387.555458]  [<ffffffffc129e615>] join_cmd_hdl+0x291/0x334 [rtl8812au]
[ 5387.555470]  [<ffffffffc127c9ee>] rtw_cmd_thread+0x287/0x3a1 [rtl8812au]
[ 5387.555483]  [<ffffffffc127c767>] ? rtw_stop_cmd_thread+0x54/0x54 [rtl8812au]
[ 5387.555485]  [<ffffffff810a0c25>] kthread+0xe5/0x100
[ 5387.555487]  [<ffffffff810a0b40>] ? kthread_create_on_node+0x1e0/0x1e0
[ 5387.555490]  [<ffffffff81840f0f>] ret_from_fork+0x3f/0x70
[ 5387.555492]  [<ffffffff810a0b40>] ? kthread_create_on_node+0x1e0/0x1e0
[ 5387.555493] ---[ end trace 323c9a875a84fbd4 ]---
[ 5388.258429] RTL871X: rtw_set_802_11_connect(wlan0)  fw_state=0x00000008
[ 5388.492068] RTL871X: start auth
[ 5388.493561] RTL871X: auth success, start assoc
[ 5388.495578] RTL871X: rtw_cfg80211_indicate_connect(wlan0) BSS not found !!
[ 5388.495587] RTL871X: assoc success
[ 5388.498693] RTL871X: send eapol packet
[ 5388.502998] RTL871X: send eapol packet
`
Not sure what is going on.

Maksym

Throne3d, 2017-08-01:
I'm getting a similar error (I'm not sure if it's the same – I'm not particularly used to deciphering this) on Arch Linux. According to the dmesg logs, it seems to be happening every 120 seconds.

```
[26015.045718] RTL871X: rtw_set_802_11_connect(wlp0s19f2u5)  fw_state=0x00000009
[26015.056052] RTL871X: ERROR rtw_chk_start_clnt_join(wlp0s19f2u5) cur_ch:0
[26015.056087] ------------[ cut here ]------------
[26015.056116] WARNING: CPU: 0 PID: 372 at /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_mlme_ext.c:13647 rtw_chk_start_clnt_join+0x34/0x8a [rtl8812au]
[26015.056117] Modules linked in: btrfs xor raid6_pq ufs hfsplus hfs minix ntfs vfat msdos fat jfs xfs libcrc32c crc32c_generic dm_mod dax fuse snd_hda_codec_realtek snd_hda_codec_generic rtl8812au(O) cfg80211 eeepc_wmi snd_hda_codec_hdmi asus_wmi ppdev sparse_keymap rfkill edac_mce_amd evdev kvm_amd input_leds led_class joydev mousedev mac_hid kvm snd_hda_intel snd_hda_codec snd_hda_core snd_hwdep snd_pcm irqbypass crct10dif_pclmul crc32_pclmul crc32c_intel r8169 snd_timer ghash_clmulni_intel pcbc aesni_intel aes_x86_64 crypto_simd glue_helper mii cryptd k10temp snd button tpm_infineon sp5100_tco i2c_piix4 parport_pc parport soundcore wmi shpchp video acpi_cpufreq tpm_tis tpm_tis_core tpm sch_fq_codel sg ip_tables x_tables ext4 crc16 jbd2 fscrypto mbcache hid_roccat_kovaplus hid_roccat hid_roccat_common
[26015.056158]  sr_mod cdrom sd_mod hid_generic uas usbhid usb_storage hid ahci ohci_pci ohci_hcd libahci xhci_pci ehci_pci xhci_hcd libata ehci_hcd usbcore scsi_mod usb_common serio radeon amdgpu i2c_algo_bit drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops ttm drm [last unloaded: pcspkr]
[26015.056175] CPU: 0 PID: 372 Comm: RTW_CMD_THREAD Tainted: G        W  O    4.12.3-1-ARCH #1
[26015.056176] Hardware name: System manufacturer System Product Name/A88XM-PLUS, BIOS 2502 09/17/2015
[26015.056177] task: ffff8f4a4b90e3c0 task.stack: ffffba30820ac000
[26015.056189] RIP: 0010:rtw_chk_start_clnt_join+0x34/0x8a [rtl8812au]
[26015.056189] RSP: 0018:ffffba30820afe50 EFLAGS: 00010282
[26015.056191] RAX: 000000000000003c RBX: ffffba3081f91000 RCX: 0000000000000000
[26015.056192] RDX: 0000000000000000 RSI: ffff8f4a5ec0dcc8 RDI: ffff8f4a5ec0dcc8
[26015.056192] RBP: ffffba30820afe50 R08: 0000000000001764 R09: 0000000000000004
[26015.056193] R10: ffffba30820afe60 R11: 0000000000000001 R12: ffffba3081f9225a
[26015.056194] R13: 0000000000000022 R14: ffffba3081f9224e R15: 0000000000000001
[26015.056195] FS:  0000000000000000(0000) GS:ffff8f4a5ec00000(0000) knlGS:0000000000000000
[26015.056196] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[26015.056197] CR2: 00002c8569d22628 CR3: 0000000258450000 CR4: 00000000000406f0
[26015.056198] Call Trace:
[26015.056212]  join_cmd_hdl+0x2a5/0x34c [rtl8812au]
[26015.056221]  rtw_cmd_thread+0x233/0x3f1 [rtl8812au]
[26015.056224]  kthread+0x125/0x140
[26015.056233]  ? rtw_stop_cmd_thread+0x57/0x57 [rtl8812au]
[26015.056234]  ? kthread_create_on_node+0x70/0x70
[26015.056238]  ret_from_fork+0x25/0x30
[26015.056239] Code: 0f 94 c0 48 85 c9 0f 94 c0 41 08 c0 75 29 48 85 f6 74 24 0f b6 87 8a 0a 00 00 84 c0 75 45 55 48 89 e5 83 3d 3c 73 08 00 04 77 17 <0f> ff b8 00 00 00 00 0f b6 c0 5d c3 0f ff b8 00 00 00 00 0f b6 
[26015.056263] ---[ end trace 4f17fdab5e51dc12 ]---
[26015.790581] RTL871X: rtw_set_802_11_connect(wlp0s19f2u5)  fw_state=0x00000008
[26015.877098] RTL871X: start auth
[26015.883092] RTL871X: auth success, start assoc
[26015.886856] RTL871X: rtw_cfg80211_indicate_connect(wlp0s19f2u5) BSS not found !!
[26015.886864] RTL871X: assoc success
[26015.890283] RTL871X: send eapol packet
[26015.897345] RTL871X: send eapol packet
[26015.898069] RTL871X: set pairwise key camid:4, addr:40:c7:29:1f:53:a8, kid:0, type:AES
[26015.903059] RTL871X: set group key camid:5, addr:40:c7:29:1f:53:a8, kid:1, type:AES
```

This is having built with 0292aa694d85e8f9120479604e98693f20d5da02.

Possibly relevant information:

```
# lsusb
Bus 002 Device 003: ID 0bda:a811 Realtek Semiconductor Corp. RTL8811AU 802.11a/b/g/n/ac WLAN Adapter
Bus 002 …

# lsb_release -a -s
1.4 Arch "Arch Linux" rolling n/a

# uname -mrs
Linux 4.12.3-1-ARCH x86_64
```

I also tried pinging `8.8.8.8` for a while, to notice when my internet cut out. It involves a lot of 'Network is unreachable' blocks and the output repeats approximately like this:

```
$ ping 8.8.8.8
PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.
64 bytes from 8.8.8.8: icmp_seq=1 ttl=54 time=22.3 ms
64 bytes from 8.8.8.8: icmp_seq=2 ttl=54 time=19.8 ms
64 bytes from 8.8.8.8: icmp_seq=3 ttl=54 time=19.5 ms
64 bytes from 8.8.8.8: icmp_seq=4 ttl=54 time=19.6 ms
…
64 bytes from 8.8.8.8: icmp_seq=9570 ttl=54 time=19.5 ms
64 bytes from 8.8.8.8: icmp_seq=9571 ttl=54 time=19.5 ms
ping: sendmsg: Network is unreachable
ping: sendmsg: Network is unreachable
ping: sendmsg: Network is unreachable
ping: sendmsg: Network is unreachable
ping: sendmsg: Network is unreachable
64 bytes from 8.8.8.8: icmp_seq=9581 ttl=54 time=19.8 ms
64 bytes from 8.8.8.8: icmp_seq=9582 ttl=54 time=22.9 ms
…
64 bytes from 8.8.8.8: icmp_seq=11057 ttl=54 time=19.2 ms
64 bytes from 8.8.8.8: icmp_seq=11058 ttl=54 time=19.9 ms
64 bytes from 8.8.8.8: icmp_seq=11059 ttl=54 time=19.8 ms
64 bytes from 8.8.8.8: icmp_seq=11060 ttl=54 time=19.9 ms
^C
--- 8.8.8.8 ping statistics ---
11060 packets transmitted, 10465 received, 5% packet loss, time 11080981ms
rtt min/avg/max/mdev = 18.575/31.876/6883.389/151.056 ms, pipe 7
```

(I'm not 100% sure this is the right driver – it seems to *mostly work* so I'd expect I haven't gone horribly wrong, and poking around online suggested I try one by this name and then this was the most up to date version I could find.)

Is there any more information I could provide that might be useful in tracking this down?

Thanks.

jcjordyn130, 2017-09-23:
well I can tell from the source code it's because cur_ch is 0.... 

        if (cur_ch == 0) {
                connect_allow = _FALSE;
                DBG_871X_LEVEL(_drv_err_, FUNC_ADPT_FMT" cur_ch:%u\n"
                               , FUNC_ADPT_ARG(padapter), cur_ch);
                rtw_warn_on(1); // This is the line that causes the oops/page fault.
                goto exit;
        }

jcjordyn130, 2017-09-23:
I commented out lines 13643 to 13649 in the file rtw_mlme_ext.c, and it seems to stop the driver from resetting every 120 seconds (even though it doesn't fix the issue, it just masks the reset).

maksyms, 2017-09-23:
Thanks for that. Real question is why cur_ch is 0.

Also, rtw_warn_on is just a define to WARN_ON on Linux, which in turn is supposed to log things into dmesg log and that's it. So your commenting out may have just prevented that logging, but not the underlying issue.

Mind you, I have no idea why cur_ch would be 0 at that stage, apart from memory corruption.

closed 2020-04-13 by maksyms