abperiasamy/rtl8812AU_8821AU_linux #216: Build Error Kernel 4.11.9

lakshman5876, 2017-07-06:
Build error on Kernel upgrade to 4.11.9 from 4.11.7.
```
[lakshman@archmain ~/bin/rtl8812AU_abperiasamy]$ sudo make -f Makefile.dkms install
make clean
make[1]: Entering directory '/home/lakshman/bin/rtl8812AU_abperiasamy'
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.11.9-1-ARCH/build M=/home/lakshman/bin/rtl8812AU_abperiasamy clean
make[2]: Entering directory '/usr/lib/modules/4.11.9-1-ARCH/build'
make[2]: Leaving directory '/usr/lib/modules/4.11.9-1-ARCH/build'
make[1]: Leaving directory '/home/lakshman/bin/rtl8812AU_abperiasamy'
mkdir -p '/usr/src/rtl8812au-4.3.14'
cp -r dkms.conf Kconfig Makefile.dkms Makefile platform core hal include os_dep '/usr/src/rtl8812au-4.3.14'
cp Makefile '/usr/src/rtl8812au-4.3.14/Makefile'
sed 's/#MODULE_VERSION#/4.3.14/' dkms.conf > '/usr/src/rtl8812au-4.3.14/dkms.conf'
dkms add -m rtl8812au -v 4.3.14 2>/dev/null || true

Creating symlink /var/lib/dkms/rtl8812au/4.3.14/source ->
                 /usr/src/rtl8812au-4.3.14

DKMS: add completed.
dkms build -m rtl8812au -v 4.3.14

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j8 KERNELRELEASE=4.11.9-1-ARCH KVER=4.11.9-1-ARCH....(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.11.9-1-ARCH (x86_64)
Consult /var/lib/dkms/rtl8812au/4.3.14/build/make.log for more information.
make: *** [Makefile.dkms:19: build] Error 10
```

lakshman5876, 2017-07-06:
**/var/lib/dkms/rtl8812au/4.3.14/build/make.log

```
DKMS make.log for rtl8812au-4.3.14 for kernel 4.11.9-1-ARCH (x86_64)
Thu Jul  6 13:59:45 IST 2017
make[1]: Entering directory '/var/lib/dkms/rtl8812au/4.3.14/build'
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.11.9-1-ARCH/build M=/var/lib/dkms/rtl8812au/4.3.14/build  modules
make[2]: Entering directory '/usr/lib/modules/4.11.9-1-ARCH/build'
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/core/efuse/rtw_efuse.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/os_dep/osdep_service.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/os_dep/linux/os_intfs.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/os_dep/linux/usb_intf.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/os_dep/linux/usb_ops_linux.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/os_dep/linux/ioctl_linux.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/os_dep/linux/xmit_linux.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/os_dep/linux/mlme_linux.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/os_dep/linux/recv_linux.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/os_dep/linux/wifi_regd.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/os_dep/linux/rtw_android.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/os_dep/linux/rtw_proc.o
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/hal/hal_intf.o
/var/lib/dkms/rtl8812au/4.3.14/build/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_add_monitor_if’:
/var/lib/dkms/rtl8812au/4.3.14/build/os_dep/linux/ioctl_cfg80211.c:3654:12: error: ‘struct net_device’ has no member named ‘destructor’; did you mean ‘priv_destructor’?
  mon_ndev->destructor = rtw_ndev_destructor;
            ^~~~~~~~~~
            priv_destructor
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/hal/hal_com.o
make[3]: *** [scripts/Makefile.build:294: /var/lib/dkms/rtl8812au/4.3.14/build/os_dep/linux/ioctl_cfg80211.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [Makefile:1492: _module_/var/lib/dkms/rtl8812au/4.3.14/build] Error 2
make[2]: Leaving directory '/usr/lib/modules/4.11.9-1-ARCH/build'
make[1]: *** [Makefile:1584: modules] Error 2
make[1]: Leaving directory '/var/lib/dkms/rtl8812au/4.3.14/build'
```

lakshman5876, 2017-07-06:
Solved by using the suggestion in make.log above

`/var/lib/dkms/rtl8812au/4.3.14/build/os_dep/linux/ioctl_cfg80211.c:3654:12: error: ‘struct net_device’ has no member named ‘destructor’; did you mean ‘priv_destructor’?
`

ptpt52, 2017-07-07:
seems from kernel >= 4.11.9  the `struct net_device` changed

ptpt52, 2017-07-07:
shall be fixed in https://github.com/abperiasamy/rtl8812AU_8821AU_linux/pull/217

closed 2017-07-09 by ptpt52

tsoernes, 2017-07-24:
I still get this on Fedora 26 4.11.9-300.
On master as well as v5.1.5: 
```
/home/torstein/progs/rtl8812au-astsam/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_add_monitor_if’:
/home/torstein/progs/rtl8812au-astsam/os_dep/linux/ioctl_cfg80211.c:3954:12: error: ‘struct net_device’ has no member named ‘destructor’; did you mean ‘priv_destructor’?
  mon_ndev->destructor = rtw_ndev_destructor;
```