abperiasamy/rtl8812AU_8821AU_linux #219: Add CSL - AC450 to supported hardware

ManuGithubSteam, 2017-07-10:
I'm happy to report, that the this driver works with: 

CSL - AC450 WLAN Dual Band Stick | 433 MBit/s | 2,4 GHz / 5 GHz 

it also work with the newest Arch Kernel 4.9.11 :-)

Thanks a lot for your support!

Amazon Link: https://www.amazon.de/CSL-WLAN-Standards-Protected-Infrastruktur-Netz-Status-LED/dp/B01E2J5XDI/ref=sr_1_2?ie=UTF8&qid=1499667949&sr=8-2&keywords=csl+5ghz+wlan

iam-TJ, 2017-07-29:
I have problems with the driver and the CSL AC450; see my issue #225 - does your device have the same USB ID ?

ManuGithubSteam, 2017-07-29:
Hi,
USB ID is not displayed in lsusb for me. However i do not have the issues with the public ssid s not showing up or hidden ssids showing up. It worked flawless till now but the most recent kernel upgrade did break it again :-(

iam-TJ, 2017-07-29:
If "lsusb" doesn't display the ID then the device (or the hub/socket it is connected to) is faulty. Even without a (device) driver USB device IDs are listed. Try alternative USB ports, or ensure another USB device works from that same port, to try to narrow down the cause.

iam-TJ, 2017-07-31:
Adding a note of interest; I noticed that the **CSL AC450** dual-band device (802.11ac 1T1R) I have (which has a lenticular case which can be seen through under bright light) the PCB inside has the marking "**WN738N-V2**" which is *very* much like the TP-Link model designations. It may be an indication this device (PCB) is made my TP-Link for OEM and other brand's.