abperiasamy/rtl8812AU_8821AU_linux #220: TP-Link Archer T4U V2 showing twice

vinyanalista, 2017-07-13:
First of all, let me thank you for the great driver that makes it possible for me to use my [TP-Link Archer T4U V2](http://www.tp-link.com/en/products/details/cat-11_Archer-T4U.html) ([2357:010d](https://wikidevi.com/wiki/TP-LINK_Archer_T4U_v2)) device on Linux.

I use [openSUSE Leap 42.2](https://kamarada.github.io/en/2016/11/16/optimal-release-for-linux-professionals-arrives-with-opensuse-leap-42-2/) on a laptop. Often maintainers update the [Linux kernel](https://www.kernel.org/), always on the 4.4.x longterm branch.

When I reboot with a new kernel, I take that opportunity to update the rtl8812au driver: I do `git pull` (connected using the laptop's builtin Wi-Fi adapter) and run a script I created just for that

```
#!/bin/bash
set -ex
dkms remove rtl8812au/4.3.14 --all
make -f Makefile.dkms install
lsmod | grep rtl8812au
sleep 5s
modprobe rtl8812au
lsmod | grep rtl8812au
```

But this time when I updated both the kernel and the rtl8812au driver, Wi-Fi stopped working.

When I used [kernel-default](https://software.opensuse.org/package/kernel-default) version 4.4.74 with the latest rtl8812au driver (abperiasamy/rtl8812AU_8821AU_linux@2c6399ff11810055d2cac2aaa59f07f37245e20d by now), my TP-Link Archer T4U appeared twice if I did a `ip -c a`. Both instances connected to the same Wi-Fi network (same SSID), both received IP addresses, but connection was very unstable so I was not able to navigate.

I downgraded the rtl8812au driver to a version I knew was working for sure: abperiasamy/rtl8812AU_8821AU_linux@b018b446fb0729fa659e7eead475d6c13f192642 was working for me by [23-Apr-2017](https://forums.opensuse.org/showthread.php/524410-Qualcomm-Atheros-Bluetooth-4-0-(04CA-3014)-is-not-detected?p=2820716#post2820716). Now, my TP-Link Archer T4U works as expected:

```
# lsusb
Bus 003 Device 002: ID 8087:8001 Intel Corp. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 0424:5537 Standard Microsystems Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 008: ID 0bda:0129 Realtek Semiconductor Corp. RTS5129 Card Reader Controller
Bus 001 Device 007: ID 1bcf:2c81 Sunplus Innovation Technology Inc. 
Bus 001 Device 004: ID 045e:0745 Microsoft Corp. Nano Transceiver v1.0 for Bluetooth
Bus 001 Device 005: ID 2357:010d  
Bus 001 Device 003: ID 0424:2137 Standard Microsystems Corp. 
Bus 001 Device 002: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
# ip -c a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether 1c:39:47:0c:c4:1d brd ff:ff:ff:ff:ff:ff
3: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether ac:e0:10:7a:7b:19 brd ff:ff:ff:ff:ff:ff
4: wlan2: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 18:d6:c7:08:e4:a6 brd ff:ff:ff:ff:ff:ff
    inet 10.0.0.60/24 brd 10.0.0.255 scope global dynamic wlan2
       valid_lft 364sec preferred_lft 364sec
```

Right now, it shows as `wlan2`, MAC address `18:d6:c7:08:e4:a6`.

I think that is a bug introduced between abperiasamy/rtl8812AU_8821AU_linux@b018b446fb0729fa659e7eead475d6c13f192642 (exclusive) and abperiasamy/rtl8812AU_8821AU_linux@2c6399ff11810055d2cac2aaa59f07f37245e20d (inclusive). I'm going to do `git checkout master` to reproduce what hapens and comment here.

vinyanalista, 2017-07-13:
I'm now at abperiasamy/rtl8812AU_8821AU_linux@2c6399ff11810055d2cac2aaa59f07f37245e20d

Besides building and installing the module, I needed to reboot my laptop.

Now see the result:

```
# lsusb
Bus 003 Device 002: ID 8087:8001 Intel Corp. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 0424:5537 Standard Microsystems Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 008: ID 0bda:0129 Realtek Semiconductor Corp. RTS5129 Card Reader Controller
Bus 001 Device 007: ID 1bcf:2c81 Sunplus Innovation Technology Inc. 
Bus 001 Device 004: ID 045e:0745 Microsoft Corp. Nano Transceiver v1.0 for Bluetooth
Bus 001 Device 005: ID 2357:010d  
Bus 001 Device 003: ID 0424:2137 Standard Microsystems Corp. 
Bus 001 Device 002: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
# ip -c a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether 1c:39:47:0c:c4:1d brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether ac:e0:10:7a:7b:19 brd ff:ff:ff:ff:ff:ff
    inet 10.0.0.10/24 brd 10.0.0.255 scope global dynamic wlan0
       valid_lft 561sec preferred_lft 561sec
4: rename4: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 18:d6:c7:08:e4:a6 brd ff:ff:ff:ff:ff:ff
    inet 10.0.0.60/24 brd 10.0.0.255 scope global dynamic rename4
       valid_lft 456sec preferred_lft 456sec
5: wlan2: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 1a:d6:c7:08:e4:a6 brd ff:ff:ff:ff:ff:ff
    inet 10.0.0.251/24 brd 10.0.0.255 scope global dynamic wlan2
       valid_lft 439sec preferred_lft 439sec
```

`wlan0` is my laptop's builtin Wi-Fi adapter, a Qualcomm Atheros QCA9565 / AR9565 Wireless Network Adapter ([168c:0036](https://wikidevi.com/wiki/Qualcomm_Atheros_QCWB335)). Both `rename4` (MAC address `18:d6:c7:08:e4:a6`) and `wlan2` (MAC address `1a:d6:c7:08:e4:a6`) refer to my TP-Link Archer T4U.

Curiously, both MAC addresses are equal, except by the first byte (`18` in `rename4` and `1a` in `wlan2`). The second one, however, seems invalid: [MAC Vendors](https://macvendors.com/) returns me `No Vendor` for `1a:d6:c7:08:e4:a6`.

hschletz, 2017-07-16:
Seems to be related to CONFIG_CONCURRENT_MODE enabled in https://github.com/abperiasamy/rtl8812AU_8821AU_linux/commit/2c6399ff11810055d2cac2aaa59f07f37245e20d. I see the same issue with my D-Link DWA-182. MAC addresses are 10:... and 12:... (difference is in Bit 1 of the first byte, just like yours).

Only 1 of these interfaces appears to be functional. Initially, I was unable to connect. Then I reverted the change (disabling CONFIG_CONCURRENT_MODE), resulting in only 1 interface, and it worked. Curiously, after re-enabling, I get 2 interfaces again, but I am able to connect now. Maybe NetworkManager picked the wrong interface.

hschletz, 2017-07-16:
After the next reboot, both interfaces stopped working again. I reverted Makefile to its previous state to keep the driver usable for me.

iam-TJ, 2017-07-29:
I suspect CONFIG_CONCURRENT_MODE is there to support using WDS (a.k.a. Miracast) for display streaming (which uses peer-to-peer I believe) as well as connection to a regular AP at the same time, so as to maintain a WiFi network when using Miracast.

Possibly the secondary interface should have capability flags indicating only it only supports P2P modes (see "iw phy" output) ?

hschletz, 2017-07-31:
ed2b8009bda848e83f8b671ba9fed99baeb30fa4 disabled CONFIG_CONCURRENT_MODE again. Latest code works out of the box for me.

vinyanalista, 2017-08-22:
I'm now at abperiasamy/rtl8812AU_8821AU_linux@0292aa6 (latest commit by now).

I ran my script to build and load the kernel driver. At a first glance, now it works out of the box.

Right now, my laptop's builtin Wi-Fi adapter shows as `wlan0` (MAC address `ac:e0:10:7a:7b:19`) and TP-Link Archer T4U V2 shows as `wlan2` (MAC address `18:d6:c7:08:e4:a6`):

```
# ip -c a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether 1c:39:47:0c:c4:1d brd ff:ff:ff:ff:ff:ff
3: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether ac:e0:10:7a:7b:19 brd ff:ff:ff:ff:ff:ff
4: wlan2: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 18:d6:c7:08:e4:a6 brd ff:ff:ff:ff:ff:ff
    inet 10.0.0.60/24 brd 10.0.0.255 scope global dynamic wlan2
       valid_lft 602sec preferred_lft 602sec
```

I'm using [openSUSE Leap 42.3](https://kamarada.github.io/en/2017/07/26/refresh-of-linux-distribution-continues-leveraging-community-enterprise-benefits/) with Linux kernel 4.4.79:

```
# uname -a
Linux viny-notebook 4.4.79-4-default #1 SMP Thu Aug 3 14:49:17 UTC 2017 (4dc78e3) x86_64 x86_64 x86_64 GNU/Linux
```

vinyanalista, 2017-08-27:
I'm not sure it is working as it should. Even though my TP-Link Archer T4U V2 adapter is not appearing twice anymore, browsing sometimes seems slower than it should be. I'm going to investigate further.

visbs, 2019-08-23:
@iam-TJ it is not working for my p2p using concurrent mode  it gives lot of errors can u give any suggestions are hacks to fix the same.