abperiasamy/rtl8812AU_8821AU_linux #222: Can't compile with fedora 25 kernel 4.11.10

hrumph, 2017-07-18:
Hi, I can't compile this with fedora 25 kernel 4.11.10 (I just upgraded from kernel 4.11.5 and it worked fine with that kernel)

It all seems go horribly wrong when I try to compile __linux/ioctl_cfg80211.c__. This is the error i'm getting:

```
  CC [M]  /home/hrumph/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_cfg80211.o
/home/hrumph/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_indicate_connect’:
/home/hrumph/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_cfg80211.c:784:10: error: variable ‘roam_info’ has initializer but incomplete type
   struct cfg80211_roam_info roam_info = {};
          ^~~~~~~~~~~~~~~~~~
/home/hrumph/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_cfg80211.c:784:29: error: storage size of ‘roam_info’ isn’t known
   struct cfg80211_roam_info roam_info = {};
                             ^~~~~~~~~
In file included from ./include/linux/slab.h:14:0,
                 from /home/hrumph/rtl8812AU_8821AU_linux/include/osdep_service_linux.h:29,
                 from /home/hrumph/rtl8812AU_8821AU_linux/include/osdep_service.h:41,
                 from /home/hrumph/rtl8812AU_8821AU_linux/include/drv_types.h:32,
                 from /home/hrumph/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_cfg80211.c:22:
./include/linux/gfp.h:239:20: error: passing argument 3 of ‘cfg80211_roamed’ makes pointer from integer without a cast [-Werror=int-conversion]
 #define GFP_ATOMIC (__GFP_HIGH|__GFP_ATOMIC|__GFP_KSWAPD_RECLAIM)
                    ^
/home/hrumph/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_cfg80211.c:803:50: note: in expansion of macro ‘GFP_ATOMIC’
   cfg80211_roamed(padapter->pnetdev, &roam_info, GFP_ATOMIC);
                                                  ^~~~~~~~~~
In file included from /home/hrumph/rtl8812AU_8821AU_linux/include/osdep_service_linux.h:88:0,
                 from /home/hrumph/rtl8812AU_8821AU_linux/include/osdep_service.h:41,
                 from /home/hrumph/rtl8812AU_8821AU_linux/include/drv_types.h:32,
                 from /home/hrumph/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_cfg80211.c:22:
./include/net/cfg80211.h:5233:6: note: expected ‘const u8 * {aka const unsigned char *}’ but argument is of type ‘unsigned int’
 void cfg80211_roamed(struct net_device *dev,
      ^~~~~~~~~~~~~~~
/home/hrumph/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_cfg80211.c:803:3: error: too few arguments to function ‘cfg80211_roamed’
   cfg80211_roamed(padapter->pnetdev, &roam_info, GFP_ATOMIC);
   ^~~~~~~~~~~~~~~
In file included from /home/hrumph/rtl8812AU_8821AU_linux/include/osdep_service_linux.h:88:0,
                 from /home/hrumph/rtl8812AU_8821AU_linux/include/osdep_service.h:41,
                 from /home/hrumph/rtl8812AU_8821AU_linux/include/drv_types.h:32,
                 from /home/hrumph/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_cfg80211.c:22:
./include/net/cfg80211.h:5233:6: note: declared here
 void cfg80211_roamed(struct net_device *dev,
      ^~~~~~~~~~~~~~~
/home/hrumph/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_cfg80211.c:784:29: error: unused variable ‘roam_info’ [-Werror=unused-variable]
   struct cfg80211_roam_info roam_info = {};
                             ^~~~~~~~~
cc1: all warnings being treated as errors
scripts/Makefile.build:294: recipe for target '/home/hrumph/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_cfg80211.o' failed
make[2]: *** [/home/hrumph/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_cfg80211.o] Error 1
Makefile:1496: recipe for target '_module_/home/hrumph/rtl8812AU_8821AU_linux' failed
make[1]: *** [_module_/home/hrumph/rtl8812AU_8821AU_linux] Error 2
make[1]: Leaving directory '/usr/src/kernels/4.11.10-200.fc25.x86_64'
```

MilhouseVH, 2017-07-18:
Pretty sure the kernel version checks in the following lines are all wrong - the current code only works so long as there's no more 4.11.y releases after 4.11.9... hence why it doesn't work with 4.11.10.

1. https://github.com/abperiasamy/rtl8812AU_8821AU_linux/pull/217/files#diff-c39a0cc7e3701496c432e2cfe95097baR783
should be `>= KERNEL_VERSION(4, 12, 0)` not `> KERNEL_VERSION(4, 11, 9)`:

2. https://github.com/abperiasamy/rtl8812AU_8821AU_linux/pull/217/files#diff-c39a0cc7e3701496c432e2cfe95097baR796
should be `>= KERNEL_VERSION(4, 12, 0)` not `> KERNEL_VERSION(4, 11, 9)`:

3. https://github.com/abperiasamy/rtl8812AU_8821AU_linux/pull/217/files#diff-c39a0cc7e3701496c432e2cfe95097baR5928
should be `< KERNEL_VERSION(4, 12, 0)` not `< KERNEL_VERSION(4, 11, 9)`:

Someone will need to push a fix for this repo.

ptpt52, 2017-07-22:
fixed in HEAD

closed 2017-07-22 by ptpt52

hrumph, 2017-07-22:
It compiles alright but it doesn't work with NetworkManager in Fedora 25 now (AFACS). I'll create another issue later with details.

ptpt52, 2017-07-22:
@hrumph 
disable CONFIG_CONCURRENT_MODE in Makefile and try it again?

`
diff --git a/Makefile b/Makefile
index 022c193..1499f1d 100644
--- a/Makefile
+++ b/Makefile
@@ -870,7 +870,7 @@ endif
 
 ifeq ($(CONFIG_PLATFORM_I386_PC), y)
 EXTRA_CFLAGS += -DCONFIG_LITTLE_ENDIAN
-EXTRA_CFLAGS += -DCONFIG_CONCURRENT_MODE
+#EXTRA_CFLAGS += -DCONFIG_CONCURRENT_MODE
 EXTRA_CFLAGS += -DCONFIG_IOCTL_CFG80211 -DRTW_USE_CFG80211_STA_EVENT
 #EXTRA_CFLAGS += -DCONFIG_P2P_IPS
 SUBARCH := $(shell uname -m | sed -e s/i.86/i386/)
`