abperiasamy/rtl8812AU_8821AU_linux #223: Does not work with NetworkManager in Fedora 25

hrumph, 2017-07-22:
Hi,
I got it to compilie (See issue #222). @ptp52, I did try making after disabling CONFIG_CONCURRENT_MODE, without any success.

When the NetworkManager service is started it's as though it (something, I don't know what) is trying to create two devices,

1. wlp0s18f2u4
2. wlan0

I think that only the wlp0s18f2u4 device should be created. Also the system log is spammed furiously. The error messages look like this.

When the NetworkManager service is started the log gets spammed furiously. This is what I get:
```
ul 22 19:29:29 localhost.localdomain systemd[1]: Starting Network Manager...
Jul 22 19:29:29 localhost.localdomain NetworkManager[8215]: <info>  [1500766169.8472] NetworkManager (version 1.4.4-5.fc25) is starting...
Jul 22 19:29:29 localhost.localdomain NetworkManager[8215]: <info>  [1500766169.8473] Read config: /etc/NetworkManager/NetworkManager.conf (lib: 20-connectivity-fedora.conf)
Jul 22 19:29:29 localhost.localdomain NetworkManager[8215]: <info>  [1500766169.8533] manager[0x55670544a0c0]: monitoring kernel firmware directory '/lib/firmware'.
Jul 22 19:29:29 localhost.localdomain NetworkManager[8215]: <info>  [1500766169.8557] dns-mgr[0x556705427aa0]: init: dns=default, rc-manager=symlink
Jul 22 19:29:29 localhost.localdomain NetworkManager[8215]: <info>  [1500766169.8573] rfkill8: found WiFi radio killswitch (at /sys/devices/pci0000:00/0000:00:12.2/usb1/1-4/1-4:1.0/ieee80211/phy0/rfkill8) (driver rtl8812au)
Jul 22 19:29:29 localhost.localdomain NetworkManager[8215]: <info>  [1500766169.8576] rfkill9: found WiFi radio killswitch (at /sys/devices/pci0000:00/0000:00:12.2/usb1/1-4/1-4:1.0/ieee80211/phy1/rfkill9) (driver rtl8812au)
Jul 22 19:29:29 localhost.localdomain NetworkManager[8215]: <info>  [1500766169.8580] manager[0x55670544a0c0]: WiFi hardware radio set enabled
Jul 22 19:29:29 localhost.localdomain NetworkManager[8215]: <info>  [1500766169.8580] manager[0x55670544a0c0]: WWAN hardware radio set enabled
Jul 22 19:29:29 localhost.localdomain systemd[1]: Started Network Manager.
Jul 22 19:29:29 localhost.localdomain audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=NetworkManager comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Jul 22 19:29:29 localhost.localdomain dbus-daemon[785]: [system] Activating via systemd: service name='org.freedesktop.nm_dispatcher' unit='dbus-org.freedesktop.nm-dispatcher.service' requested by ':1.118' (uid=0 pid=8215 comm="/usr/sbin/NetworkManager --no-daemon " label="system_u:system_r:NetworkManager_t:s0")
Jul 22 19:29:29 localhost.localdomain systemd[1]: Starting Network Manager Script Dispatcher Service...
Jul 22 19:29:29 localhost.localdomain dbus-daemon[785]: [system] Successfully activated service 'org.freedesktop.nm_dispatcher'
Jul 22 19:29:29 localhost.localdomain audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=NetworkManager-dispatcher comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Jul 22 19:29:29 localhost.localdomain systemd[1]: Started Network Manager Script Dispatcher Service.
Jul 22 19:29:29 localhost.localdomain NetworkManager[8215]: <info>  [1500766169.8790] settings: loaded plugin ifcfg-rh: (c) 2007 - 2015 Red Hat, Inc.  To report bugs please use the NetworkManager mailing list. (/usr/lib64/NetworkManager/libnm-settings-plugin-ifcfg-rh.so)
Jul 22 19:29:29 localhost.localdomain NetworkManager[8215]: <info>  [1500766169.8792] settings: loaded plugin iBFT: (c) 2014 Red Hat, Inc.  To report bugs please use the NetworkManager mailing list. (/usr/lib64/NetworkManager/libnm-settings-plugin-ibft.so)
Jul 22 19:29:29 localhost.localdomain NetworkManager[8215]: <info>  [1500766169.8793] settings: loaded plugin keyfile: (c) 2007 - 2015 Red Hat, Inc.  To report bugs please use the NetworkManager mailing list.
Jul 22 19:29:29 localhost.localdomain NetworkManager[8215]: <info>  [1500766169.8808] ifcfg-rh: new connection /etc/sysconfig/network-scripts/ifcfg-Wired_connection_2 (b1a72e80-c268-484c-8761-1ddfce1fcd2c,"Wired connection 2")
Jul 22 19:29:29 localhost.localdomain NetworkManager[8215]: <info>  [1500766169.8819] ifcfg-rh: new connection /etc/sysconfig/network-scripts/ifcfg-doggins (2a41723b-503a-4010-b915-d657b07ed312,"doggins")
Jul 22 19:29:29 localhost.localdomain NetworkManager[8215]: <info>  [1500766169.8827] ifcfg-rh: new connection /etc/sysconfig/network-scripts/ifcfg-enp3s0 (414ee9e8-1401-356c-b978-0c770366566d,"enp3s0")
Jul 22 19:29:29 localhost.localdomain NetworkManager[8215]: <info>  [1500766169.8956] keyfile: new connection /etc/NetworkManager/system-connections/us-ca2.nordvpn.com.tcp443_ (3b527459-c6a0-47a2-ba98-d56880d111c9,"us-ca2.nordvpn.com.tcp443")
Jul 22 19:29:29 localhost.localdomain dbus-daemon[785]: [system] Activating via systemd: service name='org.freedesktop.hostname1' unit='dbus-org.freedesktop.hostname1.service' requested by ':1.118' (uid=0 pid=8215 comm="/usr/sbin/NetworkManager --no-daemon " label="system_u:system_r:NetworkManager_t:s0")
Jul 22 19:29:29 localhost.localdomain systemd[1]: Starting Hostname Service...
Jul 22 19:29:30 localhost.localdomain dbus-daemon[785]: [system] Successfully activated service 'org.freedesktop.hostname1'
Jul 22 19:29:30 localhost.localdomain systemd[1]: Started Hostname Service.
Jul 22 19:29:30 localhost.localdomain audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=systemd-hostnamed comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Jul 22 19:29:30 localhost.localdomain NetworkManager[8215]: <info>  [1500766170.0227] settings: hostname: using hostnamed
Jul 22 19:29:30 localhost.localdomain NetworkManager[8215]: <info>  [1500766170.0230] settings: hostname changed from (none) to "localhost.localdomain"
Jul 22 19:29:30 localhost.localdomain NetworkManager[8215]: <info>  [1500766170.0234] dhcp-init: Using DHCP client 'dhclient'
Jul 22 19:29:30 localhost.localdomain NetworkManager[8215]: <info>  [1500766170.0234] manager: WiFi enabled by radio killswitch; enabled by state file
Jul 22 19:29:30 localhost.localdomain nm-dispatcher[8219]: req:1 'hostname': new request (5 scripts)
Jul 22 19:29:30 localhost.localdomain NetworkManager[8215]: <info>  [1500766170.0235] manager: WWAN enabled by radio killswitch; enabled by state file
Jul 22 19:29:30 localhost.localdomain nm-dispatcher[8219]: req:1 'hostname': start running ordered scripts...
Jul 22 19:29:30 localhost.localdomain NetworkManager[8215]: <info>  [1500766170.0235] manager: Networking is enabled by state file
Jul 22 19:29:30 localhost.localdomain NetworkManager[8215]: <info>  [1500766170.0236] Loaded device plugin: NMVxlanFactory (internal)
Jul 22 19:29:30 localhost.localdomain NetworkManager[8215]: <info>  [1500766170.0236] Loaded device plugin: NMVlanFactory (internal)
Jul 22 19:29:30 localhost.localdomain NetworkManager[8215]: <info>  [1500766170.0236] Loaded device plugin: NMVethFactory (internal)
Jul 22 19:29:30 localhost.localdomain NetworkManager[8215]: <info>  [1500766170.0236] Loaded device plugin: NMTunFactory (internal)
Jul 22 19:29:30 localhost.localdomain NetworkManager[8215]: <info>  [1500766170.0237] Loaded device plugin: NMMacvlanFactory (internal)
Jul 22 19:29:30 localhost.localdomain NetworkManager[8215]: <info>  [1500766170.0237] Loaded device plugin: NMIPTunnelFactory (internal)
Jul 22 19:29:30 localhost.localdomain NetworkManager[8215]: <info>  [1500766170.0237] Loaded device plugin: NMInfinibandFactory (internal)
Jul 22 19:29:30 localhost.localdomain NetworkManager[8215]: <info>  [1500766170.0237] Loaded device plugin: NMEthernetFactory (internal)
Jul 22 19:29:30 localhost.localdomain NetworkManager[8215]: <info>  [1500766170.0238] Loaded device plugin: NMBridgeFactory (internal)
Jul 22 19:29:30 localhost.localdomain NetworkManager[8215]: <info>  [1500766170.0238] Loaded device plugin: NMBondFactory (internal)
Jul 22 19:29:30 localhost.localdomain NetworkManager[8215]: <info>  [1500766170.0247] Loaded device plugin: NMAtmManager (/usr/lib64/NetworkManager/libnm-device-plugin-adsl.so)
Jul 22 19:29:30 localhost.localdomain NetworkManager[8215]: <info>  [1500766170.0264] Loaded device plugin: NMTeamFactory (/usr/lib64/NetworkManager/libnm-device-plugin-team.so)
[... cut ...]
Jul 22 19:29:33 localhost.localdomain NetworkManager[8215]: <info>  [1500766173.3998] device (wlp0s18f2u4): supplicant interface state: disabled -> inactive
Jul 22 19:29:33 localhost.localdomain NetworkManager[8215]: <warn>  [1500766173.7273] device (wlp0s18f2u4): set-hw-addr: new MAC address 6A:E8:B0:AE:24:27 not successfully set (scanning)
Jul 22 19:29:33 localhost.localdomain kernel: IPv6: ADDRCONF(NETDEV_UP): wlp0s18f2u4: link is not ready
Jul 22 19:29:33 localhost.localdomain NetworkManager[8215]: <info>  [1500766173.7280] device (wlan0): supplicant interface state: inactive -> disabled
```

ptpt52, 2017-07-23:
make sure you reload/reinstall the kmod rtl8812au.ko

ptpt52, 2017-07-23:
maybe it is something wrong with NetworkManager

hrumph, 2017-07-23:
Taking a hint from https://bugzilla.redhat.com/show_bug.cgi?id=1382741#c10 I added 
```
[device]
wifi.scan-rand-mac-address=no
```

to __/etc/NetworkManager/NetworkManager.conf__. Reading the same bug report it appears that this workaround may be done automatically in more recent versions of NetworkManager (see https://cgit.freedesktop.org/NetworkManager/NetworkManager/commit/?id=023417292339e34dd07e5b902576c1472ff0c3dd).

Using this trick I did get it to work, but there's still two fully equivalent devices, which are

1. wlp0s18f2u4
2. wlan0 or wlan1 (Network manager can't even make up it's mind about which of the two it is when I boot).

The devices always have the same set of connections, so you can use either one of the two and I can't figure out a way to get rid of wlan0/wlan1. Also when I boot, if I have my PC join the wireless network automatically, it seems to randomly choose to connect on either one of the two devices. That means that when setting up my firewalling, I have to triplicate my rules. I have to have the same rule for each of wlp0s18f2u4, wlan0 and wlan1. (Routing is ok though, because routing is done per connection in network manager).

hrumph, 2017-07-24:
I have taken some further measures which remove **wlan0**/**wlan1** from the picture so I don't have to triplicate my firewall rules any more. I have added the following two identical files.

1. /etc/sysconfig/network-scripts/ifcfg-wlan0
2. /etc/sysconfig/network-scripts/ifcfg-wlan1

The content of the files is as follows:

```
NM_CONTROLLED=no
HWADDR=70:4D:7B:12:C9:8C
```

So now things are basically back to normal but that's after two kludgey workarounds.

ptpt52, 2017-07-24:
I think it is something wrong with your kmod, you are probably using a CONFIG_CONCURRENT_MODE kmod. 
You should do an clean-up and comment out the CONFIG_CONCURRENT_MODE in Makefile and re-build/re-install the kmod

hrumph, 2017-07-24:
@ptpt52 I think I already tried that. (I may give it one more shot just do double check I guess). Anyway it turns out that I can't identify interfaces with a MAC address because sometimes **wlan*** gets one and sometimes **wlp0s18f2u4** gets the other and the actual choice of which gets which seems random. So instead of having the extra **ifcfg** files, I have added the following to __/etc/NetworkManager/NetworkManager.conf.

```
[keyfile]
unmanaged-devices=interface-name:wlan*
```

hrumph, 2017-07-24:
I'll make two new issues (maybe tomorrow) and close this one since these do appear to be separate.