abperiasamy/rtl8812AU_8821AU_linux #224: Wifi appears twice.

NGTmeaty, 2017-07-28:
two nics appear, i'm using a AC600 

there is only one wifi adapter, and after the latest pull, i seem to have some crazy issues, i ping at around 20ish ms, then randomly spikes to 200ms, this did not happen after the last pull. nothing about my wireless has changes

and there are some issues with each "nic" in network manager

running arch linux

please help, my gaming experience is trash. 
```

64 bytes from : icmp_seq=54 ttl=118 time=53.3 ms
64 bytes from : icmp_seq=55 ttl=118 time=22.1 ms
64 bytes from : icmp_seq=56 ttl=118 time=60.6 ms
64 bytes from : icmp_seq=57 ttl=118 time=40.4 ms
64 bytes from : icmp_seq=58 ttl=118 time=160 ms
64 bytes from : icmp_seq=59 ttl=118 time=35.8 ms
64 bytes from : icmp_seq=60 ttl=118 time=38.6 ms
```

NGTmeaty, 2017-07-28:
#220

ptpt52, 2017-07-28:
I disable CONFIG_CONCURRENT_MODE by default for now.

NGTmeaty, 2017-07-28:
Should that help the issues I have talked about aboe?

NGTmeaty, 2017-07-28:
*above?

NGTmeaty, 2017-07-30:
It does not fix the issues with ping, nor the two adapters.

ptpt52, 2017-07-31:
@NGTmeaty make sure u remove old kmod, and clean reinstall the new one