abperiasamy/rtl8812AU_8821AU_linux #226: Does not work with kernel 4.12.3-1-ARCH

ManuGithubSteam, 2017-07-29:
The device is not recognized anymore....

iam-TJ, 2017-07-29:
What does the kernel log (dmesg) report when the device is inserted? Has the module been built for the newer kernel, and installed?

ManuGithubSteam, 2017-07-29:
Dmesg
: 
[ 5563.215802] usb 2-2: USB disconnect, device number 5
[ 5565.167616] usb 2-2: new high-speed USB device number 6 using xhci_hcd

The module does not build with the new kernel:

make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.12.3-1-ARCH/build M=/home/manuel/rtl8812au  modules
make[1]: Entering directory '/usr/lib/modules/4.12.3-1-ARCH/build'
  CC [M]  /home/manuel/rtl8812au/os_dep/linux/ioctl_cfg80211.o
/home/manuel/rtl8812au/os_dep/linux/ioctl_cfg80211.c:6392:25: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .change_virtual_intf = cfg80211_rtw_change_iface,
                         ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/manuel/rtl8812au/os_dep/linux/ioctl_cfg80211.c:6392:25: note: (near initialization for ‘rtw_cfg80211_ops.change_virtual_intf’)
/home/manuel/rtl8812au/os_dep/linux/ioctl_cfg80211.c:6415:22: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .add_virtual_intf = cfg80211_rtw_add_virtual_intf,
                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/manuel/rtl8812au/os_dep/linux/ioctl_cfg80211.c:6415:22: note: (near initialization for ‘rtw_cfg80211_ops.add_virtual_intf’)
/home/manuel/rtl8812au/os_dep/linux/ioctl_cfg80211.c: In function ‘cfg80211_rtw_get_channel’:
/home/manuel/rtl8812au/os_dep/linux/ioctl_cfg80211.c:4392:18: warning: ‘width’ may be used uninitialized in this function [-Wmaybe-uninitialized]
   chandef->width = width;
   ~~~~~~~~~~~~~~~^~~~~~~
/home/manuel/rtl8812au/os_dep/linux/ioctl_cfg80211.c:4393:25: warning: ‘center_freq’ may be used uninitialized in this function [-Wmaybe-uninitialized]
   chandef->center_freq1 = center_freq;
   ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:303: /home/manuel/rtl8812au/os_dep/linux/ioctl_cfg80211.o] Error 1
make[1]: *** [Makefile:1512: _module_/home/manuel/rtl8812au] Error 2
make[1]: Leaving directory '/usr/lib/modules/4.12.3-1-ARCH/build'
make: *** [Makefile:1857: modules] Error 2

iam-TJ, 2017-07-29:
Are you sure the source directory you're building is from this repository!?

Your report shows:

/home/manuel/rtl8812au/os_dep/linux/ioctl_cfg80211.c:6392:25: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]

which means line 6392 of os_dep/linux/ioctl_cfg80211.c

But on my system that file doesn't contain that many lines!

tj@T300CHI:/usr/src/rtl8812au_8821au-4.3.14.github.abperiasamy$ git pull
Already up-to-date.
tj@T300CHI:/usr/src/rtl8812au_8821au-4.3.14.github.abperiasamy$ wc -l os_dep/linux/ioctl_cfg80211.c
6163 os_dep/linux/ioctl_cfg80211.c

However, the reason is that the kernel network drivers were changed so that these functions don't pass a separate flags argument; instead it is included in the struct vif_params.

$ git describe --contains 818a986e4eb
v4.12-rc1~129^2~155^2~8

$ git show 818a986e4eb
commit 818a986e4ebacea2020622e48c8bc04b7f500d89
Author: Johannes Berg <johannes.berg@intel.com>
Date:   Wed Apr 12 11:23:28 2017 +0200

    cfg80211: move add/change interface monitor flags into params
    
    Instead passing both flags, which can be NULL, and vif_params,
    which are never NULL, move the flags into the vif_params and
    use BIT(0), which is invalid from userspace, to indicate that
    the flags were changed.
    
    While updating all drivers, fix a small bug in wil6210 where
    it was setting the flags to 0 instead of leaving them unchanged.

ManuGithubSteam, 2017-07-30:
You where right. I believe i cloned the wrong repo.

closed 2017-07-30 by ManuGithubSteam

iam-TJ, 2017-07-30:
It's simple to check; you should see:

```
$ git config remote.origin.url
https://github.com/abperiasamy/rtl8812AU_8821AU_linux.git

```