abperiasamy/rtl8812AU_8821AU_linux #229: Scan doesn't see any APs (with kernel 4.13-rc2)

iam-TJ, 2017-07-30:
I've just tried an Ubuntu mainline kernel build of v4.13-rc2 and found that although the module loads correctly the device [0bda:0811] cannot see any APs at all!

Combined with my earlier report #225 it is beginning to look like more recent kernels are in some way incompatible. With kernel 4.5 all APs (public and hidden) are found on both 2.4GHz and 5.xGHz. With kernel 4.10 it isn't seeing public APs on 2.4GHz, and now with 4.13-rc2 it cannot see any APs.

iam-TJ, 2017-07-31:
After a lot of methodical testing across kernel versions from 4.5.0 I've come to the conclusion that this driver has a problem with scanning and maintaining a list of APs when connected via USB3.0 (XHCI), or there is a problem with the XHCI driver not using the 3.0 hub/ports, ***or*** the PC's micro-USB 3.0 port is faulty.

I've bought another device, the **Comfast CF-912AC** `[0BDA:8812]`, which uses the same driver and it exhibits similar problems. The two devices are:
```
Bus 001 Device 087: ID 0bda:0811 Realtek Semiconductor Corp. RTL8821AU 802.11ac WLAN Adapter 1T1R 
Bus 001 Device 085: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter
```
I've found that at various times the device(s) will:

a) see no public APs on 2.4GHz (but will on 5.8GHz); will see hidden SSIDs on 2.4GHz
b) see no APs at all
c) see all APs (hidden and public SSIDs) on both bands

The PC, an Asus T300CHI (tablet 2-in-1 format) has a single v3.0 micro-USB port (so I have no option of testing in another port). With the WiFi device connected directly to the port, or via a v3.0 unpowered hub, I see these issues. Looking more closely I noticed that xhci_hcd offers both 2.0 and 3.0 hubs, but every device (internal and external) is connected via the 2.0 hub (max rate 480Mbps). I recall issues like this in the early days of the Linux kernel XHCI implementation so I'll dig deeper on that.
```
$ lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/11p, 480M
    |__ Port 2: Dev 88, If 0, Class=Vendor Specific Class, Driver=rtl8812au, 480M
    |__ Port 4: Dev 3, If 1, Class=Wireless, Driver=btusb, 12M
    |__ Port 4: Dev 3, If 0, Class=Wireless, Driver=btusb, 12M
    |__ Port 5: Dev 5, If 1, Class=Video, Driver=uvcvideo, 480M
    |__ Port 5: Dev 5, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 6: Dev 7, If 0, Class=Mass Storage, Driver=usb-storage, 480M
    |__ Port 7: Dev 9, If 0, Class=Human Interface Device, Driver=usbhid, 12M
```
On another PC, a Dell XPS M1530, connected via a v2.0 hub and port, both devices have always seen all the available APs on both bands.

Turbito, 2017-08-06:
Using 4.13.0-rc3 here. I have tested with the latest version at today and cannot see wireless networks unitl connected via hidden ap (my ap isn't hidden actually)

A few months ago tested with the driver 4.2.22-beta and it worked but I can't compile with the new kernel.
The bus id of my device is 2357:010d

rawkahn, 2017-10-28:
I have similar issues with my surface pro 2, running Kali dual-boot. Single USB 3.0 connection, unstable AP list, poor if any reception etc, my onboard wifi has zero problems. On another forum, some guy ditched Alfa's cable & runs his device on a regular micro-USB cable so its at least stable & functional in USB 2.0. Possibly a conflict between USB 2/3 within the driver? Talk about a headache.