abperiasamy/rtl8812AU_8821AU_linux #230: rtl8812AU_8821AU-kmp-default-20170613_k4.4.74_18.20-2.1.x86_64 with TP-LINK ARcher T4UH on openSUSE Leap 42.2

sjvudp, 2017-08-10:
After buying the TP-LINK Archer T4UH that claims "Linux support" I realized that the driver TP-LINK offers cannot be compiled easily (even after hacking) in openSUSE Leap 42.2 (4.4.74-18.20-default).
So I used rtl8812AU_8821AU-kmp-default-20170613_k4.4.74_18.20-2.1.x86_64 from the openSUSE build service (http://download.opensuse.org/repositories/home:/Akoellh:/Kernelmodules/openSUSE_Leap_42.2_Update/x86_64/rtl8812AU_8821AU-kmp-default-20170613_k4.4.74_18.20-2.1.x86_64.rpm). Basically it worked quite hassle-free, but there are some minor issues.

The first day the WLAN AP was close by (about 1m): The WLANs were detected and I could log in. At some point (maybe the AP restarted after reconfiguring) I got these messages:
```
Aug 08 23:12:24 i7g4x4a NetworkManager[2071]: <info>  (wlan0): roamed from BSSID CC:CE:1E:C8:F9:92 (FMQ-WLAN) to (none) ((none))
Aug 08 23:12:24 i7g4x4a NetworkManager[2071]: <info>  (wlan0): supplicant interface state: completed -> associating
Aug 08 23:12:24 i7g4x4a kernel: RTL871X: rtw_set_802_11_connect(wlan0)  fw_state=0x00000009
Aug 08 23:12:24 i7g4x4a kernel: RTL871X: ERROR rtw_chk_start_clnt_join(wlan0) cur_ch:0
Aug 08 23:12:24 i7g4x4a kernel: ------------[ cut here ]------------
Aug 08 23:12:24 i7g4x4a kernel: WARNING: CPU: 5 PID: 3700 at /home/abuild/rpmbuild/BUILD/rtl8812AU_8821AU_linux/obj/default/core/rtw_mlme_ext.c:13647 rtw_chk_start_clnt_join+0x7b/
Aug 08 23:12:24 i7g4x4a kernel: Modules linked in: af_packet rtl8812au(O) cfg80211 nls_iso8859_1 nls_cp437 vfat fat ses enclosure scsi_transport_sas fuse nf_log_ipv6 xt_pkttype nf
Aug 08 23:12:24 i7g4x4a kernel:  i2c_i801
Aug 08 23:12:24 i7g4x4a kernel:  mei_me
Aug 08 23:12:24 i7g4x4a kernel:  coretemp
Aug 08 23:12:24 i7g4x4a kernel:  snd_timer
Aug 08 23:12:24 i7g4x4a kernel:  mei lpc_ich mii snd shpchp mfd_core sparse_keymap soundcore rfkill kvm mxm_wmi wmi tpm_infineon irqbypass processor thermal fan battery fjes pcspk
Aug 08 23:12:24 i7g4x4a kernel: CPU: 5 PID: 3700 Comm: RTW_CMD_THREAD Tainted: G           O     4.4.74-18.20-default #1
Aug 08 23:12:24 i7g4x4a kernel: Hardware name: ASUS All Series/Z87-C, BIOS 2103 08/15/2014
Aug 08 23:12:24 i7g4x4a kernel:  0000000000000000 ffffffff81329737 0000000000000000 ffffffffa0bd5748
Aug 08 23:12:24 i7g4x4a kernel:  ffffffff8107eda1 ffffc90001c77000 0000000000000022 ffffc90001c782da
Aug 08 23:12:24 i7g4x4a kernel:  ffffc90001c78252
Aug 08 23:12:24 i7g4x4a kernel:  0000000000000000
Aug 08 23:12:24 i7g4x4a kernel:  ffffffffa0b5e19e ffffffffa0b5e44a
Aug 08 23:12:24 i7g4x4a kernel: Call Trace:
Aug 08 23:12:24 i7g4x4a kernel:  [<ffffffff81019ea9>] dump_trace+0x59/0x320
Aug 08 23:12:24 i7g4x4a kernel:  [<ffffffff8101a26a>] show_stack_log_lvl+0xfa/0x180
Aug 08 23:12:24 i7g4x4a kernel:  [<ffffffff8101b011>] show_stack+0x21/0x40
Aug 08 23:12:24 i7g4x4a kernel:  [<ffffffff81329737>] dump_stack+0x5c/0x85
Aug 08 23:12:24 i7g4x4a kernel:  [<ffffffff8107eda1>] warn_slowpath_common+0x81/0xb0
Aug 08 23:12:24 i7g4x4a kernel:  [<ffffffffa0b5e19e>] rtw_chk_start_clnt_join+0x7b/0x95 [rtl8812au]
Aug 08 23:12:24 i7g4x4a kernel:  [<ffffffffa0b5e44a>] join_cmd_hdl+0x292/0x324 [rtl8812au]
Aug 08 23:12:24 i7g4x4a kernel:  [<ffffffffa0b3d9f1>] rtw_cmd_thread+0x285/0x39f [rtl8812au]
Aug 08 23:12:24 i7g4x4a kernel:  [<ffffffff8109d8d2>] kthread+0xd2/0xf0
Aug 08 23:12:24 i7g4x4a kernel:  [<ffffffff8160fb0f>] ret_from_fork+0x3f/0x70
Aug 08 23:12:24 i7g4x4a kernel: DWARF2 unwinder stuck at ret_from_fork+0x3f/0x70
Aug 08 23:12:24 i7g4x4a kernel: 
Aug 08 23:12:24 i7g4x4a kernel: Leftover inexact backtrace:
Aug 08 23:12:24 i7g4x4a kernel:  [<ffffffff8109d800>] ? kthread_park+0x50/0x50
Aug 08 23:12:24 i7g4x4a kernel: ---[ end trace 48b638781d32a31e ]---
Aug 08 23:12:24 i7g4x4a NetworkManager[2071]: <info>  (wlan0): supplicant interface state: associating -> disconnected
Aug 08 23:12:24 i7g4x4a NetworkManager[2071]: <info>  (wlan0): supplicant interface state: disconnected -> scanning
Aug 08 23:12:25 i7g4x4a kernel: RTL871X: rtw_set_802_11_connect(wlan0)  fw_state=0x00000008
Aug 08 23:12:25 i7g4x4a NetworkManager[2071]: <info>  (wlan0): supplicant interface state: scanning -> associating
Aug 08 23:12:29 i7g4x4a kernel: RTL871X: start auth
Aug 08 23:12:29 i7g4x4a kernel: RTL871X: auth success, start assoc
Aug 08 23:12:31 i7g4x4a kernel: RTL871X: rtw_cfg80211_indicate_connect(wlan0) BSS not found !!
Aug 08 23:12:31 i7g4x4a kernel: RTL871X: assoc success
Aug 08 23:12:31 i7g4x4a kernel: RTL871X: send eapol packet
Aug 08 23:12:31 i7g4x4a NetworkManager[2071]: <info>  (wlan0): supplicant interface state: associating -> 4-way handshake
Aug 08 23:12:31 i7g4x4a kernel: RTL871X: send eapol packet
Aug 08 23:12:31 i7g4x4a kernel: RTL871X: set pairwise key camid:4, addr:cc:ce:1e:c8:f9:92, kid:0, type:AES
Aug 08 23:12:31 i7g4x4a kernel: RTL871X: set group key camid:5, addr:cc:ce:1e:c8:f9:92, kid:1, type:AES
Aug 08 23:12:31 i7g4x4a NetworkManager[2071]: <info>  (wlan0): supplicant interface state: 4-way handshake -> completed
Aug 08 23:12:31 i7g4x4a NetworkManager[2071]: <info>  (wlan0): roamed from BSSID (none) ((none)) to CC:CE:1E:C8:F9:92 (FMQ-WLAN)
````

Another thing that puzzles me is "BSS not found" (I don't understand what it means):
```
Aug 08 23:22:31 i7g4x4a NetworkManager[2071]: <info>  Config: set interface ap_scan to 1
Aug 08 23:22:31 i7g4x4a NetworkManager[2071]: <info>  (wlan0): supplicant interface state: inactive -> associating
Aug 08 23:22:31 i7g4x4a kernel: RTL871X: rtw_set_802_11_connect(wlan0)  fw_state=0x00000008
Aug 08 23:22:31 i7g4x4a kernel: RTL871X: start auth
Aug 08 23:22:31 i7g4x4a kernel: RTL871X: auth success, start assoc
Aug 08 23:22:31 i7g4x4a kernel: RTL871X: rtw_cfg80211_indicate_connect(wlan0) BSS not found !!
Aug 08 23:22:31 i7g4x4a kernel: RTL871X: assoc success
Aug 08 23:22:31 i7g4x4a kernel: RTL871X: send eapol packet
Aug 08 23:22:31 i7g4x4a NetworkManager[2071]: <info>  (wlan0): supplicant interface state: associating -> 4-way handshake
Aug 08 23:22:31 i7g4x4a kernel: RTL871X: send eapol packet
Aug 08 23:22:31 i7g4x4a kernel: RTL871X: set pairwise key camid:4, addr:cc:ce:1e:c8:f9:92, kid:0, type:AES
[... 41 lines cut ...]
Aug 08 23:26:07 i7g4x4a NetworkManager[2071]: <info>  (wlan0): supplicant interface state: associating -> disconnected
Aug 08 23:26:07 i7g4x4a NetworkManager[2071]: <info>  (wlan0): supplicant interface state: disconnected -> scanning
Aug 08 23:26:07 i7g4x4a kernel: RTL871X: rtw_set_802_11_connect(wlan0)  fw_state=0x00000008
Aug 08 23:26:07 i7g4x4a NetworkManager[2071]: <info>  (wlan0): supplicant interface state: scanning -> associating
Aug 08 23:26:10 i7g4x4a kernel: RTL871X: start auth
Aug 08 23:26:10 i7g4x4a kernel: RTL871X: auth success, start assoc
Aug 08 23:26:10 i7g4x4a kernel: RTL871X: rtw_cfg80211_indicate_connect(wlan0) BSS not found !!
Aug 08 23:26:10 i7g4x4a kernel: RTL871X: assoc success
Aug 08 23:26:10 i7g4x4a NetworkManager[2071]: <info>  (wlan0): supplicant interface state: associating -> associated
Aug 08 23:26:10 i7g4x4a kernel: RTL871X: send eapol packet
Aug 08 23:26:10 i7g4x4a NetworkManager[2071]: <info>  (wlan0): supplicant interface state: associated -> 4-way handshake
Aug 08 23:26:10 i7g4x4a kernel: RTL871X: send eapol packet
Aug 08 23:26:10 i7g4x4a kernel: RTL871X: set pairwise key camid:4, addr:cc:ce:1e:c8:f9:92, kid:0, type:AES
Aug 08 23:26:10 i7g4x4a kernel: RTL871X: set group key camid:5, addr:cc:ce:1e:c8:f9:92, kid:2, type:AES
Aug 08 23:26:10 i7g4x4a NetworkManager[2071]: <info>  (wlan0): supplicant interface state: 4-way handshake -> completed
Aug 08 23:26:10 i7g4x4a NetworkManager[2071]: <info>  (wlan0): roamed from BSSID (none) ((none)) to CC:CE:1E:C8:F9:92 (FMQ-WLAN)
Aug 08 23:27:25 i7g4x4a kernel: RTL871X: send eapol packet
Aug 08 23:27:25 i7g4x4a kernel: RTL871X: set group key camid:6, addr:cc:ce:1e:c8:f9:92, kid:1, type:AES
````

The second day I had the WLAN AP move about 10m away, so the 5GHz was rather weak, but I couldn't see the issues I had seen on the first day.

However it seems the WPS-key on the stick has no effect at all. Can you confirm?

Are these problems already known and fixed?

Throne3d, 2017-08-13:
This problem looks plausibly related to #215. I've been getting intermittent disconnections, with a similar error involving `rtw_set_802_11_connect` and `rtw_chk_start_clnt_join`. I'm not sure what triggers it, but when it starts it usually repeats once every 120 seconds. (Moving the device to another port seems to fix it, sometimes?)

sjvudp, 2017-08-24:
Could it be some roaming between 2.4 and 5GHz (same SSID), or the access point changing to more silent channels ("self optimizing")?