abperiasamy/rtl8812AU_8821AU_linux #232: Compiling this driver for Raspberry Pi 3 (kernel version  4.9) does not work

Cider1987, 2017-09-02:
Hi! 
I know there is a separate issue about failing to compile this for raspberry pi, and yes, the 
`kernel=vmlinuz-3.10-3-rpi
initramfs initrd.img-3.10-3-rpi followkernel` part no longer works

Maybe there is a working fork or something?

I also tried to "just do it" without this part and it almost sort of worked (insmod did not produce errors) but wpa-supplicant has troubles trying to reconfigure it.

Please help

tonsimple, 2017-09-03:
I've compiled it for a Fedora with kernel 4.4.31-11 (Qubes OS) without issue.
Later kernels don't work because of Qubes not having the right kernel header packages for them on Fedora.
What exact issues are you seeing?

andreasloe, 2017-09-09:
I do not know what the issue above was but I am getting the following error
`sudo make`
`make ARCH=arm CROSS_COMPILE= -C /lib/modules/4.9.35-v7+/build M=/usr/src/rtl8812AU_8821AU_linux  modules`
`make[1]: Entering directory '/usr/src/linux-4.9.35'`

```
  ERROR: Kernel configuration is invalid.
         include/generated/autoconf.h or include/config/auto.conf are missing.
         Run 'make oldconfig && make prepare' on kernel src to fix it.
```

`  WARNING: Symbol version dump ./Module.symvers`
`           is missing; modules will have no dependencies and modversions.`

`  CC [M]  /usr/src/rtl8812AU_8821AU_linux/core/rtw_cmd.o`
`In file included from <command-line>:0:0:`
`././include/linux/kconfig.h:4:32: fatal error: generated/autoconf.h: Datei oder Verzeichnis nicht gefunden
`` #include <generated/autoconf.h>
`                                ^
`compilation terminated.
``scripts/Makefile.build:293: recipe for target '/usr/src/rtl8812AU_8821AU_linux/core/rtw_cmd.o' failed
``make[2]: *** [/usr/src/rtl8812AU_8821AU_linux/core/rtw_cmd.o] Error 1
``Makefile:1490: recipe for target '_module_/usr/src/rtl8812AU_8821AU_linux' failed
``make[1]: *** [_module_/usr/src/rtl8812AU_8821AU_linux] Error 2
``make[1]: Leaving directory '/usr/src/linux-4.9.35'
``Makefile:1584: recipe for target 'modules' failed
``make: *** [modules] Error 2`

I am using 4.9.35-v7+.

andreasloe, 2017-09-09:
I think I know which mistake I made. I forgot to
` sudo apt-get install raspberrypi-kernel-headers`
Now, the first steps go through, but the last step gives
```
sudo modprobe -a rtl8812au
modprobe: WARNING: Module rtl8812au not found.
```

andreasloe, 2017-09-09:
I am sorry to bother you all but since I am trying this now for hours it seems I found a solution:
[https://askubuntu.com/questions/861717/how-to-install-d-link-dwa-172-id-20013318-driver](https://askubuntu.com/questions/861717/how-to-install-d-link-dwa-172-id-20013318-driver)