abperiasamy/rtl8812AU_8821AU_linux #235: Build error from unused variable warnings

thedewi, 2017-10-08:
On current master (0292aa6) a trivial warning is breaking the build for me because ```-Werror``` is enabled:

```
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/os_dep/linux/usb_intf.o
/var/lib/dkms/rtl8812au/4.3.14/build/os_dep/linux/usb_intf.c: In function ‘autoresume_enter’:
/var/lib/dkms/rtl8812au/4.3.14/build/os_dep/linux/usb_intf.c:1208:24: error: unused variable ‘pmlmeinfo’ [-Werror=unused-variable]
/var/lib/dkms/rtl8812au/4.3.14/build/os_dep/linux/usb_intf.c:1206:24: error: unused variable ‘psecuritypriv’ [-Werror=unused-variable]
cc1: all warnings being treated as errors
make[5]: *** [/var/lib/dkms/rtl8812au/4.3.14/build/os_dep/linux/usb_intf.o] Error 1
```

When I comment out ```-Werror``` in ```Makefile``` (thanks to #176) it builds successfully (but it's probably better to delete the variables 😸).

thedewi, 2017-10-08:
Update: I was preparing a PR to remove the unused variables, but that revealed more warnings and my C is too rusty:
```
  CC [M]  /var/lib/dkms/rtl8812au/4.3.14/build/os_dep/linux/ioctl_cfg80211.o
/var/lib/dkms/rtl8812au/4.3.14/build/os_dep/linux/ioctl_cfg80211.c:6000:2: error: initialization from incompatible pointer type [-Werror]
/var/lib/dkms/rtl8812au/4.3.14/build/os_dep/linux/ioctl_cfg80211.c:6000:2: error: (near initialization for ‘rtw_cfg80211_ops.add_station’) [-Werror]
/var/lib/dkms/rtl8812au/4.3.14/build/os_dep/linux/ioctl_cfg80211.c:6001:2: error: initialization from incompatible pointer type [-Werror]
/var/lib/dkms/rtl8812au/4.3.14/build/os_dep/linux/ioctl_cfg80211.c:6001:2: error: (near initialization for ‘rtw_cfg80211_ops.del_station’) [-Werror]
cc1: all warnings being treated as errors
```

plumdog, 2018-02-27:
Hi,

While using this driver (at latest, currently bed205c14a363fedd8b3a497ef0141588b610d50), when running `modprobe rtl8812au`, I get a warning in syslog that points at [this line](https://github.com/torvalds/linux/blob/master/net/wireless/core.c#L393) in the kernel code, which checks `WARN_ON(ops->add_station && !ops->del_station);`

I'm no C programmer, I should add, but it looks to me that this warning is validating that if `add_station` is there, `del_station` must be too, and the fix for this issue - in #240 - removed `del_station`.

It's not immediately clear to me why `del_station` was removed and `add_station` was kept, as the both appear to be triggering errors in the errors above, so I'm not sure whether #240 will have fixed the build for anyone.

So it is my (very tentative) guess that #240 should be reverted as it doesn't fix this issue and it triggers a warning when the module is inserted.