abperiasamy/rtl8812AU_8821AU_linux #238: Stability issues with Fedora 27 and Kernel 4.14.11

keachi, 2018-01-10:
I get a lot of disconnects. Sometimes it is stable for multiple minutes and then I get a disconnect and reconnect from the NetworkManager.
I use Fedora 27 (which has kernel version 4.14.11) and installed this as a dkms module.
This is the dmesg output:
```
RTL871X: rtw_set_802_11_connect(wlp5s0u2)  fw_state=0x00000009
RTL871X: ERROR rtw_chk_start_clnt_join(wlp5s0u2) cur_ch:0
------------[ cut here ]------------
WARNING: CPU: 2 PID: 28559 at /var/lib/dkms/rtl8812AU_8821AU_linux/1.0/build/core/rtw_mlme_ext.c:13647 rtw_chk_start_clnt_join+0x30/0x83 [rtl8812au]
Modules linked in: rtl8812au(OE) rfcomm fuse cmac md4 nls_utf8 cifs dns_resolver fscache ccm nf_conntrack_netbios_ns nf_conntrack_broadcast xt_CT ip6t_rpfilter ip6t_REJECT nf_reject_ipv6 xt_conntrack ip_set nfnetlink ebtable_nat ebtable_broute bridge stp llc ip6table_nat nf_conntrack_ipv6 nf_defrag_ipv6 nf_nat_ipv6 ip6table_mangle ip6table_raw ip6table_security iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack libcrc32c iptable_mangle iptable_raw iptable_security ebtable_filter ebtables ip6table_filter ip6_tables bnep sunrpc vfat fat intel_rapl x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel kvm arc4 iwldvm irqbypass crct10dif_pclmul crc32_pclmul mac80211 ghash_clmulni_intel iTCO_wdt iTCO_vendor_support intel_cstate intel_uncore intel_rapl_perf btusb
 btrtl iwlwifi btbcm btintel bluetooth rtsx_usb_ms joydev memstick cfg80211 snd_hda_codec_hdmi snd_hda_codec_realtek snd_hda_codec_generic ecdh_generic snd_hda_intel snd_hda_codec rfkill ir_rc6_decoder snd_hda_core snd_hwdep snd_seq rc_rc6_mce snd_seq_device ir_lirc_codec snd_pcm lirc_dev nuvoton_cir mei_me rc_core snd_timer mei i2c_i801 snd lpc_ich soundcore ie31200_edac shpchp hid_logitech_hidpp i915 hid_logitech_dj rtsx_usb_sdmmc mmc_core i2c_algo_bit drm_kms_helper drm crc32c_intel r8169 serio_raw rtsx_usb mii video [last unloaded: rtl8812au]
CPU: 2 PID: 28559 Comm: RTW_CMD_THREAD Tainted: G        W  OE   4.14.11-300.fc27.x86_64 #1
Hardware name: ZOTAC ZBOX-ID88/ID89/ID90/ZBOX-ID88/ID89/ID90, BIOS 4.6.5 07/17/2013                                                                                      
task: ffff96ce2c431f40 task.stack: ffffbef840594000                                                                                                                      
RIP: 0010:rtw_chk_start_clnt_join+0x30/0x83 [rtl8812au]                                                                                                                  
RSP: 0018:ffffbef840597e80 EFLAGS: 00010282                                                                                                                              
RAX: 0000000000000039 RBX: ffffbef8406c3000 RCX: 0000000000000000                                                                                                        
RDX: 0000000000000000 RSI: ffff96cec0313a38 RDI: ffff96cec0313a38                                                                                                        
RBP: ffffbef8406c425a R08: 000000000000eecf R09: 0000000000000004                                                                                                        
R10: ffffbef840597df0 R11: 0000000000000001 R12: 0000000000000022                                                                                                        
R13: ffffbef8406c424e R14: 0000000000000000 R15: 0000000000000001                                                                                                        
FS:  0000000000000000(0000) GS:ffff96cec0300000(0000) knlGS:0000000000000000                                                                                             
CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033                                                                                                                        
CR2: 000055c5e27e6a10 CR3: 000000000d20a001 CR4: 00000000001606e0                                                                                                        
Call Trace:                                                                                                                                                              
 join_cmd_hdl+0x28d/0x33a [rtl8812au]                                                                                                                                    
 rtw_cmd_thread+0x1fc/0x3bf [rtl8812au]                                                                                                                                  
 ? rtw_stop_cmd_thread+0x52/0x52 [rtl8812au]                                                                                                                             
 kthread+0x118/0x130                                                                                                                                                     
 ? kthread_park+0x60/0x60                                                                                                                                                
 ret_from_fork+0x1f/0x30                                                                                                                                                 
Code: 48 85 d2 41 0f 94 c0 48 85 c9 0f 94 c0 41 08 c0 75 22 48 85 f6 74 1d 0f b6 87 8a 0a 00 00 84 c0 75 3e 83 3d 54 24 0a 00 04 77 14 <0f> ff b8 00 00 00 00 eb 07 0f ff b8 00 00 00 00 0f b6 c0 c3 48                                                                                                                                            
---[ end trace a4e0c70992b76f0a ]---
```

nigelterry, 2018-01-22:
I am getting a similar issue with Fedora27 and 4.14.13-300.fc27.x86_64. 

> [10375.683670] ------------[ cut here ]------------
[10375.683702] WARNING: CPU: 0 PID: 30984 at /home/nterry/rtl8812AU_8821AU_linux/core/rtw_mlme_ext.c:13647 rtw_chk_start_clnt_join+0x34/0x88 [rtl8812au]
[10375.683704] Modules linked in: fuse ccm nf_conntrack_netbios_ns nf_conntrack_broadcast xt_CT ip6t_rpfilter ip6t_REJECT nf_reject_ipv6 xt_conntrack ip_set nfnetlink ebtable_nat ebtable_broute bridge stp llc ip6table_nat nf_conntrack_ipv6 nf_defrag_ipv6 nf_nat_ipv6 ip6table_mangle ip6table_raw ip6table_security iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack libcrc32c iptable_mangle iptable_raw iptable_security ebtable_filter ebtables ip6table_filter ip6_tables sunrpc btrfs xor zstd_compress raid6_pq zstd_decompress xxhash snd_hda_codec_hdmi snd_hda_codec_idt rtl8812au(OE) uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_core videodev media snd_hda_codec_generic snd_hda_intel snd_hda_codec snd_hda_core snd_hwdep snd_seq snd_seq_device arc4 iwldvm mac80211
[10375.683751]  iTCO_wdt intel_rapl wmi_bmof iTCO_vendor_support iwlwifi x86_pkg_temp_thermal intel_powerclamp coretemp hp_wmi snd_pcm sparse_keymap cfg80211 rfkill rtsx_pci_ms memstick snd_timer snd kvm_intel soundcore joydev shpchp kvm irqbypass mei_me mei lpc_ich i2c_i801 crct10dif_pclmul crc32_pclmul hp_accel ghash_clmulni_intel lis3lv02d input_polldev intel_cstate intel_uncore intel_rapl_perf wmi vboxpci(OE) vboxnetadp(OE) vboxnetflt(OE) vboxdrv(OE) binfmt_misc amdkfd i915 amd_iommu_v2 radeon rtsx_pci_sdmmc mmc_core i2c_algo_bit drm_kms_helper ttm crc32c_intel drm r8169 serio_raw rtsx_pci mii video
[10375.683789] CPU: 0 PID: 30984 Comm: RTW_CMD_THREAD Tainted: G        W  OE   4.14.13-300.fc27.x86_64 #1
[10375.683791] Hardware name: Hewlett-Packard EW6-I7D2281TBR1/1657, BIOS F.1B 10/05/2011
[10375.683793] task: ffff9e785d3f8000 task.stack: ffffaed0c5c9c000
[10375.683815] RIP: 0010:rtw_chk_start_clnt_join+0x34/0x88 [rtl8812au]
[10375.683817] RSP: 0018:ffffaed0c5c9fe60 EFLAGS: 00010286
[10375.683820] RAX: 000000000000003a RBX: ffffaed0c21c9000 RCX: 0000000000000000
[10375.683822] RDX: 0000000000000000 RSI: ffff9e7b9fa16a38 RDI: ffff9e7b9fa16a38
[10375.683824] RBP: ffffaed0c5c9fe60 R08: 000000000000084a R09: 0000000000000007
[10375.683826] R10: ffffaed0c5c9fe70 R11: ffffffffb95f71cd R12: ffffaed0c21ca25a
[10375.683827] R13: 0000000000000022 R14: ffffaed0c21ca24e R15: 0000000000000001
[10375.683830] FS:  0000000000000000(0000) GS:ffff9e7b9fa00000(0000) knlGS:0000000000000000
[10375.683832] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[10375.683834] CR2: 00007fa785f17e70 CR3: 00000002a720a004 CR4: 00000000000606f0
[10375.683836] Call Trace:
[10375.683860]  join_cmd_hdl+0x2a5/0x34c [rtl8812au]
[10375.683877]  rtw_cmd_thread+0x219/0x3e1 [rtl8812au]
[10375.683881]  kthread+0x125/0x140
[10375.683891]  ? rtw_stop_cmd_thread+0x57/0x57 [rtl8812au]
[10375.683892]  ? kthread_park+0x60/0x60
[10375.683895]  ret_from_fork+0x32/0x40
[10375.683896] Code: 48 85 d2 41 0f 94 c0 48 85 c9 0f 94 c0 41 08 c0 75 22 48 85 f6 74 1d 0f b6 87 8a 0a 00 00 84 c0 75 3f 83 3d 4c 72 08 00 04 77 15 <0f> ff b8 00 00 00 00 eb 07 0f ff b8 00 00 00 00 0f b6 c0 5d c3 
[10375.683920] ---[ end trace 7a56a5598fe44ba6 ]---

johnperry-math, 2018-02-16:
My impression is that Fedora's 4.14 kernel has trashed wireless. My TP-Link won't even register as a wireless device if I run that kernel, but the moment I revert to 4.13 everything is as it was before (which wasn't great, which is why I'm looking at an 8812au device).