abperiasamy/rtl8812AU_8821AU_linux #241: Build failure on kernel 4.15: include/osdep_service_linux.h:274:8: error: ‘_timer’ has no member named ‘data’

lgg4373, 2018-02-05:
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.15.1-2.g9df97e2-default/build M=./rtl8812AU_8821AU_linux  modules
make[1]: Entering directory '/usr/src/linux-4.15.1-2.g9df97e2-obj/x86_64/default'
  CC [M]  ./rtl8812AU_8821AU_linux/core/rtw_cmd.o
In file included from ./rtl8812AU_8821AU_linux/include/osdep_service.h:41:0,
                 from ./rtl8812AU_8821AU_linux/include/drv_types.h:32,
                 from ./rtl8812AU_8821AU_linux/core/rtw_cmd.c:22:
**./rtl8812AU_8821AU_linux/include/osdep_service_linux.h: In function ‘_init_timer’:
./rtl8812AU_8821AU_linux/include/osdep_service_linux.h:274:8: error: ‘_timer’ has no member named ‘data’
  ptimer->data = (unsigned long)cntx;
        ^
./rtl8812AU_8821AU_linux/include/osdep_service_linux.h:275:2: error: implicit declaration of function ‘init_timer’ [-Werror=implicit-function-declaration]
  init_timer(ptimer);
  ^
cc1: all warnings being treated as errors**
/usr/src/linux-4.15.1-2.g9df97e2/scripts/Makefile.build:317: recipe for target './rtl8812AU_8821AU_linux/core/rtw_cmd.o' failed
make[4]: *** [./rtl8812AU_8821AU_linux/core/rtw_cmd.o] Error 1
/usr/src/linux-4.15.1-2.g9df97e2/Makefile:1513: recipe for target '_module_./rtl8812AU_8821AU_linux' failed
make[3]: *** [_module_./rtl8812AU_8821AU_linux] Error 2
Makefile:146: recipe for target 'sub-make' failed
make[2]: *** [sub-make] Error 2
Makefile:24: recipe for target '__sub-make' failed
make[1]: *** [__sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-4.15.1-2.g9df97e2-obj/x86_64/default'
Makefile:1584: recipe for target 'modules' failed
make: *** [modules] Error 2

kimocoder, 2018-02-05:
this driver does not support the changes in timers of kernel v4.15

lakshman5876, 2018-02-07:
Same here on 4.15.1

```make
In file included from /var/lib/dkms/rtl8812au/4.3.14/build/include/osdep_service.h:41:0,
                 from /var/lib/dkms/rtl8812au/4.3.14/build/include/drv_types.h:32,
                 from /var/lib/dkms/rtl8812au/4.3.14/build/core/rtw_ioctl_set.c:22:
/var/lib/dkms/rtl8812au/4.3.14/build/include/osdep_service_linux.h: In function ‘_init_timer’:
/var/lib/dkms/rtl8812au/4.3.14/build/include/osdep_service_linux.h:274:8: error: ‘_timer {aka struct timer_list}’ has no member named ‘data’
  ptimer->data = (unsigned long)cntx;

```

eReGeBe, 2018-02-07:
I made #242 which allows the driver to compile. However, I cannot verify that this makes the driver work on kernels >= 4.15.

MissingNoIOI, 2018-02-08:
I just tried the patch on Arch Linux with Kernel 4.15.1 and it doesn't work. The dkms module compiles successfully, but my PC freezes during boot afterwards. Had to restart in fallback mode to remove the module

closed 2018-02-08 by harshavardhana

eReGeBe, 2018-02-10:
@MissingNoIOI sorry for the inconveniences, I probably shouldn't have opened a PR for a patch I hadn't verified.

@harshavardhana perhaps my patch should be rolled back until someone can corroborate its correctness?

Sorry again for the trouble.

mikeh007, 2018-07-11:
As a very keen supporter of Linux, and a user of Ubuntu and Lubuntu,   but not an expert developer,   can I just ask -- innocently -- how is it that,  a mainline kernel is released with such radical changes as to timer code, which (it would seem) then SCREWS UP untold numbers of legacy drivers for things like WiFi adaptors ?     

I've loved Ubuntu for 7 years now and run my entire business on it - desktops and laptops - yesterday I upgraded my main laptop from 16.04.02  to  18.04  to then find the Wifi as flaky as hell.  

Spent all last evening, and half this morning,   learning about drivers - mine is for an innocent, excellent,  $10 USB wireless adaptor from TPLINK),  and on about the 20th attempt to 'fix the driver module'  for the RTL8192 which this device uses - now using TPLINK's own linux driver-install instructions --- I find the   sudo make install   step FALLS OVER,   because..   well from one point of view,  pardon me, but  Kernel 4.15 seems not fit for purpose ?     ie it WILL NOT ALLOW me to use a driver created by the manufacturer and which plainly has worked fine on 16.04.  

I have the highest regard for the kernel developers  but didn't this sort of this ought to be checked out BEFORE urging us all to upgrade ?      16.04 was the last LTS,  this is the next:  I assumed it would be a safe move up.   Evidently not.     Resolved to have to backtrack to a reinstall of 16.04LTS where I was before and lose 2 days for no progress. 

I hope the kernel is fit for purpose again by the time 16.04 LTS ends.  

Thanks all - any and all advice and wisdom would be much appreciated.

Mike H
London

chinlean, 2020-03-18:
> As a very keen supporter of Linux, and a user of Ubuntu and Lubuntu, but not an expert developer, can I just ask -- innocently -- how is it that, a mainline kernel is released with such radical changes as to timer code, which (it would seem) then SCREWS UP untold numbers of legacy drivers for things like WiFi adaptors ?
> 
> I've loved Ubuntu for 7 years now and run my entire business on it - desktops and laptops - yesterday I upgraded my main laptop from 16.04.02 to 18.04 to then find the Wifi as flaky as hell.
> 
> Spent all last evening, and half this morning, learning about drivers - mine is for an innocent, excellent, $10 USB wireless adaptor from TPLINK), and on about the 20th attempt to 'fix the driver module' for the RTL8192 which this device uses - now using TPLINK's own linux driver-install instructions --- I find the sudo make install step FALLS OVER, because.. well from one point of view, pardon me, but Kernel 4.15 seems not fit for purpose ? ie it WILL NOT ALLOW me to use a driver created by the manufacturer and which plainly has worked fine on 16.04.
> 
> I have the highest regard for the kernel developers but didn't this sort of thing ought to be checked out BEFORE urging us all to upgrade ? 16.04 was the last LTS, this is the next: I assumed it would be a safe move up. Evidently not. Resolved to have to backtrack to a reinstall of 16.04LTS where I was before and lose 2 days for no progress.
> 
> I hope the kernel is fit for purpose again by the time 16.04 LTS ends.
> 
> Thanks all - any and all advice and wisdom would be much appreciated.
> 
> Mike H
> London

I agree with you Mike. Just imagine how many million hours are wasted around the world with this kind of interfaces changed between version. I'm working mostly on Windows and relatively little on Linux platform. Such interfaces changes never happen (as I experienced)  from version to version. While working in Linux, spends tens (if not hundred) of hours working on the compilation issues make me sick. These hours should be spend on the real business problem, not programming issues

MilhouseVH, 2020-03-19:
Out of tree drivers like this repo are not the concern of kernel developers. Period. That's why in-tree drivers exist, and why hardware manufacturers should be submitting drivers for inclusion in the kernel source code.

> mine is for an innocent, excellent, $10 USB wireless adaptor from TPLINK)

And of that $10 exactly $0.00 from the hardware manufacturer went into upstreaming in-tree kernel drivers so if you want to complain to anyone then complain to Realtek.

To be quite honest the open source community enables Realtek to be a bad open source citizen by continuing to hack these drivers to work with the latest kernel when it would be better for all concerned if all of these rtl* repsitories died off so that Realtek would have no choice but to get their drivers accepted into the kernel tree (or stop pretending they support Linux).

MilhouseVH, 2020-03-19:
> I agree with you Mike. Just imagine how many million hours are wasted around the world with this kind of interfaces changed between version. I'm working mostly on Windows and relatively little on Linux platform. Such interfaces changes never happen (as I experienced) from version to version. 

"Linux does not have a binary kernel interface, nor does it have a stable kernel interface." [1]

> While working in Linux, spends tens (if not hundred) of hours working on the compilation issues make me sick. These hours should be spend on the real business problem, not programming issues

If you're forced to use out of tree drivers, then demand that the author (usually a hardware manufacturer that is now profiting from your efforts) submits their work to the main kernel tree as this will save you and everyone else a lot of time and effort in future.

1. https://www.kernel.org/doc/html/v4.10/process/stable-api-nonsense.html

chinlean, 2020-03-20:
> "Linux does not have a binary kernel interface, nor does it have a stable kernel interface." [1]
>
>
> If you're forced to use out of tree drivers, then demand that the author (usually a hardware [...]

Thanks MilhouseVH educating me on the reasons behind this.

PR #242 (2018-02-08): Adapt to kernel 4.15 timers interface changes
Credits to https://github.com/mkubecek/vmware-host-modules/commit/562121d7bc06 for
helping me understand the changes.

Closes: #241 (?)