abperiasamy/rtl8812AU_8821AU_linux #243: Stalled connection every 120 seconds

KjellMorgenstern, 2018-02-07:
Hello,

I see frequent connection stalls every 120 seconds. I tried with two different USB stick, one rtl8812au, named AC1200, and the other one I think is rtl8811. For example, fping to my router shows this, every two minutes:
```
...
fritz.box : [88], 84 bytes, 2.10 ms (2.25 avg, 0% loss)
fritz.box : [89], 84 bytes, 2.33 ms (2.25 avg, 0% loss)
fritz.box : [90], 84 bytes, 3986 ms (46.0 avg, 3% loss)
fritz.box : [91], 84 bytes, 2986 ms (78.0 avg, 2% loss)
fritz.box : [92], 84 bytes, 1985 ms (98.5 avg, 1% loss)
fritz.box : [93], 84 bytes, 984 ms (107 avg, 0% loss)
fritz.box : [94], 84 bytes, 5.83 ms (106 avg, 0% loss)
...
```
No other devices (linux with a rtl8821ae,  OS X Mac mini) in my network show this when monitoring with fping at the same time. Also, when monitoring the transfer rate with speedometer, is see that the used bandwith drops to zero for about 5-seconds each 120 seconds. WIFI signal strength is very good and does not show any problems.

Looking into the source code shows that there is only one place where a timer of 2 x 60 seconds is used, and thats only in the comment, the actual value assigned is 60 seconds. It is in core/rtw_sta_mgt.c, line 
146, but I think it is unlikely that this is related.

What else could I try to narrow down the problem?
Is this maybe a known issue?

Xim8, 2018-04-25:
I found this post because I was looking for a solution to stop my wifi from periodically stalling some seconds every two minutes.

In my case it was NetworkManager doing scans for networks to do roaming. Setting the BSSID of the connection disables the scans and the stalls are gone.

quartzo, 2018-06-22:
setting the BSSID fixed my regular pauses here too on Ubuntu 18.04.

KjellMorgenstern, 2019-04-16:
I was using Ubuntu 18.04, too. It's very likely that my issue was also caused by network scans from NetworkManager. Probably running scans while having an active connection is not a supported use case, and the issue can be closed.