abperiasamy/rtl8812AU_8821AU_linux #245: No compila en Ubuntu 17.10

fers1, 2018-02-17:
`$ sudo make && sudo make install && sudo modprobe -a rtl8812au 
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.13.0-32-generic/build M=/usr/src/rtl8812au  modules
make[1]: se entra en el directorio '/usr/src/linux-headers-4.13.0-32-generic'
  CC [M]  /usr/src/rtl8812au/core/rtw_cmd.o
In file included from /usr/src/rtl8812au/include/drv_types.h:32:0,
                 from /usr/src/rtl8812au/core/rtw_cmd.c:22:
/usr/src/rtl8812au/include/osdep_service.h: In function ‘thread_enter’:
/usr/src/rtl8812au/include/osdep_service.h:343:2: error: implicit declaration of function ‘allow_signal’; did you mean ‘do_signal’? [-Werror=implicit-function-declaration]
  allow_signal(SIGTERM);
  ^~~~~~~~~~~~
  do_signal
/usr/src/rtl8812au/include/osdep_service.h: In function ‘flush_signals_thread’:
/usr/src/rtl8812au/include/osdep_service.h:353:6: error: implicit declaration of function ‘signal_pending’; did you mean ‘timer_pending’? [-Werror=implicit-function-declaration]
  if (signal_pending (current)) {
      ^~~~~~~~~~~~~~
      timer_pending
/usr/src/rtl8812au/include/osdep_service.h:354:3: error: implicit declaration of function ‘flush_signals’; did you mean ‘do_signal’? [-Werror=implicit-function-declaration]
   flush_signals(current);
   ^~~~~~~~~~~~~
   do_signal
cc1: all warnings being treated as errors
scripts/Makefile.build:308: fallo en las instrucciones para el objetivo '/usr/src/rtl8812au/core/rtw_cmd.o'
make[2]: *** [/usr/src/rtl8812au/core/rtw_cmd.o] Error 1
Makefile:1550: fallo en las instrucciones para el objetivo '_module_/usr/src/rtl8812au'
make[1]: *** [_module_/usr/src/rtl8812au] Error 2
make[1]: se sale del directorio '/usr/src/linux-headers-4.13.0-32-generic'
Makefile:1570: fallo en las instrucciones para el objetivo 'modules'
make: *** [modules] Error 2

`

barreira89, 2018-02-18:
Having the same issue with Ubuntu 16.04.3 LTS.

barreira89, 2018-02-25:
@fers1, I tried a different repo and branch with a similar set up and it seemed to work for the version of the kernel that I have:

https://github.com/VitalKoshalew/rtl8812AU/tree/driver-4.3.22-beta-mod

AlexKvazos, 2018-02-25:
`sudo apt-get install dkms`

fers1, 2018-02-26:
@barreira89  tampoco compila. No work.

`cd /usr/src/rtl8812au && sudo make clean && sudo make && sudo make install && sudo modprobe -a rtl8812au Clonar en «rtl8812AU»...
remote: Counting objects: 2154, done.
remote: Total 2154 (delta 0), reused 0 (delta 0), pack-reused 2154
Receiving objects: 100% (2154/2154), 7.88 MiB | 453.00 KiB/s, done.
Resolving deltas: 100% (1131/1131), done.
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.13.0-36-generic/build M=/usr/src/rtl8812au clean
make[1]: se entra en el directorio '/usr/src/linux-headers-4.13.0-36-generic'
  CLEAN   /usr/src/rtl8812au/.tmp_versions
make[1]: se sale del directorio '/usr/src/linux-headers-4.13.0-36-generic'
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.13.0-36-generic/build M=/usr/src/rtl8812au  modules
make[1]: se entra en el directorio '/usr/src/linux-headers-4.13.0-36-generic'
  CC [M]  /usr/src/rtl8812au/core/rtw_cmd.o
In file included from /usr/src/rtl8812au/include/drv_types.h:32:0,
                 from /usr/src/rtl8812au/core/rtw_cmd.c:22:
/usr/src/rtl8812au/include/osdep_service.h: In function ‘thread_enter’:
/usr/src/rtl8812au/include/osdep_service.h:343:2: error: implicit declaration of function ‘allow_signal’; did you mean ‘do_signal’? [-Werror=implicit-function-declaration]
  allow_signal(SIGTERM);
  ^~~~~~~~~~~~
  do_signal
/usr/src/rtl8812au/include/osdep_service.h: In function ‘flush_signals_thread’:
/usr/src/rtl8812au/include/osdep_service.h:353:6: error: implicit declaration of function ‘signal_pending’; did you mean ‘timer_pending’? [-Werror=implicit-function-declaration]
  if (signal_pending (current)) {
      ^~~~~~~~~~~~~~
      timer_pending
/usr/src/rtl8812au/include/osdep_service.h:354:3: error: implicit declaration of function ‘flush_signals’; did you mean ‘do_signal’? [-Werror=implicit-function-declaration]
   flush_signals(current);
   ^~~~~~~~~~~~~
   do_signal
cc1: all warnings being treated as errors
scripts/Makefile.build:308: fallo en las instrucciones para el objetivo '/usr/src/rtl8812au/core/rtw_cmd.o'
make[2]: *** [/usr/src/rtl8812au/core/rtw_cmd.o] Error 1
Makefile:1550: fallo en las instrucciones para el objetivo '_module_/usr/src/rtl8812au'
make[1]: *** [_module_/usr/src/rtl8812au] Error 2
make[1]: se sale del directorio '/usr/src/linux-headers-4.13.0-36-generic'
Makefile:1570: fallo en las instrucciones para el objetivo 'modules'
make: *** [modules] Error 2
`

barreira89, 2018-02-27:
@fers1, you tried using the other repo and checking out the specific branch? That repo also has some instructions for how to use dkms to install. But I've found with the fixes in that branch that you can build using the commands that you used. Could you show the results of `git status` and `git remote -v`?

RustanHakansson, 2018-03-02:
@barreira89 I am trying the same thing and seem to be getting the same results as @fers1 

I followed these instructions on https://github.com/VitalKoshalew/rtl8812AU/tree/driver-4.3.22-beta-mod
```
# DRV_NAME=rtl8812AU
# DRV_VERSION=4.3.14
# mkdir /usr/src/${DRV_NAME}-${DRV_VERSION}
# git archive driver-${DRV_VERSION} | tar -x -C /usr/src/${DRV_NAME}-${DRV_VERSION}
# dkms add -m ${DRV_NAME} -v ${DRV_VERSION}
# dkms build -m ${DRV_NAME} -v ${DRV_VERSION}
# dkms install -m ${DRV_NAME} -v ${DRV_VERSION}
```

Errors on the second to last step:
```

# dkms build -m ${DRV_NAME} -v ${DRV_VERSION}

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make'....(bad exit status: 2)
ERROR (dkms apport): binary package for rtl8812AU: 4.3.14 not found
Error! Bad return status for module build on kernel: 4.13.0-36-generic (x86_64)
Consult /var/lib/dkms/rtl8812AU/4.3.14/build/make.log for more information.

```
Showing the contents of that file:
```
# cat /var/lib/dkms/rtl8812AU/4.3.14/build/make.log
DKMS make.log for rtl8812AU-4.3.14 for kernel 4.13.0-36-generic (x86_64)
fre  2 mar 2018 13:22:10 CET
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.13.0-36-generic/build M=/var/lib/dkms/rtl8812AU/4.3.14/build  modules
make[1]: Går till katalogen ”/usr/src/linux-headers-4.13.0-36-generic”
  CC [M]  /var/lib/dkms/rtl8812AU/4.3.14/build/core/rtw_cmd.o
In file included from /var/lib/dkms/rtl8812AU/4.3.14/build/include/drv_types.h:32:0,
                 from /var/lib/dkms/rtl8812AU/4.3.14/build/core/rtw_cmd.c:22:
/var/lib/dkms/rtl8812AU/4.3.14/build/include/osdep_service.h: In function ‘thread_enter’:
/var/lib/dkms/rtl8812AU/4.3.14/build/include/osdep_service.h:343:2: error: implicit declaration of function ‘allow_signal’; did you mean ‘do_signal’? [-Werror=implicit-function-declaration]
  allow_signal(SIGTERM);
  ^~~~~~~~~~~~
  do_signal
/var/lib/dkms/rtl8812AU/4.3.14/build/include/osdep_service.h: In function ‘flush_signals_thread’:
/var/lib/dkms/rtl8812AU/4.3.14/build/include/osdep_service.h:353:6: error: implicit declaration of function ‘signal_pending’; did you mean ‘timer_pending’? [-Werror=implicit-function-declaration]
  if (signal_pending (current))
      ^~~~~~~~~~~~~~
      timer_pending
/var/lib/dkms/rtl8812AU/4.3.14/build/include/osdep_service.h:355:3: error: implicit declaration of function ‘flush_signals’; did you mean ‘do_signal’? [-Werror=implicit-function-declaration]
   flush_signals(current);
   ^~~~~~~~~~~~~
   do_signal
cc1: some warnings being treated as errors
scripts/Makefile.build:308: receptet för målet ”/var/lib/dkms/rtl8812AU/4.3.14/build/core/rtw_cmd.o” misslyckades
make[2]: *** [/var/lib/dkms/rtl8812AU/4.3.14/build/core/rtw_cmd.o] Fel 1
Makefile:1550: receptet för målet ”_module_/var/lib/dkms/rtl8812AU/4.3.14/build” misslyckades
make[1]: *** [_module_/var/lib/dkms/rtl8812AU/4.3.14/build] Fel 2
make[1]: Lämnar katalogen ”/usr/src/linux-headers-4.13.0-36-generic”
Makefile:1576: receptet för målet ”modules” misslyckades
make: *** [modules] Fel 2
```

Here is the output you asked for

```
# git status
På grenen driver-4.3.14
Din gren är à jour med "origin/driver-4.3.14".

inget att checka in, arbetskatalogen ren

# git remote -v
origin	https://github.com/VitalKoshalew/rtl8812AU.git (fetch)
origin	https://github.com/VitalKoshalew/rtl8812AU.git (push)
```

Any ideas of what I am doing wrong?

RustanHakansson, 2018-03-02:
For reference/searchability: I have a Linksys Max-Stream AC1200 Mu-Mimo USB adapter, model WUSB6400M-EU, plugging it in gives me this:

```
# dmesg
[ 3854.625076] usb 1-3: new high-speed USB device number 4 using xhci_hcd
[ 3854.851762] usb 1-3: New USB device found, idVendor=13b1, idProduct=0043
[ 3854.851766] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 3854.851768] usb 1-3: Product: WUSB6400M
[ 3854.851770] usb 1-3: Manufacturer: Linksys
[ 3854.851772] usb 1-3: SerialNumber: 123456

```

barreira89, 2018-03-02:
@RustanHakansson You'll need to checkout the branch with the latest updates:
`git checkout driver-4.3.22-beta-mod`

Also, I'm not sure if it's necessary, but I change the name of the `# DRV_VERSION=4.3.14` variable to match the version of the driver I was trying to install (`4.3.22-beta-mod`).

RustanHakansson, 2018-03-02:
@barreira89 Thanks, now all commands were ok! But what to do next? 

This gives no output: 

`$ sudo modprobe -a rtl8812au`

Settings page for wifi (I am on Ubuntu 17.10) says no found adapters. 

I saw on one page that I should reboot, tried that, unplugging and replugging, no effect.

RustanHakansson, 2018-03-02:
It is not listed when I run iwconfig. Here is output from modinfo:

```
$ sudo modinfo rtl8812au
filename:       /lib/modules/4.13.0-36-generic/updates/dkms/rtl8812au.ko
version:        v4.3.14_13455.20150212_BTCOEX20150128-51
author:         Realtek Semiconductor Corp.
description:    Realtek Wireless Lan Driver
license:        GPL
srcversion:     C29830757A11C5C39B68DE8
alias:          usb:v0BDApA811d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v056Ep4007d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0411p029Bd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0411p025Dd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0411p0242d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2019pAB32d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0846p9052d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0E66p0023d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p3318d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p3314d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v04BBp0953d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDApA811d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v7392pA813d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v7392pA812d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v7392pA811d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp0823d*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v0BDAp0820d*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v0BDAp8822d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp0821d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp0811d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v148Fp9097d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p0122d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p010Fd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p0103d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p010Ed*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p010Dd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p0101d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v20F4p805Bd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p3316d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p3315d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v07B8p8812d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2019pAB30d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v1740p0100d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v13B1p003Fd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v1058p0632d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p3313d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0586p3426d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0E66p0022d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0B05p17D2d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0409p0408d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0789p016Ed*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v04BBp0952d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0DF6p0074d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v7392pA822d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p330Ed*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v050Dp1109d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v050Dp1106d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp881Cd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp881Bd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp881Ad*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp8812d*dc*dsc*dp*ic*isc*ip*in*
depends:        cfg80211
name:           rtl8812au
vermagic:       4.13.0-36-generic SMP mod_unload 
parm:           rtw_ips_mode:The default IPS mode (int)
parm:           rtw_usb_rxagg_mode:int
parm:           rtw_qos_opt_enable:int
parm:           ifname:The default name to allocate for first interface (charp)
parm:           if2name:The default name to allocate for second interface (charp)
parm:           rtw_initmac:charp
[... 28 lines cut ...]
parm:           rtw_hwpwrp_detect:int
parm:           rtw_hw_wps_pbc:int
parm:           rtw_max_roaming_times:The max roaming times to try (uint)
parm:           rtw_mc2u_disable:int
parm:           rtw_80211d:Enable 802.11d mechanism (int)
parm:           rtw_notch_filter:0:Disable, 1:Enable, 2:Enable only for P2P (uint)
parm:           rtw_led_enable:Enable status LED (int)
parm:           rtw_hiq_filter:0:allow all, 1:allow special, 2:deny all (uint)
parm:           rtw_adaptivity_en:0:disable, 1:enable (uint)
parm:           rtw_adaptivity_mode:0:normal, 1:carrier sense (uint)
parm:           rtw_adaptivity_dml:0:disable, 1:enable (uint)
parm:           rtw_amplifier_type_2g:BIT3:2G ext-PA, BIT4:2G ext-LNA (uint)
parm:           rtw_amplifier_type_5g:BIT6:5G ext-PA, BIT7:5G ext-LNA (uint)
parm:           rtw_RFE_type:default init value:64 (uint)
parm:           rtw_TxBBSwing_2G:default init value:0xFF (uint)
parm:           rtw_TxBBSwing_5G:default init value:0xFF (uint)
parm:           rtw_OffEfuseMask:default open Efuse Mask vaule:0 (uint)
parm:           rtw_FileMaskEfuse:default drv Mask Efuse vaule:0 (uint)
parm:           rtw_tx_pwr_lmt_enable:0:Disable, 1:Enable, 2: Depend on efuse (int)
parm:           rtw_tx_pwr_by_rate:0:Disable, 1:Enable, 2: Depend on efuse (int)
parm:           rtw_phy_file_path:The path of phy parameter (charp)
parm:           rtw_load_phy_file:PHY File Bit Map (int)
parm:           rtw_decrypt_phy_file:Enable Decrypt PHY File (int)

```

barreira89, 2018-03-02:
@RustanHakansson try
``sudo modprobe -a 8812au`

RustanHakansson, 2018-03-02:
> sudo modprobe -a 8812au

No difference to iwconfig after running this and plugging it in. 

Is your device recognized when you plug it in? Do you have the same as I?

fers1, 2018-03-03:
@barreira89 

`**cd /usr/src/rtl8812au && sudo make clean && sudo make && sudo make install && sudo modprobe -a rtl8812au** 

Clonar en «rtl8812AU_8821AU_linux»...
remote: Counting objects: 1578, done.
remote: Compressing objects: 100% (9/9), done.
remote: Total 1578 (delta 4), reused 4 (delta 4), pack-reused 1565
Receiving objects: 100% (1578/1578), 3.58 MiB | 437.00 KiB/s, done.
Resolving deltas: 100% (976/976), done.
[sudo] password for fernando: 
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.13.0-36-generic/build M=/usr/src/rtl8812au clean
make[1]: se entra en el directorio '/usr/src/linux-headers-4.13.0-36-generic'
  CLEAN   /usr/src/rtl8812au/.tmp_versions
make[1]: se sale del directorio '/usr/src/linux-headers-4.13.0-36-generic'
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.13.0-36-generic/build M=/usr/src/rtl8812au  modules
make[1]: se entra en el directorio '/usr/src/linux-headers-4.13.0-36-generic'
  CC [M]  /usr/src/rtl8812au/core/rtw_cmd.o
In file included from /usr/src/rtl8812au/include/drv_types.h:32:0,
                 from /usr/src/rtl8812au/core/rtw_cmd.c:22:
/usr/src/rtl8812au/include/osdep_service.h: In function ‘thread_enter’:
/usr/src/rtl8812au/include/osdep_service.h:343:2: error: implicit declaration of function ‘allow_signal’; did you mean ‘do_signal’? [-Werror=implicit-function-declaration]
  allow_signal(SIGTERM);
  ^~~~~~~~~~~~
  do_signal
/usr/src/rtl8812au/include/osdep_service.h: In function ‘flush_signals_thread’:
/usr/src/rtl8812au/include/osdep_service.h:353:6: error: implicit declaration of function ‘signal_pending’; did you mean ‘timer_pending’? [-Werror=implicit-function-declaration]
  if (signal_pending (current)) {
      ^~~~~~~~~~~~~~
      timer_pending
/usr/src/rtl8812au/include/osdep_service.h:354:3: error: implicit declaration of function ‘flush_signals’; did you mean ‘do_signal’? [-Werror=implicit-function-declaration]
   flush_signals(current);
   ^~~~~~~~~~~~~
   do_signal
cc1: all warnings being treated as errors
scripts/Makefile.build:308: fallo en las instrucciones para el objetivo '/usr/src/rtl8812au/core/rtw_cmd.o'
make[2]: *** [/usr/src/rtl8812au/core/rtw_cmd.o] Error 1
Makefile:1550: fallo en las instrucciones para el objetivo '_module_/usr/src/rtl8812au'
make[1]: *** [_module_/usr/src/rtl8812au] Error 2
make[1]: se sale del directorio '/usr/src/linux-headers-4.13.0-36-generic'
Makefile:1570: fallo en las instrucciones para el objetivo 'modules'
make: *** [modules] Error 2

**f@h:/usr/src/rtl8812au$ git status**
En la rama master
Su rama está actualizada con «origin/master».

Archivos sin seguimiento:
  (use «git add <archivo>...» para incluir en lo que se ha de confirmar)

	core/.rtw_cmd.o.d
	rtl8812AU/
	rtl8812AU_8821AU_linux/
	rtl8812au/

no se ha agregado nada al commit pero existen archivos sin seguimiento (use «git add» para darle seguimiento)

**f@h:/usr/src/rtl8812au$ git remote -v**

origin	https://github.com/abperiasamy/rtl8812AU_8821AU_linux (fetch)
origin	https://github.com/abperiasamy/rtl8812AU_8821AU_linux (push)

`
using the other repo https://github.com/VitalKoshalew/rtl8812AU

`**sudo make clean && sudo make && sudo make install && sudo modprobe -a rtl8812au** Clonar en «rtl8812AU»...

remote: Counting objects: 2154, done.
remote: Total 2154 (delta 0), reused 0 (delta 0), pack-reused 2154
Receiving objects: 100% (2154/2154), 7.88 MiB | 459.00 KiB/s, done.
[... 30 lines cut ...]
/usr/src/rtl8812AU/include/osdep_service.h:355:3: error: implicit declaration of function ‘flush_signals’; did you mean ‘do_signal’? [-Werror=implicit-function-declaration]
   flush_signals(current);
   ^~~~~~~~~~~~~
   do_signal
cc1: some warnings being treated as errors
scripts/Makefile.build:308: fallo en las instrucciones para el objetivo '/usr/src/rtl8812AU/core/rtw_cmd.o'
make[2]: *** [/usr/src/rtl8812AU/core/rtw_cmd.o] Error 1
Makefile:1550: fallo en las instrucciones para el objetivo '_module_/usr/src/rtl8812AU'
make[1]: *** [_module_/usr/src/rtl8812AU] Error 2
make[1]: se sale del directorio '/usr/src/linux-headers-4.13.0-36-generic'
Makefile:1576: fallo en las instrucciones para el objetivo 'modules'
make: *** [modules] Error 2

**$ git status**

En la rama driver-4.3.14
Su rama está actualizada con «origin/driver-4.3.14».

nothing to commit, working tree clean

**$ git remote -v**

origin	https://github.com/VitalKoshalew/rtl8812AU (fetch)
origin	https://github.com/VitalKoshalew/rtl8812AU (push)
`

RustanHakansson, 2018-03-04:
@fers1 Seems you need to do the same thing that I did, as instructed by barreira89 above:

git checkout driver-4.3.22-beta-mod

and

# DRV_VERSION=4.3.22-beta-mod

RustanHakansson, 2018-03-07:
I gave up and returned my Linksys Max-Stream AC1200.