abperiasamy/rtl8812AU_8821AU_linux #246: Not working on Debian 9, Kernal 4.9.0-5-amd64

soorajvnair, 2018-02-21:
Hey there,

I'm able to compile the driver for the kernel version using the DKMS module,
but unfortunately, it just isn't working correctly.

chaotaklon, 2019-03-07:
added support for linux v4.20
https://github.com/chaotaklon/rtl8812AU_8821AU_linux

sudo-bmitch, 2019-10-07:
I'm also facing issues with this kernel.

```
$ uname -rv
4.9.0-9-amd64 #1 SMP Debian 4.9.168-1+deb9u5 (2019-08-11)
```

I can see networks show up in a scan, but trying to connect to one of them leaves me in an auth loop. I'm seeing the following in /var/log/messages:

```
Oct  6 20:05:43 bmitch-asusr556l NetworkManager[26303]: <info>  [1570406743.6550] device (wlx34e894d64636): supplicant interface state: disconnected -> scanning
Oct  6 20:05:47 bmitch-asusr556l kernel: [327185.319694] RTL871X: rtw_set_802_11_connect(wlx34e894d64636)  fw_state=0x00000008                                                            
Oct  6 20:05:47 bmitch-asusr556l NetworkManager[26303]: <info>  [1570406747.3231] device (wlx34e894d64636): supplicant interface state: scanning -> associating
Oct  6 20:05:47 bmitch-asusr556l kernel: [327185.537811] RTL871X: start auth                                                                                                              
Oct  6 20:05:49 bmitch-asusr556l NetworkManager[26303]: <info>  [1570406749.1489] device (wlx34e894d64636): supplicant interface state: associating -> disconnected
Oct  6 20:05:59 bmitch-asusr556l NetworkManager[26303]: <info>  [1570406759.4101] device (wlx34e894d64636): supplicant interface state: disconnected -> scanning
Oct  6 20:06:00 bmitch-asusr556l kernel: [327198.048202] RTL871X: rtw_set_802_11_connect(wlx34e894d64636)  fw_state=0x00000008                                  
Oct  6 20:06:00 bmitch-asusr556l NetworkManager[26303]: <info>  [1570406760.0530] device (wlx34e894d64636): supplicant interface state: scanning -> associating                           
Oct  6 20:06:00 bmitch-asusr556l kernel: [327198.197898] RTL871X: start auth                                                                                    
Oct  6 20:06:01 bmitch-asusr556l NetworkManager[26303]: <info>  [1570406761.7888] device (wlx34e894d64636): supplicant interface state: associating -> disconnected
Oct  6 20:06:08 bmitch-asusr556l NetworkManager[26303]: <warn>  [1570406768.7368] device (wlx34e894d64636): Activation: (wifi) association took too long        
Oct  6 20:06:08 bmitch-asusr556l NetworkManager[26303]: <info>  [1570406768.7368] device (wlx34e894d64636): state change: config -> need-auth (reason 'none') [50 60 0]                   
Oct  6 20:06:08 bmitch-asusr556l NetworkManager[26303]: <warn>  [1570406768.7396] device (wlx34e894d64636): Activation: (wifi) asking for new secrets    
```

Anther network adapter (RTL8723BE) on the same laptop using the Debian "firmware-realtek" package does not experience this issue, NetworkManager completes the authentication. This is based on a recent clone of the repo:

```
$ git rev-parse --short HEAD
fa68771
```

Let me know if there's further debugging details I can provide.

guyru, 2019-10-09:
I'm facing a similar errors using kernel 4.19.0-6-amd64 on Debian Buster.