abperiasamy/rtl8812AU_8821AU_linux #247: TP-LINK AC 1300 - Archer T4U

AlexKvazos, 2018-02-25:
I am rying to use this driver to use it with the device in subject. I can see the proper device id as it matches the one I see when I plug my device and run `lsusb`.

https://github.com/abperiasamy/rtl8812AU_8821AU_linux/blob/master/os_dep/linux/usb_intf.c#L300

The compilation succeeds though no network card appears even after restarting the `network-manager` service or rebooting the whole system.

Running Kali Linux on Linux Kernel 4.14 inside VMWare with the device attached to the virtual machine.

Caleb9, 2018-09-05:
Just purchased this device today because it was supposed to work on Linux without issues ;).

Not sure if anyone's still interested but the only way to make it work without problems so far was to use [this fork](https://github.com/hexonxons/rtl8812AU/tree/add_t4u_ac1300_def) of the [original driver](https://github.com/diederikdehaas/rtl8812AU), specifically from the _add_t4u_ac1300_def_ branch. There is an already quite ripe [pull request](https://github.com/diederikdehaas/rtl8812AU/pull/105) to the original repo, but the owner seems to be busy elsewhere or something...

Anyway I had to manually build and install it as it does not provide `Makefile.dkms` and I didn't want to bother setting it up, so there will be manual work on each kernel update unless [abperiasamy](https://github.com/abperiasamy/rtl8812AU_8821AU_linux/issues/247) version gets fixed at some point. Then I added the module to `/etc/modules.d/` so it loads automatically on boot.

Nevertheless I can report that the connection works without issues, also after rebooting. Oh, I'm on Linux Mint, 4.15.0.33 kernel.

daylanc, 2018-10-02:
@Caleb9 I am new to Ubuntu and was given the TP-LINK ac1300 Archer T4U version 2.0 USB wifi adapter. I need to install the driver on Ubuntu 16.04 with kernel 4.15.0.36. Could you tell me exactly how to install it? I am having a hard time following your comment above due to not knowing much about Ubuntu.
Thank you!

Caleb9, 2018-10-02:
@daylanc I don't use Ubuntu but I did try it in VirtualBox and unfortunately it seems there's some compiler related problem with Ubuntu in particular. Fixing it seems to be too involved however, as it would require patching and building a custom Linux kernel, if I am not mistaken

So, what I would suggest, if you have such possibility, is to try to change the distribution from Ubuntu to Linux Mint, where this driver compiles and works, I'm guessing because there are some patches applied to the kernel - Mint is based on Ubuntu and is very similar to it, using the same software packages etc.

I realize this might not be possible for you, but if you decide to go down that path, then ping me and I'll describe the steps needed to make it work in Linux Mint. Otherwise maybe someone who is more skilled in Ubuntu can provide easy-for-a-rookie steps to resolve this?

AlexKvazos, 2018-10-05:
Closing due to innactivity

closed 2018-10-05 by AlexKvazos

mukesh2020, 2019-01-04:
I am using Kali Linux plasma 
Recently I bought TU4 V2 ac1300 adapter for Linux but it is not working and I am new to Linux environment so kindly help me.

kimocoder, 2019-01-05:
@mukesh2020 what is the VID/PID? Post the output og "lsusb" with adapter plugged in

mukesh2020, 2019-01-07:
/bin/sh: 1: bc: not found
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.18.0-kali3-amd64/build M=/root/Desktop/rtl8812AU_linux_v5.2.20_25672.20171213  modules
make[1]: Entering directory '/usr/src/linux-headers-4.18.0-kali3-amd64'
/bin/sh: 1: bc: not found
  CC [M]  /root/Desktop/rtl8812AU_linux_v5.2.20_25672.20171213/core/rtw_cmd.o
In file included from /root/Desktop/rtl8812AU_linux_v5.2.20_25672.20171213/include/osdep_service.h:47:0,
                 from /root/Desktop/rtl8812AU_linux_v5.2.20_25672.20171213/include/drv_types.h:27,
                 from /root/Desktop/rtl8812AU_linux_v5.2.20_25672.20171213/core/rtw_cmd.c:17:
/root/Desktop/rtl8812AU_linux_v5.2.20_25672.20171213/include/osdep_service_linux.h: In function ‘_init_timer’:
/root/Desktop/rtl8812AU_linux_v5.2.20_25672.20171213/include/osdep_service_linux.h:299:8: error: ‘_timer {aka struct timer_list}’ has no member named ‘data’
  ptimer->data = (unsigned long)cntx;
        ^~
/root/Desktop/rtl8812AU_linux_v5.2.20_25672.20171213/include/osdep_service_linux.h:300:2: error: implicit declaration of function ‘init_timer’; did you mean ‘_init_timer’? [-Werror=implicit-function-declaration]
  init_timer(ptimer);
  ^~~~~~~~~~
  _init_timer
cc1: some warnings being treated as errors
make[4]: *** [/usr/src/linux-headers-4.18.0-kali3-common/scripts/Makefile.build:323: /root/Desktop/rtl8812AU_linux_v5.2.20_25672.20171213/core/rtw_cmd.o] Error 1
make[3]: *** [/usr/src/linux-headers-4.18.0-kali3-common/Makefile:1519: _module_/root/Desktop/rtl8812AU_linux_v5.2.20_25672.20171213] Error 2
make[2]: *** [Makefile:146: sub-make] Error 2
make[1]: *** [Makefile:8: all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.18.0-kali3-amd64'
make: *** [Makefile:1828: modules] Error 2

mukesh2020, 2019-01-07:
running command make
for installing the tplink  t4u 
ac1300