abperiasamy/rtl8812AU_8821AU_linux #249: Issues on Fedora 27

christiano, 2018-03-18:
Hello,

First of all, thanks for this awesome contribution! 

I'm having an issue with Fedora 27, kernel 4.15.9-300.fc27.x86_64

From time to time, the NetworkManager disconnects and connect again one second later. This issue occurs at least twice per hour. 

The Dmesg shows the message below. Do you have an idea what could be wrong? Any fix for this? Thanks again!

> [28137.584533] wlp2s0: associated
> [28152.227064] wlp2s0: authenticate with xx:xx:xx:xx:xx:xx
> [28152.245422] wlp2s0: send auth to xx:xx:xx:xx:xx:xx (try 1/3)
> [28152.252303] wlp2s0: authenticated
> [28152.252910] wlp2s0: associate with xx:xx:xx:xx:xx:xx (try 1/3)
> [28152.446469] wlp2s0: RX AssocResp from xx:xx:xx:xx:xx:xx (capab=0x401 status=0 aid=1)
> [28152.446599] wlp2s0: associated
> [28156.188319] wlp2s0: authenticate with xx:xx:xx:xx:xx:xx
> [28156.205289] wlp2s0: send auth to xx:xx:xx:xx:xx:xx (try 1/3)
> [28156.372942] wlp2s0: send auth to xx:xx:xx:xx:xx:xx (try 2/3)
> [28156.562448] wlp2s0: authenticated
> [28156.563686] wlp2s0: associate with xx:xx:xx:xx:xx:xx (try 1/3)
> [28156.752743] wlp2s0: associate with xx:xx:xx:xx:xx:xx (try 2/3)
> [28156.904722] wlp2s0: associate with xx:xx:xx:xx:xx:xx (try 3/3)
> [28157.069407] wlp2s0: association with xx:xx:xx:xx:xx:xx timed out
>

dschwartz783, 2018-04-10:
I’m having this exact problem.

dschwartz783, 2018-04-10:
What model of AP are you using?

dschwartz783, 2018-04-10:
You might want to try https://github.com/aircrack-ng/rtl8812au

It is working better for me so far, as well as supporting more features