abperiasamy/rtl8812AU_8821AU_linux #253: AnewKodi AC1200 - Ubuntu 18.04 kernel 4.16.12-041612-generic

helektron, 2018-06-04:
Hi, I installed the drivers without problems but usb wifi acapter doesn't turn on the light and I can't scan 5ghz wifi. How can I fix it?

Thanks!

helektron, 2018-06-04:
It's "Realtek Semiconductor Corp." Output:

`➜  ~ lsusb
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 006: ID 0bda:0129 Realtek Semiconductor Corp. RTS5129 Card Reader Controller
Bus 001 Device 007: ID 0cf3:3004 Atheros Communications, Inc. AR3012 Bluetooth 4.0
Bus 001 Device 004: ID 046d:c063 Logitech, Inc. DELL Laser Mouse
Bus 001 Device 003: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 003: ID 0bda:0411 Realtek Semiconductor Corp. 
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 005: ID 5986:029d Acer, Inc 
Bus 003 Device 003: ID 03eb:8815 Atmel Corp. 
Bus 003 Device 007: ID 0bda:b812 Realtek Semiconductor Corp. 
Bus 003 Device 006: ID 0bda:5411 Realtek Semiconductor Corp. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`

helektron, 2018-06-04:
rfkill list
0: ideapad_wlan: Wireless LAN
	Soft blocked: no
	Hard blocked: no
1: ideapad_bluetooth: Bluetooth
	Soft blocked: no
	Hard blocked: no
2: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
3: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no

kimocoder, 2018-06-04:
Yup. I don't think it's added.. you can add it manually to "os_dep/linux/usb_intfs.c" file yourself, then compile it again :)

helektron, 2018-06-04:
What do I need to add to that file?

Thanks!

kimocoder, 2018-06-04:
You will need to add the "0bda:b812" in the correct place (under 8812), I may do it tomorrow as I'm not close to my computer at the time.

helektron, 2018-06-04:
I Will try It! I Will keep you posted. Thanks!

kimocoder, 2018-06-04:
Just copy this line and edit the VID/PID and compile

[image]

kimocoder, 2018-06-04:
Just copy this line and edit the VID/PID after the "0x" sections

[image]

helektron, 2018-06-04:
I did it but It doesn't work. Any idea? Many thanks!

helektron, 2018-06-04:
Some outputs:

**modinfo rtl8812au | grep 0BDA**
alias:          usb:v0BDApA811d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDApA811d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp0823d*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v0BDAp0820d*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v0BDAp8822d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp0821d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp0811d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp881Cd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp881Bd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp881Ad*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp8812d*dc*dsc*dp*ic*isc*ip*in*

**rfkill list all**
0: ideapad_wlan: Wireless LAN
	Soft blocked: no
	Hard blocked: no
1: ideapad_bluetooth: Bluetooth
	Soft blocked: no
	Hard blocked: no
2: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
3: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no

**sudo dkms status**
rtl8812au, 4.2.2: added
rtl8812AU, 5: added
rtl8812AU_8821AU_linux, 1.0, 4.16.12-041612-generic, x86_64: installed (WARNING! Diff between built and installed module!)
rtl8812AU_8821AU_linux, 1.0, 4.17.0-041700-generic, x86_64: installed

**lsmod | grep 8812**         
rtl8812au            1359872  0
cfg80211              659456  5 rtl8812au,ath9k_common,ath9k,ath,mac80211

**dmesg | grep -i -e rtl -e 8812**
[    5.669411] r8169 0000:02:00.0 eth0: RTL8105e at 0x        (ptrval), 20:89:84:df:8b:17, XID 00c00000 IRQ 25
[  112.590772] RTL871X: module init start
[  112.590774] RTL871X: rtl8812au v4.3.14_13455.20150212_BTCOEX20150128-51
[  112.590775] RTL871X: rtl8812au BT-Coex version = BTCOEX20150128-51
[  112.590828] usbcore: registered new interface driver rtl8812au
[  112.590829] RTL871X: module init ret=0

helektron, 2018-06-04:
It seems to be recognized but I doesn't scan wifi:

**iwconfig**
lo        no wireless extensions.

**wlx00e04c881200  unassociated  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0**

wlp3s0    IEEE 802.11  ESSID:"JAZZTEL_uTN4"  
          Mode:Managed  Frequency:2.462 GHz  Access Point: B8:05:AB:F5:49:80   
          Bit Rate=58.5 Mb/s   Tx-Power=15 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=69/70  Signal level=-41 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:121   Missed beacon:0

enp2s0    no wireless extensions.

helektron, 2018-06-06:
@kimocoder sorry but network with new adapter is still disabled following your steps. Can you help me? Many thanks!

helektron, 2018-06-06:
Hi @kimocoder thanks but I've the same problem (I removed old module before installing the new one). The error seems to be with the interface trying to set it up:

```
Jun  6 11:48:21 helektron wpa_supplicant[1101]: Could not set interface wlx00e04c881200 flags (UP): Operation not permitted
Jun  6 11:48:21 helektron wpa_supplicant[1101]: nl80211: Could not set interface 'wlx00e04c881200' UP
Jun  6 11:48:21 helektron wpa_supplicant[1101]: nl80211: deinit ifname=wlx00e04c881200 disabled_11b_rates=0
Jun  6 11:48:24 helektron wpa_supplicant[1101]: Could not set interface wlx00e04c881200 flags (UP): Operation not permitted
Jun  6 11:48:24 helektron wpa_supplicant[1101]: WEXT: Could not set interface 'wlx00e04c881200' UP
Jun  6 11:48:24 helektron wpa_supplicant[1101]: wlx00e04c881200: Failed to initialize driver interface
Jun  6 11:48:24 helektron NetworkManager[1074]: <error> [1528278504.1902] sup-iface[0x5596fc421580,wlx00e04c881200]: error adding interface: wpa_supplicant couldn't grab this interface.
Jun  6 11:48:24 helektron NetworkManager[1074]: <info>  [1528278504.1902] device (wlx00e04c881200): supplicant interface state: starting -> down
Jun  6 11:48:34 helektron NetworkManager[1074]: <warn>  [1528278514.0445] device (wlx00e04c881200): re-acquiring supplicant interface (#3).
Jun  6 11:48:36 helektron wpa_supplicant[1101]: Could not set interface wlx00e04c881200 flags (UP): Operation not permitted
Jun  6 11:48:36 helektron wpa_supplicant[1101]: nl80211: Could not set interface 'wlx00e04c881200' UP
Jun  6 11:48:36 helektron wpa_supplicant[1101]: nl80211: deinit ifname=wlx00e04c881200 disabled_11b_rates=0
Jun  6 11:48:39 helektron wpa_supplicant[1101]: Could not set interface wlx00e04c881200 flags (UP): Operation not permitted
Jun  6 11:48:39 helektron wpa_supplicant[1101]: WEXT: Could not set interface 'wlx00e04c881200' UP
Jun  6 11:48:39 helektron wpa_supplicant[1101]: wlx00e04c881200: Failed to initialize driver interface
Jun  6 11:48:39 helektron NetworkManager[1074]: <error> [1528278519.1663] sup-iface[0x5596fc421630,wlx00e04c881200]: error adding interface: wpa_supplicant couldn't grab this interface.
Jun  6 11:48:39 helektron NetworkManager[1074]: <info>  [1528278519.1664] device (wlx00e04c881200): supplicant interface state: starting -> down
```

helektron, 2018-06-08:
Fixed! At the end it was a wrong driver. The correct driver for that usb adapter is: rtl8822bu

I used this fork for compiling 4.15+ kernel:

https://github.com/FomalhautWeisszwerg/rtl8822bu

I hope it helps more people.

Thanks for your help!

Regards.

closed 2018-06-08 by helektron