abperiasamy/rtl8812AU_8821AU_linux #256: dropped packets..

dirtbag, 2018-06-23:
Im using a TP-LINK Archer T4U
Bus 004 Device 002: ID 2357:0101  
on fedora28   4.16.13-300.fc28.x86_64

For the most part, the device works great with this driver. But over ssh sessions, I get lag every so often.
Im trying to figure out whats causing it.. The server im connecting to has a really good connection to the internet and at home, ive got a 100Meg connection. 

wlp0s20u4  IEEE 802.11bgn  ESSID:"Wireless WiFi"  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency:2.412 GHz  Access Point: E0:CB:BC:34:AF:36   
          Bit Rate:174 Mb/s   Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:****-****-****-****-****-****-****-****   Security mode:open
          Power Management:off
          Link Quality=100/100  Signal level=100/100  Noise level=0/100
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

the wireless connection looks good, but I see dropped packets 

wlp0s20u4: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.1.1.127  netmask 255.255.255.0  broadcast 10.1.1.255
        inet6 fe80::b3ba:76d3:d856:c74f  prefixlen 64  scopeid 0x20<link>
        ether 84:16:f9:1b:54:a7  txqueuelen 1000  (Ethernet)
        RX packets 38434  bytes 49292956 (47.0 MiB)
        RX errors 0  dropped 240  overruns 0  frame 0
        TX packets 19921  bytes 2436955 (2.3 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

thats over the course of about 10 minutes or so.

Any ideas?

-db

xbgmsharp, 2018-07-11:
I also do see packet drop.
```
# ifconfig wlxe46f13ea6bc6
wlxe46f13ea6bc6: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet6 fe80::e66f:13ff:feea:6bc6  prefixlen 64  scopeid 0x20<link>
        ether e4:6f:13:ea:6b:c6  txqueuelen 1000  (Ethernet)
        RX packets 27271509  bytes 865342263 (825.2 MiB)
        RX errors 0  dropped 32896  overruns 0  frame 0
        TX packets 47197146  bytes 2908532240 (2.7 GiB)
        TX errors 0  dropped 4183 overruns 0  carrier 0  collisions 0
```

```
# iwconfig wlxe46f13ea6bc6
wlxe46f13ea6bc6  IEEE 802.11AC  ESSID:"_KaYa"  Nickname:"<WIFI@REALTEK>"
          Mode:Master  Frequency:5.2 GHz  Access Point: E4:6F:13:EA:6B:C6   
          Bit Rate:41 Mb/s   Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality=87/100  Signal level=65/100  Noise level=0/100
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```
```
# lsusb -v -s 1:2

Bus 001 Device 002: ID 2001:3315 D-Link Corp. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x2001 D-Link Corp.
  idProduct          0x3315 
  bcdDevice            0.00
  iManufacturer           1 Realtek
  iProduct                2 D-Link Wireless Adapter
  iSerial                 3 123456
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           53
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           5
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x85  EP 5 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      1
Device Status:     0x0002
  (Bus Powered)
  Remote Wakeup Enabled
```

Not sure how i could debug to identify the root cause. Any thoughts?
I am running Kernel 4.9.111 with rtl8812AU drivers as a module.

DrDoctor13, 2019-05-09:
Getting dropped packets here, too. Running kernel 5.0.13.
```

wlp0s20u10: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.19  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::8c99:6929:5c35:74  prefixlen 64  scopeid 0x20<link>
        ether 14:59:c0:3f:ae:11  txqueuelen 1000  (Ethernet)
        RX packets 9552923  bytes 36875778537 (34.3 GiB)
        RX errors 0  dropped 99180  overruns 0  frame 0
        TX packets 2602114  bytes 683042523 (651.4 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

```

```
Bus 001 Device 004: ID 0846:9052 NetGear, Inc. A6100 AC600 DB Wireless Adapter [Realtek RTL8811AU]
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0846 NetGear, Inc.
  idProduct          0x9052 A6100 AC600 DB Wireless Adapter [Realtek RTL8811AU]
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x003c
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           6
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              2 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x05  EP 5 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x06  EP 6 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x87  EP 7 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               3
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x08  EP 8 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x09  EP 9 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0

```